doe300/VC4C #1: Add linker

doe300, 2017-10-09:
Since [SPIRV-Tools](https://github.com/KhronosGroup/SPIRV-Tools) now implements a linker, we could use it, to provide a linker for the VC4C compiler, at least for SPIR-V code.

Provide an interface (can be separate from compiler-interface) to link together two SPIR-V binaries.

pierremoreau, 2017-12-04:
The linker in SPIRV-Tools won’t complain for missing symbols if those are annotated with the BuiltIn decoration, so you might be able to use that. But otherwise you are right, it does not support partial linking.

doe300, 2017-12-04:
I didn't know that. To make it work, I would need to hand-craft an SPIR-V module with all intrinsics declared and marked as BuiltIn (or hand-modify a generated module to mark all functions as BuiltIns), which I could then link into the other modules... could be worth a try.

> Couldn’t you create an OpenCL library out of your headers, which you would link against each time clLinkProgram() is called?

I can't create a "native" (as in compiled to the GPU's machine-code) library, since VC4C doesn't support this feature (it can only generate completely "linked" machine-code). The reason for this is, that all function-calls are inlined into the kernel-function (to save the trouble of defining an calling convention and saving registers/reserving registers for return values).

pierremoreau, 2017-12-05:
> I can't create a "native" (as in compiled to the GPU's machine-code) library, since VC4C doesn't support this feature (it can only generate completely "linked" machine-code). The reason for this is, [...]

If you extend the list of headers provided by the user to `clCreateProgramWithSource()` with your own headers, shouldn’t the compiler take care of inlining the functions if you marked them as “force inline”?
Otherwise, you can always compile the headers to a SPIR-V module and provide it to the linker. Then run `spirv-opt` which has a function inlining pass. The intrinsic functions should hopefully have the “Inline” function control on them, and the function inlining pass should honour it (I haven’t checked what it does).

doe300, 2017-12-05:
> If you extend the list of headers provided by the user to clCreateProgramWithSource() with your own headers, shouldn’t the compiler take care of inlining the functions if you marked them as “force inline”?

I do this and this works for most functions.

But the problem is, I have functions (intrinsics) which are declared in my headers, but the definition is only inserted (replacing the function-call) in m own compiler, which is run after the LLVM/SPIR-V compiler has finished.

E.g. I have a function `vc4cl_mutex_lock()` which locks the single hardware-mutex. Since neither LLVM nor SPIR-V has a fitting operation to lock a mutex, I need to replace this function with the corresponding machine-code manually. Also, since LLVM/SPIR-V has no such operation, I need to replace the function-call with the machine-code after linking the SPIR-V modules. This means, I would need to link the SPIR-V modules without (yet) being able to provide a definition for this function, which the linker does not accept.

So the only way I could think of making it work, is to link in an additional SPIR-V module with all the intrinsic functions declared as Builtin, as you suggested.

pierremoreau, 2017-12-05:
Ah my bad, I hadn’t thought you would have instructions that could not be expressed in SPIR-V.

It wouldn’t be difficult to add a flag to the SPIR-V linker to allow partial linking. Feel free to open an issue for that on SPIRV-Tools with your use case; I wonder if others might be interested by it as well. That would give you a less hacky way than marking those functions as BuiltIn.

closed 2018-02-18 by doe300

pierremoreau, 2018-02-27:
FYI the work has been merged into master.

commit a1d55949 (2018-02-18): Adds support for linking SPIR-V modules, closes #1

- configures SPIRV-Tools to link partially
- exports an easy interface to Precompiler for rewriting VC4CL
program-compilation
- improves selection of front-ends based on supported formats