doe300/VC4C #3: Pre-compiler deletes /dev/stdout

doe300, 2017-11-13:
Sometimes, when failing to compile an OpenCL C source-code, the pre-compiler (clang) deletes the symbolic link `/dev/stdout`. This results in any successive compilations with VC4C (and probably any other program using `stdout`) to fail.

To "fix" this, the `/dev/stdout` symbolic link needs to be re-created. This can be done by rebooting the Raspberry Pi.

Another "fix" is to configure VC4C with the CMake option `PRECOMPILER_DROP_RIGHTS` enabled, which will run the pre-compilation process with the user `pi` (instead of root, if run with sudo). This should prevent the process from deleting `/dev/stdout`.

Since I cannot reliably reproduce the issue, I don't know if the fix works. If anyone can create a stable reproduction, please notify me.

doe300, 2017-11-13:
I managed to reproduce the issue:

- When clang fails to compile a file, it deletes the output-file, if it can. This is on purpose since most of the time, the output-file is a temporary file, which is no longer needed when the compilation fails.
- Since we need to run any OpenCL application as root (to mmap the V3D registers), all processes started by this application are also run as root. Now clang has the rights required to delete its output-file `/dev/stdout`!

doe300, 2017-11-13:
To fix the issue, rewrite the pre-compilation:

- [ ] use temporary files for intermediate representations
- [ ] delete temporary files, even if compilation fails; but only after they are completely read (after the parser returns/after the second pre-compilation step has finished executing)
- [ ] if possible, enable `PRECOMPILER_DROP_RIGHTS`, so at least the pre-compiler does not run with root access. Need to make sure, it can still read/write the temporary files.

doe300, 2017-11-19:
Fixed with the latest commit

closed 2017-11-19 by doe300

paulreimer, 2018-01-15:
Could you update the "Known Issues" section of the `Readme.md`?