doe300/VC4C #6: Segmentation fault when running VC4C

senshu, 2017-11-27:
Running VC4C on a Raspberry Pi 3 with a fresh Raspbian Stretch setup.
VC4C has been built with only the LLVM front-end.

I am trying to run the following command against [fibonacci.cl](https://github.com/doe300/VC4C/blob/master/example/fibonacci.cl) from the `examples` folder:

```
VC4C --llvm --hex -o fibonacci.hex fibonacci.cl
```

I get the following message:

```
Compiling 'fibonacci.cl' into 'fibonacci.hex' with options '' ...
[E] Mon Nov 27 14:27:43 2017: Errors in precompilation:
[E] Mon Nov 27 14:27:43 2017: Stack dump:
0.	Program arguments: /usr/lib/llvm-3.9/bin/clang -cc1 -triple armv6-unknown-linux-gnueabihf -emit-llvm -disable-free -disable-llvm-verifier -discard-value-names -main-file-name fibonacci.cl -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu arm1136jf-s -target-feature +strict-align -target-abi aapcs-linux -mfloat-abi hard -dwarf-column-info -debugger-tuning=gdb -coverage-file /tmp/vc4c-sbCfYb -resource-dir /usr/lib/llvm-3.9/bin/../lib/clang/3.9.1 -include-pch /home/pi/VC4C/include/VC4CLStdLib.h.pch -I . -internal-isystem /usr/local/include -internal-isystem /usr/lib/llvm-3.9/bin/../lib/clang/3.9.1/include -internal-externc-isystem /usr/include/arm-linux-gnueabihf -internal-externc-isystem /include -internal-externc-isystem /usr/include -O3 -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -fdebug-compilation-dir /home/pi/VC4CL-test -ferror-limit 19 -fmessage-length 0 -fallow-half-arguments-and-returns -fno-signed-char -fobjc-runtime=gcc -fdiagnostics-show-option -vectorize-loops -vectorize-slp -o /tmp/vc4c-sbCfYb -x cl fibonacci.cl 
clang: error: unable to execute command: Segmentation fault
clang: error: clang frontend command failed due to signal (use -v to see invocation)
clang version 3.9.1-9+rpi1 (tags/RELEASE_391/rc2)
Target: armv6-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin
```

I have tried to run the given `/usr/lib/llvm-3.9/bin/clang` command line and got a segmentation fault as well.
The command succeeds if I remove the argument: `-include-pch /home/pi/VC4C/include/VC4CLStdLib.h.pch`:

```
/usr/lib/llvm-3.9/bin/clang -cc1 -triple armv6-unknown-linux-gnueabihf -emit-llvm -disable-free -disable-llvm-verifier -discard-value-names -main-file-name fibonacci.cl -mrelocation-model static -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu arm1136jf-s -target-feature +strict-align -target-abi aapcs-linux -mfloat-abi hard -dwarf-column-info -debugger-tuning=gdb -coverage-file /tmp/vc4c-kK1RbC -resource-dir /usr/lib/llvm-3.9/bin/../lib/clang/3.9.1 -I . -internal-isystem /usr/local/include -internal-isystem /usr/lib/llvm-3.9/bin/../lib/clang/3.9.1/include -internal-externc-isystem /usr/include/arm-linux-gnueabihf -internal-externc-isystem /include -internal-externc-isystem /usr/include -O3 -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -fdebug-compilation-dir /home/pi/VC4CL-test -ferror-limit 19 -fmessage-length 0 -fallow-half-arguments-and-returns -fno-signed-char -fobjc-runtime=gcc -fdiagnostics-show-option -vectorize-loops -vectorize-slp -o /tmp/vc4c-kK1RbC -x cl fibonacci.cl
```

Here is an extract of `strace` run with the faulty command:

```
stat64("/home/pi/VC4C/include", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
open("/home/pi/VC4C/include/VC4CLStdLib.h.pch", O_RDONLY) = 5
readlink("/proc/self/fd/5", "/home/pi/VC4C/include/VC4CLStdLi"..., 4096) = 39
fstat64(5, {st_mode=S_IFREG|0644, st_size=12789540, ...}) = 0
mmap2(NULL, 12789540, PROT_READ, MAP_PRIVATE, 5, 0) = 0x737a9000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x157f000} ---
```

doe300, 2017-11-27:
With what version of LLVM/CLang have you built the PCH-file, has your clang been updated after you built the PCH-file? Have you tried re-building the PCH-file with the version of CLang in use? To do so, you need to delete the file and re-build the `vc4cl-stdlib` target with the correct CLang configured.

Where do the long list of additional command line options for the clang invocation come from?Did you manually specify them?

Is (your) Raspberry Pi 3 running 32 or 64 bit linux?

senshu, 2017-11-27:
> With what version of LLVM/CLang have you built the PCH-file

Only clang 3.9 is installed.
I have not updated anything after building the PCH file.

After rebuilding the PCH file, I get the same error.

> Where do the long list of additional command line options for the clang invocation come from?Did you manually specify them?

I just copied the command line that was displayed in the error message after running VC4C. I don't know where the options come from.

> Is (your) Raspberry Pi 3 running 32 or 64 bit linux?

32-bit. I used an official Raspbian Stretch Lite image, upgraded it, and installed only the required packages to build VC4C.
BTW I had to manually create a symlink `/usr/bin/clang` to `/usr/bin/clang-3.9` to make it work.

doe300, 2017-11-27:
The very long list of command line arguments confuse me, especially, because I have never seen most of them. I suspect one of them (or a combination of those flags) causes the crash, maybe some issue with the PCH-file being compiled without them and used in a compilation where those flags are set. But in that case, usually clang outputs an error and exits properly.

>  BTW I had to manually create a symlink /usr/bin/clang to /usr/bin/clang-3.9 to make it work.

That one I can fix. Though I wonder, did cmake not complain about not finding clang?

senshu, 2017-11-28:
> That one I can fix. Though I wonder, did cmake not complain about not finding clang?

It did before I created the symlink.

senshu, 2017-11-28:
Here is the process that I followed to build and install VC4C:
https://gist.github.com/senshu/671ecb1e68729c5e1a897c62251e00cf

Reproduced the issue this morning, starting from scratch with a brand new [Raspbian Stretch Lite](https://www.raspberrypi.org/downloads/raspbian/) SD card on a Raspberry Pi 3.

doe300, 2017-11-28:
Though I cannot reproduce the SEGFAULT, I also get errors trying to compile with standard clang. Seems like the command to compile the PCH is not compatible with the command to compile the actual kernels.

doe300, 2017-11-29:
The pre-compilation error with "standard" CLang should now be fixed. But there is a larger underlying problem, see #7. Basically, using the "standard" CLang will result in much slower compilation times.

doe300, 2018-02-22:
@senshu, is this still an issue? Can you re-check or close this ticket?

senshu, 2018-02-23:
Sorry. I have moved to another project and I will not be able to check.
I think it is better to close the issue since you could not reproduce it.

closed 2018-02-23 by senshu