doe300/VC4C #9: Rewrite building of standard-library PCH

doe300, 2017-11-30:
Especially now, that cross-compilation and debian packaging is in the making (see #8), the way the VC4CLStdLib precompiled header is being built needs to be changed.

Currently, the PCH-file is being built on the development machine:
- a PCH-file only works with the exact version of the CLang used to compile it. For cross-compilation or packaging, this won't match the version of the destination machine(s).
- the PCH-file requires the original source headers to be available and their (relative?) path not to change, also not given for cross-compiled or packaged distribution.

The solution would be to build the PCH-file on the destination machine:
- requires the VC4CLStdLib header-files to be available there (e.g. installed via extra `vc4cl-stdlib` package).
- the PCH-file would need to be built on first compilation using VC4C.
- this also fixes the issues arising, that the compilation options for the PCH-file don't match the options for the actual compilations (see #6 ).

=> On every build (or in a post-installation script), if the PCH-file doesn't exist, create it first (using the same options as the actual compilations).
- [ ] how to get path to the VC4CLStdLib headers? Need to distinguish between in-source-builds (e.g. on development machines, the headers lie in a project somewhere) and packaged build (e.g. on destination machines, the headers are installed into e.g. `/usr/include/vc4cl-stdlib` via extra package)
- [ ] where to store the PCH-file to and how to save the path to the file? Similar to first point, for development build, we can simply store it on `./include/` or `./build`. For packaged builds, it should reside somewhere in `/usr/share/vc4c` or a similar path

nomaddo, 2017-11-30:
Must we compile VC4CLStdLib to PCH files?
If not, treating it as plain-text is easier.
If you worry about performance (runtime overhead of compiling), it is not the problem, i think.
Developers (who care that) can compile by VC4C and use `clBuildProgramWithBInary`.

doe300, 2017-11-30:
Technically, it is not necessary, but otherwise the compilation is very slow.
If packaging/cross-compilation in combination with PCH is too complicated, we could do without a PCH header.

nomaddo, 2017-11-30:
So how about compile it when extracting debian package?
`CPACK_INSTALL_COMMANDS` can specify script files (which execute when debian package is extracted).
We add VC4CLStdLIb to deb package, and compile them during extraction, and locate it to `${prefix}/bin/../lib/vc4c`, for example.

nomaddo, 2017-11-30:
When we cross-compile, do like above.
When we compile normally (compile it in host raspberry pi), do the same thing when `make install`, or unpacking deb package.
It works, i think.

doe300, 2017-11-30:
Compiling the standard-library into PCH with installation of the package could work. We need to make sure though, not to break the in-source development build and the `make install` work-flow. 

So I suggest following steps:
- [ ] add extra package `vc4cl-stdlib` (in project VC4CLStdLib) which - completely equal to the not yet existing `make install` command in VC4CLStdLib - only installs the header-files into e.g. `/usr/local/include/vc4cl-stdlib`.
- [ ] in CMakeLists.txt for VC4C, rewrite `VC4CL_STDLIB_HEADER` to always reside in the same folder as `VC4CL_STDLIB_HEADER_SOURCE`.
- [ ] in the CircleCI build for VC4C, `make install` VC4CLStdLib and set `VC4CL_STDLIB_HEADER_SOURCE` to the header in `/usr/local/include/vc4cl-stdlib`. This (1) verifies the correctness of the VC4CLStdLib headers by running the target `vc4cl-stdlib` on the CircleCI environment as before and (2) makes sure the PCH-file is later (on the destination machine) expected to reside in `/usr/local/include/vc4cl-stdlib`.
- [ ] on installation of the VC4C package, build the PCH-file (within `/usr/local/include/vc4cl-stdlib`), as you suggested.

This would result in following behavior:
- for in-source development builds, the PCH-file would reside in the source of the VC4CLStdLib project instead of VC4C. This change is acceptable, I think.
- for `make install`ing VC4C, the `VC4CL_STDLIB_HEADER_SOURCE` variable would need to be set to the location VC4CLStdLib was installed into. After that, the installed VC4C should find the PCH-file correctly.

closed 2017-12-07 by doe300