doe300/VC4C #12: Automatically add artifacts to releases

doe300, 2017-12-04:
The next step after #8 and fixing #9 would be to create a github-release `nightly`, were after every successful build on CircleCI (or at least once per day), the resulting artifacts (library, executable and debian-package) are released.

A more light-weight version of this could also be implemented by (simply?) linking to the latest versions of the artifacts in the CircleCI builds. In this case, the artifacts would need a stable URL across builds.

doe300, 2017-12-04:
@nomaddo any ideas for this?

nomaddo, 2017-12-04:
Do you have any plan to use github-release?
I think we don't need to use it because circleci log provide all artifacts.
Releases should be stable, but current status of the project is not, I think.

If you need to fixed URL to download latest artifacts, you can use the API.
https://circleci.com/docs/api/v1-reference/#build-artifacts-latest
I think it's enough, don't you think?

Let's set milestones to make tags and then, add artifacts as github-release