doe300/VC4C #15: Speed-up memory access

doe300, 2017-12-11:
Memory access can be sped up in several ways:

- utilize delays for DMA access for other calculations
- using VPM as cache and access DMA for blocks of memory at a time
- use TMU to read from memory?

See also #14 and [this post](https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=169576#p1095422).

doe300, 2017-12-11:
### Utilizing delays for DMA-access

Any DMA read/write introduces a delay of at least 6/10 instructions between setting the address to access and waiting for the DMA operation to finish. This delay can be used for other calculations.

#### Steps
1. Insert specific number of `nop` instructions between setting RAM address and waiting for DMA to finish
1.1 **TODO**: also insert `nop`s between setting VPM address and first load from VPM (has a 5 instructions delay); update gain
2. Allow `optimization/Reordering.cpp` to replace the newly inserted `nop`s with meaningful instructions
3. **TODO**: remove remaining `nop`s again, since the wait instruction already waits the correct number of cycles to save instructions; update gain

#### Gain
- clpeak global-bandwidth test-case has been sped-up by 4.5%
- TODO: test for clpeak floating-point benchmark

doe300, 2017-12-12:
### Reading via TMU

Reading from TMU is much simpler (set address, trigger signal, read value) than reading through VPM and does not need for a mutex-lock, allowing all QPUs to read memory parallel.

#### Work-flow

1. set `TMU0_S` to address to read (per-element)
2. trigger read via signal
3. read value from `r4`

#### Concept

* Reading of vectors: Can use `elem_num` register to automatically add the offsets for the other elements. Set unused elements to zero.
* Reading of non 32-bit types: Need to bit shift/pack (and rotate) values to correct type-size and vector-element.
* Since general memory lookup via TMU is aligned to 32-bit ("the bottom two bits of the address are ignored"), need to add an initial offset (for non 32-bit aligned addresses)
* Similar to reading through DMA, the stall between signaling the read and reading `r4` (9 to 20 instructions, see [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1143940#p1144081)) can be filled with other instructions

#### Unsolved issues

* Does reading from TMU auto-increment address (similar to reading uniforms) or do we need to re-set the address every time?
* How many TMU reads can be queued at once? [This page](http://maazl.de/project/vc4asm/doc/VideoCoreIV-addendum.html) states 4 queries.

@nomaddo can you provide any information on them?

nomaddo, 2017-12-13:
> Does reading from TMU auto-increment address (similar to reading uniforms) or do we need to re-set the address every time?
Yes it is. We need to increment `TMU0_S` or `TMU1_S` every time. 

> How many TMU reads can be queued at once? This page states 4 queries.
That's true. More precisely, TMU can hold up to 4 queries from each QPU. 

One more thing to be considered: how to use both `TMU0` and `TMU1`. You know, bandwidth of vc4 is so weak. We need to use all of machine features to perform well.
Smart instruction scheduler is required, in my imagination.

nomaddo, 2017-12-13:
In my experience， TMUs require at least 9 cycles per one load with cache hit. 
If the data is stored in 2nd cache, plus 20 cycles. 
More cycles are required in wose case.

doe300, 2017-12-13:
>That's true. More precisely, TMU can hold up to 4 queries from each QPU.

> One more thing to be considered: how to use both `TMU0` and `TMU1`.

Does that mean, we can queue up to 4 queries to each of the TMUs from every QPU? If I understand correctly, for this to work, we need to disable TMU swapping, do we?

nomaddo, 2017-12-14:
My understanding is the same. 

And, I and my colleague haven't seen the case TMU swapping is useful.
We always disable it.

doe300, 2017-12-14:
You were definitively right about the increased performance when using TMU to read from memory.

With simple reads via TMU, I managed to achieve following performance gain:

- clpeak global-bandwidth test-case has been sped-up by **528%**!
- clpeak floating-point benchmark has been sped-up by 0.8%

Things left to do:

- [ ] allow instructions to be inserted between `load_tmu` signal and reading of `r4` (similar to VPM optimizations above). Could insert up to 9 (20?) instructions, MUST NOT re-order TMU loads!
- [ ] allow up to 4(8) TMU loads to be queued at once, by allowing setting of TMU addresses and triggering of `load_tmu` signals to be re-ordered before reading `r4`.
- [ ] disable TMU swapping and manually schedule TMU loads onto the available TMUs

nomaddo, 2017-12-14:
Glad to hear that!

> allow instructions to be inserted between load_tmu signal and reading of r4 (similar to VPM optimizations above). Could insert up to 9 (20?) instructions, MUST NOT re-order TMU loads!
> allow up to [...]

This should be archived by instruction scheduler, in general compiler design.
In my view, we need

- Creatinng loops by worker thread fusion
- Unroll loops
- Instruction reordering by Instruction scheduler

doe300, 2017-12-15:
We can only store 64 vectors at once, if we write them into consecutive memory. This optimization is already done for a single basic-block of a single QPU (e.g. if there are x writes within that basic-block to consecutive memory, the x values are written into VPM and written-back only once into RAM).

The problems of the global optimization (collection values from all QPUs into VPM and writing them back once) lie in (1) determining, whether the different QPUs write to consecutive memory and (2) making sure, no writes in other areas of memory occur in the mean time.

doe300, 2018-01-02:
Open points should be added by #26

doe300, 2018-04-17:
Just some note here:
If the same memory is possible written before it is read, we need to read it via VPM. Otherwise the cache will screw things up (return the previous/old value for reads via TMU).

It would be much more efficient to solve this problem another way (e.g. flush/skip cache), but at this moment this is the only solution I know of.

See also #30 and 5373ab631adef4e83a5233bdd3183df3e3b9da17.

nomaddo, 2018-04-17:
As far as I know, there is no way to flush the cache from VC4 except that it reads values as much as the volume of the L2 cache.

Terminus-IMRC, 2018-04-17:
...Or you can synchronize with CPU or VPU (by `host_int` or `V3D_SRQCS`) and have them to write to `V3D_L2CACTL` and `V3D_SLCACTL`.

lygstate, 2018-06-24:
May us add clpeak in CI?

doe300, 2018-06-24:
> May us add clpeak in CI?

What do you mean?

lygstate, 2018-06-24:
@doe300 I found why clpeak doesnt works:( VC4c does support half & double float.

lygstate, 2018-06-24:
@doe300  according to https://github.com/doe300/VC4C/blob/master/testing/clpeak/all_kernels.cl,
VC4C disabled the code of half & double float support. May us use the feature flag disable it:)
I want to contribute about this, where should I start? from VC4C or the other two lib?

doe300, 2018-06-24:
> VC4c does support half & double float.

VC4C does NOT support half or double types.

> I want to contribute about this, where should I start?

What exactly do you want to do? Support half and/or double? This would be a feature of the VC4C compiler and the VC4CL standard-library.

lygstate, 2018-06-24:
@doe300 I am not meaning support for half or double floating point, I am talking about respect the predefined macros:), such as 
#if defined(cl_khr_fp16)
By the way, when I am running clpeak , how much time it would consume, I am waiting for a long time, but it's still not finished:( By disable (half/double float) testing files.

doe300, 2018-06-24:
Ah, okay.

> how much time it would consume

That depends. I haven't run clpeak for a long time, so it could currently be broken, in which case it takes a while before failing. If it is not broken, it should take a few (<5) minutes.

lygstate, 2018-06-24:
@doe300 Now I am finished the tests running, the result is not so good:)
```
Platform: OpenCL for the Raspberry Pi VideoCore IV GPU
  Device: VideoCore IV GPU
    Driver version  : 0.4 (Linux ARM)
    Compute units   : 1
    Clock frequency : 300 MHz

    Global memory bandwidth (GBPS)
clCreateBuffer (-5)
      Tests skipped

    Single-precision compute (GFLOPS)
      float   : clEnqueueNDRangeKernel (-54)
      Tests skipped

    Integer compute (GIOPS)
      int   : clEnqueueNDRangeKernel (-54)
      Tests skipped

    Transfer bandwidth (GBPS)
      enqueueWriteBuffer         : 1.36
      enqueueReadBuffer          : 0.22
      enqueueMapBuffer(for read) : 1061.85
        memcpy from mapped ptr   : 0.22
      enqueueUnmap(after write)  : 1061.85
        memcpy to mapped ptr     : 1.02

    Kernel launch latency : 14.63 us
```

doe300, 2018-06-24:
The integer test never worked (idk why), the others all did at some point. I will need to have a look at why they do not work anymore.

lygstate, 2018-06-24:
@doe300 Did I need to increase the GPU memory size? I am using the default setting, Now I am increasing the GPU memory size to 256MB and re-running clpeak.

doe300, 2018-06-24:
> Did I need to increase the GPU memory size?

I would recommend it, but it shouldn't be necessary. I just remembered, "vanilla" clpeak has a problem with the work-group size of VC4C (see https://github.com/krrishnarraj/clpeak/issues/41)

lygstate, 2018-06-24:
@doe300 Thanks, I am goding to try your fixed version:)

nomaddo, 2018-06-24:
@doe300  Actually, VC4 support fp16 only for conversions (fp16 -> fp32 and fp32 -> fp16). 
It is useful to reduce the size of data which is transported between GPU and CPU because bottleneck of VC4 program is often transportation.
It might be interesting to disscuss how to support it......

lygstate, 2018-06-24:
Updated clpeak test result:
```
pi@raspberrypi:~/clpeak-vc4/build $ sudo ./clpeak 

Platform: OpenCL for the Raspberry Pi VideoCore IV GPU
  Device: VideoCore IV GPU
    Driver version  : 0.4 (Linux ARM)
    Compute units   : 1
    Clock frequency : 300 MHz

    Global memory bandwidth (GBPS)
clCreateBuffer (-5)
      Tests skipped

    Single-precision compute (GFLOPS)
      float   : 0.28
      float2  : 0.78
      float4  : 1.72
      float8  : 2.84
      float16 : 4.56

    Integer compute (GIOPS)
      int   : 0.02
      int2  : 0.04
      int4  : 0.08
      int8  : 0.15
      int16 : 0.30

    Transfer bandwidth (GBPS)
      enqueueWriteBuffer         : 1.51
      enqueueReadBuffer          : 0.22
      enqueueMapBuffer(for read) : 1245.18
        memcpy from mapped ptr   : 0.22
      enqueueUnmap(after write)  : 1281.22
        memcpy to mapped ptr     : 1.03

    Kernel launch latency : clGetEventProfileInfo (-7)
      Tests skipped

terminate called without an active exception
Aborted
```

nomaddo, 2018-09-15:
The answer is No. 1: Writing to `tmu0_s` trigger the TMU load.
See my experiment: https://gist.github.com/nomaddo/31aaf839f2463816beb9563173b48728

To my understanding, vc4 try to do every memory operation asynchronously.
To do that, the operations must be split them into `trigger` and `sync`.
The trigger is, `mov (tmu0_s, addr)`, and `nop.load_tmu0` is sync operation.

doe300, 2018-09-16:
@nomaddo thanks for your tests. Although I'm not a fan of python, py-videocore seems to be a nifty tool to play around and test these things.

Just for info, I also found out that a delay of up to 8 instructions inserted between writing address and triggering signal has no effect on execution time (very, very little, probably due to instruction fetching), while 9+ instructions have a larger impact. This corresponds with the general consensus, that fetching from cache takes 9 instructions.
Also, if more than 8 requests are queued per TMU (and QPU), the execution hangs. Up to 8 requests per TMU and QPU seem to work fine.

doe300, 2018-09-20:
As far as I can see, all points in here have been implemented:
- The TMU delays will be minded by the instruction scheduler
- both TMUs are used
- the instruction scheduler also is able to queue up several requests per TMU before reading any result

closed 2018-09-20 by doe300