doe300/VC4C #17: Remove common expression

nomaddo, 2017-12-13:
```cl
kernel void sum_f16 (global float a[], global float b[], global float c[]) {
  int id = get_global_id(0);
  float16 x = vload16(id, b);
  float16 y = vload16(id, c);
  vstore16 (x * y + x * y + x * y, id, a);
}
```

https://gist.github.com/nomaddo/9a7921afd53d08cb34d736e7432b8036

In this code, it seems that the same calculation is performed repeatedly (line 59 and 60).
I imaged this such typical optimization (removing subcommon expression) is done by `clang`.

nomaddo, 2017-12-13:
Output of `clang` include `fmuladd`. So sub-common expression elimination doesn't work.
With `-ffp-contract=off`, this optimization seem to be peformed.

nomaddo, 2017-12-13:
One more thing: in line 59 ~ 63, dual-issue can be performed, but doesn't work.
In my imagination, we need instruction scheduling including such optimization?

doe300, 2017-12-13:
> With `-ffp-contract=off`, this optimization seem to be peformed.

Since the VC4 doesn't support `fmuladd`, we can simply disable fused floating-point operations.

nomaddo, 2017-12-14:
This problem seems to be solved.

closed 2017-12-14 by nomaddo