doe300/VC4C #20: Build failed?

nomaddo, 2017-12-20:
I am not sure what happened, circleci build failed.
In my local env, it is successful (also in docker image `nomaddo/cross-rpi`.

Do you have any idea?

nomaddo, 2017-12-20:
The problem is, the cross-compiler `arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64` is older than the one we use in host environment?

doe300, 2017-12-20:
Yes, I am running either CLang 4.0 or gcc 7.2, which both compile the code successfully.

I am currently trying to fix the compilation for the gcc 4.9 on my Raspberry Pi and will push it if I have done so.

nomaddo, 2017-12-20:
Or, how about change the cross-compiler?
Linaro provide gcc 7.2 as x86 to armv7 cross-compiler. 
https://releases.linaro.org/components/toolchain/binaries/latest/arm-linux-gnueabihf/

If you prefer, assign me to this issue. I will change CMakeList.txt and Dockerfile.

doe300, 2017-12-20:
If we change the cross-compiler, this will indeed fix the issue of failing builds on CircleCI, but it will not fix the failing builds on (older) Raspberry Pi machines.

We could drop support for older gcc, since Raspbian Strech apparently ships with gcc 6.3 (see #13) which should support more C++11 (and C++14) features. But for the moment I will try to make it run on modern and older compilers.

closed 2017-12-20 by doe300

Terminus-IMRC, 2017-12-21:
@nomaddo How about using https://hub.docker.com/r/idein/cross-rpi/ for cross-compiling?  Your cross-rpi image is based on raspberrypi/tools, which is not upgraded for stretch because of [some problems](https://github.com/raspberrypi/tools/issues/81).  Our image involves gcc 6.3 for `armv6-rpi-linux-gnueabihf`, `armv7-rpi2-linux-gnueabihf` and `armv8-rpi3-linux-gnueabihf`.

nomaddo, 2017-12-21:
@Terminus-IMRC Thanks your suggestion. Currently, we support older Rasberry Pi machines (which use jessie package) as @doe300 said. When we drop them, I will use `idein/cross-rpi`!