doe300/VC4C #21: Remove constant load in loops

nomaddo, 2017-12-21:
`ldi` remain in for-loop even though it can be moved to outside of the loop.

```
kernel void loop1 (global float a[], global float b[]) {
  for (int i = 0; i < 1000; i++) {
    a[i] = -i;
    b[i] = -i;
  }
}
```

```
// Module with 1 kernels, global data with 0 words (64-bit each), starting at offset 1 words and 0 words of stack-frame
// Kernel 'loop1' with 67 instructions, offset 2, with following parameters: __global out float* a (4 B, 1 items), __global out float* b (4 B, 1 items)
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or ra0, unif, unif
or r2, unif, unif
or r1, 0 (0), 0 (0)
nop.never 
or r0, r1, r1
sub r0, 0 (0), r0
itof r3, r0; v8min r0, r1, r1
shl r0, r0, 2 (2)
add r0, ra0, r0
or -, mutex_acq, mutex_acq
ldi vpw_setup, 6656
or vpm, r3, r3
ldi vpw_setup, 2155954176
ldi vpw_setup, 3221291008
or vpw_addr, r0, r0
or r0, r1, r1
or r1, r1, r1
shl r0, r0, 2 (2)
add r1, r1, 1 (1)
add r0, r2, r0
ldi ra1, 1000                                   //// constant load every time !!!
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
or -, mutex_acq, mutex_acq
ldi vpw_setup, 6656
or vpm, r3, r3
ldi vpw_setup, 2155954176
ldi vpw_setup, 3221291008
or vpw_addr, r0, r0
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
xor.setf -, r1, ra1
xor.ifzc r0, 1 (1), 1 (1); v8min.ifz r0, 1 (1), 1 (1)
or.setf -, r0, r0
or.ifz r1, r1, r1
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + 4
nop.never 
nop.never 
nop.never 
brr.ifanyz (pc+4) + -40
nop.never 
nop.never 
nop.never 
or r0, unif, unif
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -63
nop.never 
nop.never 
nop.never 
not irq, qpu_num
nop.thrend.never 
nop.never 
nop.never 
```

doe300, 2017-12-21:
The reason the constant is loaded every time is that the optimization pass to combine successive literal loads with the same value (`CombineLiteralLoads`) currently only works within a single basic block.

That said, we might profit from running this and some other optimizations in a global scope.
This should be done under the consideration, that any local with a live-time exceeding a basic block complicates the register-allocation and might lead to failing to allocate registers completely (in more complex kernel code).

nomaddo, 2017-12-21:
Thanks. Actually, it is required not in right now.
But in my experience, the impact of deleting one single instruction in inner most loops is bigger than expected.

nomaddo, 2018-03-05:
Constans like `mul24 r1, 4, elem_num` should also moved.

doe300, 2018-03-09:
Some info: https://en.wikipedia.org/wiki/Loop-invariant_code_motion

PR #56 (2018-03-21): Remove constant load in loops
Move constant load (`LoadImmediate`) operations in (nested) loop to the block before head block of the outer-most loop.

This PR resolves #21, but moves only constant load, not like `mul24 r1, 4, elem_num`.
closed 2018-03-21 by doe300