doe300/VC4C #28: Remove move instruction in mutex

nomaddo, 2018-01-10:
In the following code, `fadd r1, r2, r0` can be fused to move instruction that is located in mutex block.

```
kernel void sum_f16 (global float a[], global float b[], global float c[]) {
  int id = get_global_id(0);
  float16 x = vload16(id, b);
  float16 y = vload16(id, c);
  vstore16 (x + y, id, a);
}
```

```
or -, unif, unif
or r2, unif, unif
or ra5, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or r0, unif, unif
or -, unif, unif
or -, unif, unif
or ra4, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or ra3, unif, unif
or ra2, unif, unif
or ra1, unif, unif
or ra0, r0, r0; v8min r1, 0 (0), 0 (0)
shr r2, r2, r1
ldi r0, 255
and r3, r2, r0
or r2, ra4, ra4
shr r1, ra5, r1
and r1, r1, r0
mul24 r0, ra0, r3
add r0, r2, r0
add r0, r0, r1
shl r3, r0, 4 (4)
mul24 r2, 4 (4), elem_num
or r0, r3, r3
shl r0, r0, 2 (2)
add r0, ra2, r0
or r1, r0, r0; v8adds r0, 8 (8), 8 (8)
sub.setf -, elem_num, r0
or.ifnc r0, 0 (0), 0 (0)
add.ifn r0, r1, r2
or tmu0s, r0, r0
nop.load_tmu0.never 
or r2, r4, r4
or r0, r3, r3
shl r0, r0, 2 (2)
add r0, ra1, r0
or ra0, r0, r0
mul24 r1, 4 (4), elem_num
v8adds r0, 8 (8), 8 (8)
sub.setf -, elem_num, r0
or.ifnc r0, 0 (0), 0 (0)
add.ifn r0, ra0, r1
or tmu0s, r0, r0
nop.load_tmu0.never 
or r0, r4, r4
fadd r1, r2, r0                             !!! can be fused
or r0, r3, r3
shl r0, r0, 2 (2)
add r0, ra3, r0
or r0, r0, r0
or -, mutex_acq, mutex_acq
ldi vpw_setup, 6656
or vpm, r1, r1
ldi vpw_setup, 2156937216
ldi vpw_setup, 3221291008
or vpw_addr, r0, r0                     !!! can be fused
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
or r0, unif, unif
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -69
nop.never 
nop.never 
nop.never 
not irq, qpu_num
nop.thrend.never 
nop.never 
nop.never 
```

doe300, 2018-01-11:
Do you mean those two instruction can be fused to e.g. `fadd vpm, r2, r0`?

If so, then this would suggest to add a pass which eliminates all `move` instructions, where the source is only written once and replace the write of the source to directly write into the destination of the move.

nomaddo, 2018-01-12:
> Do you mean those two instruction can be fused to e.g. fadd vpm, r2, r0?

Yes, I do．

> If so, then this would suggest to add a pass which eliminates all move instructions, where the source is only written once and replace the write of the source to directly write into the destination of the move.

My opinion is the same. Come to think of it, it's hard to find that a variable is referred only once because it require dataflow information. This issue might be hard to solve

doe300, 2018-01-12:
> Come to think of it, it's hard to find that a variable is referred only once because it require dataflow information.

Actually, it is not. Every `Local` tracks its usages (see `Local.h`), so it is quite easy to find locals which are written and read only once.
The problem lies more in making sure, that we don't introduce any conflicts for the register-mapping by prolonging the life-times of the parameters (in the example `r2` and `r0`).

commit 92f5b8bb (2018-02-05): Adds optimization to remove redundant moves, closes #28
closed 2018-02-05 by doe300