doe300/VC4C #29: Move CircleCI to LLVM-library front-end

doe300, 2018-01-11:
Using the LLVM-library front-end for debian packages is currently the easiest and best way (for the user) to use the VC4C compiler:
- the dependency on CLang/LLVM library installed can be expressed in the debian-package
- the LLVM library front-end has the same test-sample acceptance rate as the SPIR-V front-end
- the LLVM library front-end is the fastest front-end

Current problems:
- CircleCI seems to provide `llvm-config-3.9` (should be provided by `llvm-3.9`)
- "Default" `llvm-config` seems not to find the LLVM library, headers, include flags, etc. (see CircleCI log section `configure`)
- But for SPIRV-LLVM, `llvm-config` finds all required components

*Do we have to install/provide `llvm-dev` cross-compiled for Raspberry Pi?*

@nomaddo, any ideas on this?

nomaddo, 2018-01-17:
Sorry for the late reply.
I am not sure what is the point of the issue. 
The problem is, `llvm-config` (installed via `apt`) doesn't return correct, the one of `SPIRV-LLVM` does?

```
pi@nomaddo:~$: /opt/SPIRV-LLVM/build/bin/llvm-config --libs
-lLLVMLTO -lLLVMObjCARCOpts -lLLVMLinker -lLLVMBitWriter -lLLVMIRReader -lLLVMAsmParser -lLLVMSPIRVLib -lLLVMipo -lLLVMVectorize -lLLVMTableGen -lLLVMDebugInfo -lLLVMOption -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMMCDisassembler -lLLVMARMInfo -lLLVMARMAsmPrinter -lLLVMMCJIT -lLLVMLineEditor -lLLVMInstrumentation -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCodeGen -lLLVMScalarOpts -lLLVMProfileData -lLLVMObject -lLLVMMCParser -lLLVMBitReader -lLLVMInstCombine -lLLVMTransformUtils -lLLVMipa -lLLVMAnalysis -lLLVMTarget -lLLVMMC -lLLVMCore -lLLVMSupport
pi@nomaddo:~$ llvm-config-3.9 --libs
-lLLVM-3.9
```

doe300, 2018-01-17:
Yes exactly, the `/usr/bin/llvm-config` does not return the library paths and flags for the LLVM library, while `/opt/SPIRV-LLVM/build/bin/llvm-config` does.

Since we installed `llvm-dev` after the initial build of the docker-image, could it be, that we need to re-build the docker image for the CircleCI builds?

nomaddo, 2018-01-17:
Mmm, I got confused about what package do we need.
`llvm-config` in `llvm-dev` return flags correctly, but `llvm-config-3.9` of `llvm-3.9-dev` return just `-lLLVM-3.9`.
`llvm-dev` require `llvm-3.8`, so installing this package may be incorrect?

Anyway, if we need to re-build docker images is automatically done if `Dockerfile` in github has changed.
The case of accepting your pullreq 6 day ago, the build seemed to fail by unexpected error.
https://hub.docker.com/r/nomaddo/cross-rpi/builds/bpqormb5qr5znftyv6qlxl6/

doe300, 2018-01-17:
I made an error: In the docker-image (and the `vc4c` package dependency), we use`llvm-3.9-dev`, so we don't care about which version of LLVM `llvm-dev` refers to.

>  The case of accepting your pullreq 6 day ago, the build seemed to fail by unexpected error.

So if the re-build of the docker-image failed, that could explain why the `llvm-config` command failed.

nomaddo, 2018-01-17:
I am not sure, but the reason why docker build failed, seems timeout (over 2h for building).
I turn off the auto-build and push latest image by hand.

And, it add `llvm` to the image (see [here](https://github.com/nomaddo/cross-rpi/commit/cffe70f2b0c788049d90bbbdc8d0b6be816e880c)).
Now `llvm-config` is available in the docker image environment.

```
[~/work/cross-rpi] docker run -it nomaddo/cross-rpi
root@c7de9b672e84:/# llvm-config --libs
llvm-config --libs
-lLLVMLTO -lLLVMObjCARCOpts -lLLVMSymbolize -lLLVMDebugInfoPDB -lLLVMDebugInfoDWARF -lLLVMXCoreDisassembler -lLLVMXCoreCodeGen -lLLVMXCoreDesc -lLLVMXCoreInfo -lLLVMXCoreAsmPrinter -lLLVMSystemZDisassembler -lLLVMSystemZCodeGen -lLLVMSystemZAsmParser -lLLVMSystemZDesc -lLLVMSystemZInfo -lLLVMSystemZAsmPrinter -lLLVMSparcDisassembler -lLLVMSparcCodeGen -lLLVMSparcAsmParser -lLLVMSparcDesc -lLLVMSparcInfo -lLLVMSparcAsmPrinter -lLLVMPowerPCDisassembler -lLLVMPowerPCCodeGen -lLLVMPowerPCAsmParser -lLLVMPowerPCDesc -lLLVMPowerPCInfo -lLLVMPowerPCAsmPrinter -lLLVMNVPTXCodeGen -lLLVMNVPTXDesc -lLLVMNVPTXInfo -lLLVMNVPTXAsmPrinter -lLLVMMSP430CodeGen -lLLVMMSP430Desc -lLLVMMSP430Info -lLLVMMSP430AsmPrinter -lLLVMMipsDisassembler -lLLVMMipsCodeGen -lLLVMMipsAsmParser -lLLVMMipsDesc -lLLVMMipsInfo -lLLVMMipsAsmPrinter -lLLVMHexagonDisassembler -lLLVMHexagonCodeGen -lLLVMHexagonAsmParser -lLLVMHexagonDesc -lLLVMHexagonInfo -lLLVMCppBackendCodeGen -lLLVMCppBackendInfo -lLLVMBPFCodeGen -lLLVMBPFDesc -lLLVMBPFInfo -lLLVMBPFAsmPrinter -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMInfo -lLLVMARMAsmPrinter -lLLVMAMDGPUCodeGen -lLLVMAMDGPUAsmParser -lLLVMAMDGPUDesc -lLLVMAMDGPUUtils -lLLVMAMDGPUInfo -lLLVMAMDGPUAsmPrinter -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Info -lLLVMAArch64AsmPrinter -lLLVMAArch64Utils -lLLVMMIRParser -lLLVMLibDriver -lLLVMOption -lLLVMTableGen -lLLVMLineEditor -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMX86Desc -lLLVMMCDisassembler -lLLVMX86Info -lLLVMX86AsmPrinter -lLLVMX86Utils -lLLVMMCJIT -lLLVMPasses -lLLVMipo -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMDebugInfoCodeView -lLLVMInterpreter -lLLVMCodeGen -lLLVMScalarOpts -lLLVMInstCombine -lLLVMInstrumentation -lLLVMProfileData -lLLVMBitWriter -lLLVMOrcJIT -lLLVMTransformUtils -lLLVMExecutionEngine -lLLVMTarget -lLLVMAnalysis -lLLVMRuntimeDyld -lLLVMObject -lLLVMMCParser -lLLVMBitReader -lLLVMMC -lLLVMCore -lLLVMSupport
```

doe300, 2018-01-17:
CMake now seems to find the `llvm-config-3.9`, its settings and `clang-3.9` correctly.

I just need to make some changes to the code, since the LLVM headers are not stable across the various LLVM versions (3.6 used by SPIRV-LLVM, 3.9 shipped with Raspbian and 4.0 running on my development machines), so I reverted CircleCI back to using SPIRV-LLVM for now.

doe300, 2018-01-22:
Okay, as of [CircleCI build 52](https://circleci.com/gh/doe300/VC4C/52), the LLVM library front-end is built correctly, but linking fails, since the `libLLVM-3.9.so` on the system has the wrong architecture. So we need to provide the `libLLVM-3.9.so` from the Raspbian distribution in the cross-compilation library directory.

doe300, 2018-02-19:
Linking `libVC4CC.so` now succeeds with LLVM library front-end, but it fails linking the programs (`VC4C`, `qpu_emulator`) with undefined references on the dependencies of `libLLVM.so`.
@nomaddo, do you have any idea what the problem may be? Since I explicitely added those libraries to the cross-compilation path in https://github.com/nomaddo/cross-rpi/pull/3.

nomaddo, 2018-02-20:
The problem is, `libffi6.so.6` is missing (Now i am trying to fix it...).
And, some how `std::defaultfloat` is missing in `src/main.cpp` because of gcc version?

doe300, 2018-02-20:
The `std::defaultfloat` is an error in the old gcc not completely supporting the C++11 standard. I fixed that error locally, will be uploaded with the next push.

I think `libedit.so.2` is missing too. And I also don't know why the libraries which should be there (`libtinfo.so.5` and `libz.so.1`) are not found.

nomaddo, 2018-02-20:
I fixed the dependecies in https://github.com/nomaddo/cross-rpi/commit/7a77004b6b41744de0a96d1b008ffdc0e54ce6cd .
I think `src/CMakeLists.txt` should also be fixed to add dependencies.

doe300, 2018-02-22:
Completed with 4ba597c5402de8519ad57bbd8379b71bfcbd07fa

closed 2018-02-22 by doe300