doe300/VC4C #30: Manipulating local memories get weird results

harleyzhang, 2018-01-17:
I am trying to implement a dot product operation. However, it seems like working on local memories always get weird results. The code snippet looks like this:

__kernel void dot3(__global float* a_vec, __global float* b_vec, 
    __global float* output, __local float* l_vec) {
  size_t gid = get_global_id(0);
  size_t lid = get_local_id(0);

  l_vec[lid]  = a_vec[gid] * b_vec[gid];

  barrier(CLK_LOCAL_MEM_FENCE);
  if(lid==0) {
    l_vec[lid] = l_vec[lid] + l_vec[lid+1]; //sometimes, this line is ignored.
  }
  output[gid] = l_vec[lid];
}

The output returns different values each run. The system does not sum the first two items in each workgroup. 

Thank you.

doe300, 2018-01-17:
Can you provide a list of expected and actual outputs as well as the local work-group size you use?

harleyzhang, 2018-01-17:
During my testing
a_vec: 0, 1, 2, 3, ......
b_vec: 0.1, 0.1, 0.1, 0.1, ....

My work-group size is 10. BTW, it's getting even more messy if I use other values like 5. 

The expected output is
0.1, 0,1, 0.3, 0.4, ....0.9, 2.1, 1.1, 1,2, ....

The idea was to sum all values in l_vec. However, that does not work. I had to simplified the code by just adding the first two elements and hope to figure out how the system works 

It seems that the memory of local variables is not maintained and may be flushed by other work items. Sometimes they might be flushed by other work groups.

doe300, 2018-01-17:
I saw just now, that you pass your `__local l_vec` as kernel parameter. How do you set the the buffer used for this parameter host-side?

harleyzhang, 2018-01-17:
This does not work either. It gets probably uninitialized value. The cpp file here
`
[math_test2.txt](https://github.com/doe300/VC4C/files/1640360/math_test2.txt)
`
The CL file: 
`
[mattool2.txt](https://github.com/doe300/VC4C/files/1640364/mattool2.txt)
`
I tried figure out the logic from the output pattern. Could not get it. 

The ouput log is
`start....
create program err=0  program0x1ac626c
build program err=0
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 
0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 0.1 
vector length130
a buffer created err=0
b buffer created err=0
out buffer created err=0
create kernel returns 0
write buffer returns 0
write buffer returns 0
set kernel arg returns 0
set kernel arg returns 0
set kernel arg returns 0
Elapsed time for setting parameters: 0.00193476 s
enqueue nd range kernel returns 0
finnish returns 0
Elapsed time for calculation: 0.00511558 s
read buffer returns 0
Elapsed time after buffer read: 60.0061 s
5.27046e-26 5.27046e-26 5.27046e-26 5.27046e-26 3.57331e-43 2.44136e-31 3.03913e-26 6.40066e-26 6.40066e-26 2.87757e-26 0 0 0 0 0 0 0 0 0 0 
all done!
`

Notice there is 60 seconds delay during the execution. There was not such a delay if use the dynamic local variables.

doe300, 2018-01-17:
Ok, I will look more into it tomorrow.

> Notice there is 60 seconds delay during the execution.

The kernel execution times out after 60 seconds. This occurs if the kernel code is wrong and is blocking somewhere, e.g. by waiting to read a value it cannot or by waiting on a mutex/semaphore which is never released, etc.
So basically, at this point at least one of the QPUs the code is executed on is in a hanged state and can only be recovered by restarting it.

doe300, 2018-01-18:
Quick update:
- Using a local block currently hangs all QPUs participating in the execution (probably waiting to load some value)
- Using a local parameter now "works" if the barrier function-call is removed: it does not hang, but the result is not correct (race condition)
- Strangely, the barrier function itself also works as expected, as tested in different code

Small remark: 
In the `math_test2.txt` you provided, for the option to pass the local memory as parameter, you specify a size of 20 byte which is definitively not enough to store 10 `floats`.

harleyzhang, 2018-01-19:
Thank doe300. You are right. The size of l_vec is too small. I changed it to 40, but that is not the main issue here.

Somehow the barrier seems to work to me. The following code works. 

```
....
  barrier(CLK_LOCAL_MEM_FENCE);                                                                    

  if(lid==0) {
    l_vec[lid] = l_vec[lid] + l_vec[lid+1]; //This gives correct result
    l_vec[0] = l_vec[0] + l_vec[1]; //Strangely, this does not
  }

```

Also there are other weird behaviors, for example for loop does not work.

doe300, 2018-01-30:
So, I spent the last week completely rewriting the memory-access as well as trying to fix this issue. Here is the status so far:

- I fixed the bug which hung the barrier-function only for LLVM front-ends
- I fixed another bug resulting in incorrect calculation of local-ids
- I found that the commit 0d9ef10360ec7d68ee1b59b0f0e1514a2e12a82a introduced a bug hanging the execution in some cases. Reverting the optimization results in the output below (with `barrier()` disabled)
- Both versions still/again hang with barrier enabled

The partial output of for local parameter:

> 0.1 0.1 0.2 0.3 0.4 0.5 0.6 0.7 0.8 0.9 1.1 1.1 1.2 1.3 1.4 1.5 1.6 1.7 1.8 1.9 2.1 2.1 2.2 2.3 2.4 2.5 2.6 2.7 2.8 2.9 

The partial output for local memory:

> 1.2578488e-19 0.1 0.2 0.3 0.4 0.5 0.6 0.7 0.8 0.9 1.1 1.1 1.2 1.3 1.4 1.5 1.6 1.7 1.8 1.9 2.1 2.1 2.2 2.3 2.4 2.5 2.6 2.7 2.8 2.9

doe300, 2018-02-16:
Finally some progress:

- `barrier()` now longer hangs anything
- Using local memory now gives the (hopefully correct) result below
- Using local parameter still has issues with reading a temporary value written previously (maybe a caching issue?), see results below
- The version with local memory now uses a new optimization (see below)

The result for using local memory is:
> 0.1 0.1 0.2 0.3 0.4 0.5 0.6 0.7 0.8 0.9 2.1 1.1 1.2 1.3 1.4 1.5 1.6 1.7 1.8 1.9 4.1 2.1 2.2 2.3 2.4 2.5 2.6 2.7 2.8 2.9 6.1 3.1 3.2 3.3 3.4 3.5 3.6 3.7 3.8 3.9 8.1 4.1 4.2 4.3 4.4 4.5 4.6 4.7 4.8 4.9 [...]

The result for using local parameter is:
> 0 0.1 0.2 0.3 0.4 0.5 0.6 0.7 0.8 0.9 1 1.1 1.2 1.3 1.4 1.5 1.6 1.7 1.8 1.9 2 2.1 2.2 2.3 2.4 2.5 2.6 2.7 2.8 2.9 3 3.1 3.2 3.3 3.4 3.5 3.6 3.7 3.8 3.9 4 4.1 4.2 4.3 4.4 4.5 4.6 4.7 4.8 4.9 [...]

New optimization:
Local memory space (if small enough) is now stored in VPM. This has several advantages:

- No need to access RAM, which saves some time (10+ instruction cycles per access) waiting for the DMA operation to have finished. This also reduces the critical section blocking the other QPUs.
- No need to reserve additional RAM to store the local memory

 I will push the changes in the next few days.

doe300, 2018-03-17:
Update (nothing really new):
- With the latest code, both versions run correctly on the emulator.
- The local memory version runs correctly on the VC4 GPU

Since the code for the parameter version is correct (runs on emulator) and the starting of kernels is correct (the local memory version works), I assume the error to be somewhere in caching (the value written into the local parameter is not read, but the previous value zero).

doe300, 2018-04-16:
This issue should be fixed now.
@harleyzhang, can you re-test whether it works for you?

closed 2018-06-01 by doe300