doe300/VC4C #31: Manipulate instructions in `BasicBlock`

nomaddo, 2018-01-17:
I am trying to implement instruction scheduler.
I am wondering how to treat instructions.

`BasicBlock.instructions` has type 
`RandomModificationList<std::unique_ptr<intermediate::IntermediateInstruction>>` and `InstructionItelator` treat raw pointers of `intermediate::intermediateinstruction`.
I need to remove all instructions in `BasicBlock::instructions` for scheduling but I reuse them.
I want to change it to `shared_ptr`, and change raw pointers to `shared_ptr`.

Do you have any reason to use `unique_ptr`?
@doe300

nomaddo, 2018-01-17:
I am struggling to cache instruction: When I cache instructions (as raw pointers) and remove instructions from `BasicBlock`, data is released. The problem is come from raw pointers.

doe300, 2018-01-17:
The `InstructionWalker` is basically an enhanced iterator over the `RandomModificationList<std::unique_ptr<intermediate::IntermediateInstruction>>`.

`InstructionWalker` "contains" (as per access with `->` and `*`) raw pointers to `intermediate::intermediateinstruction` for easier access to the instruction. With these operators, the instruction stored in the `InstructionWalker` can be read and modified, but cannot be exchanged by another object.

To change the instruction-object an `InstructionWalker` points to, use the `release()` and `reset()` functions, which call the `release()` and `reset()` functions of the `unique_ptr` actually pointing to the instructions.

To insert/erase instructions, use the `erase()` and `emplace()` instructions of `InstructionWalker`.

Some examples: 

    InstructionWalker it = ...;
    //1: modifies the current instruction, though the instruction-object itself cannot be exchanged this way
    it->conditional = COND_ALWAYS;
    
    //2: changes the instruction at the current position (deletes the old one and replaces it with the new one)
    it.reset(new MoveOperation(dest, src));

    //3:inserts a new instruction BEFORE the current position, it then points to the INSERTED instruction (similar to std::list#emplace())
    it.emplace(new MoveOperation(dest, src));
    //move it to point back to the original position
    it.nextInBlock();
    //erases the instruction pointed to by it and moves to the FOLLOWING instruction(similar to std::list#ease())
    it.erase();
    //move back to the previous (emplaced) instruction
    it.previousInBlock();

The examples 2 and 3 are semantically equivalent, though 2 is much faster and easier to write.

To remove an instruction from a basic block, cache it and insert it at some other point, you can follow this procedure:

    //removes the instruction from the BB without destroying it and inserting it into the cache
    std::unique_ptr<IntermediateInstruction> cache(currentIt.release());
    
    //...
    
    //removes the instruction from the cache and inserts it into the BB
    insertIt.emplace(cache.release());
    //currentIt and insertIt are both InstructionWalkers

nomaddo, 2018-01-31:
I continue to develop an instruction scheduler, but the debug is hard because of memory corruptions...
Can you change the interface? In my opinion, it's strange to use both unique_ptr (for automate memory control) and raw pointers (introduced by dynamic_cast!!!).

Probably I miss unique_ptr::release in some cases, but original idea of smart pointers is "automatic memory management" to avoid hell of memory-corruptions. But memory manipulation is done by man-hand (`unique_ptr::release`  or dynamic_cast or so on...) in this project...

One way to change is giving up mamory management (change "unique_ptr"s to raw pointers). It's okey for compiler because it does not require efficient memory management.

Use of shared_ptr instead is possible choice. For me, shared_ptr can be treated in similar way of raw pointers. Then, we can delete dynamic_cast (from such smart pointers to raw pointers).
For methods returning `this` itself, std::enable_shared_from_this can be applied.

doe300, 2018-02-01:
> [...], it's strange to use both unique_ptr (for automate memory control) and raw pointers

From the conceptual point of view, using unique_ptr is correct, since an instruction exists only at one position of the code-flow, so this position owns the instructions.
The use of raw pointers to access the actual instruction could be rewritten to using references, but I don't think this is what you mean, is it?

> One way to change is giving up mamory management (change "unique_ptr"s to raw pointers). It's okey for compiler because it does not require efficient memory management.

This is not an option for me.

> Use of shared_ptr instead is possible choice.

Maybe, if there is no other way.

As far as I can tell, create an DAG from a basic block (in DAG constructor) and immediately remove all instructions from the basic block (in Scheduler::doScheduling). 
So if the issue is there, you could either convert all unique_ptrs to shared_ptrs in DAG constructor (e.g. by `std::shared_ptr<IntermediateInstruction>(it.release())`) and then work with the shared ptr throughout the instruction scheduler step until the instruction is re-inserted again (in InstructionSelector::choose), where you would need to release it from the shared_ptr holding it.
Or maybe you could also work with the InstructionWalker (the position containing the instruction-pointer) instead of raw pointers to the instruction itself, not empty the basic block and in InstructionSelector::choose release and erase the position before returning its contained pointer (e.g. `auto ptr = it.release(); it.erase(); return ptr;`).

closed 2018-02-19 by nomaddo