doe300/VC4C #32: Remove dependency from `/opt/SPIRV-LLVM`

nomaddo, 2018-01-17:
Currently, `vc4c-0.4.deb` depends of `/opt/SPIRV-LLVM/build/bin/clang` in preinst stage.
It is inconvenient for ordinary users because it requires compilation of `SPIRV-LLVM` by hand.

My suggestion is,

- Use `clang` as default front-end
- If some environment variable (`VC4C_SPIRV_LLVM` or so on) are set, use it as front-end

In my understand, `VC4C` can use official `clang` as front-end, but use of `SPIRV-LLVM` is better for better performance. Is it correct?

doe300, 2018-01-17:
That is actually what #29 is for. So if #29 is finished, the dependency on `SPIRV-LLVM` can be removed.

> In my understand, `VC4C` can use official `clang` as front-end, but use of `SPIRV-LLVM` is better for better performance. Is it correct?

Yes, from the supported features it makes almost no difference which clang is used. The current relative performances are as follows:

1. LLVM library front-end with SPIRV-LLVM (best)
2. SPIR-V front-end with SPIRV-LLVM (slightly inferior performance)
3. LLVM library front-end with "default" `clang` (the front-end has the same performance as with SPIRV-LLVM, but the "default" `clang` runs longer, at least of my machine)
4. LLVM IR front-end with SPIRV-LLVM
5. LLVM IR front-end with "default" clang (horribly slow!)

But especially for shipping a debian-package, where there is no automatic installation of SPIRV-LLVM, using the "default" clang would be much easier, since we can add a dependency on it.

doe300, 2018-02-22:
Also fixed with https://github.com/doe300/VC4C/commit/4ba597c5402de8519ad57bbd8379b71bfcbd07fa

closed 2018-02-22 by doe300