doe300/VC4C #34: CLANG_PATH not found during build

josiahw, 2018-02-04:
A current git pull on a box with a previously working build has produced compile errors. CLANG_PATH appears to be undefined in the Precompiler source for the current commit. I have clang-3.9 installed in /usr/bin/.

Command used for building:
`cd VC4C && git pull && git checkout . && cmake . && make clean && make`

Output:
[ 37%] Building CXX object build/CMakeFiles/VC4CC.dir/Module.cpp.o
[ 37%] Building CXX object build/CMakeFiles/VC4CC.dir/Locals.cpp.o
[ 38%] Building CXX object build/CMakeFiles/VC4CC.dir/Precompiler.cpp.o
/home/pi/VC4C/src/Precompiler.cpp: In static member function ‘static vc4c::SourceType vc4c::Precompiler::getSourceType(std::istream&)’:
/home/pi/VC4C/src/Precompiler.cpp:125:38: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     else if(std::atol(buffer.data()) == QPUASM_MAGIC_NUMBER || std::atol(buffer.data()) == QPUASM_NUMBER_MAGIC)
             ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/pi/VC4C/src/Precompiler.cpp:125:89: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     else if(std::atol(buffer.data()) == QPUASM_MAGIC_NUMBER || std::atol(buffer.data()) == QPUASM_NUMBER_MAGIC)
                                                                ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/pi/VC4C/src/Precompiler.cpp: In function ‘void compileOpenCLToLLVMIR(std::istream&, std::ostream&, const string&, bool, const vc4c::Optional<std::__cxx11::basic_string<char> >&, const vc4c::Optional<std::__cxx11::basic_string<char> >&)’:
/home/pi/VC4C/src/Precompiler.cpp:254:31: error: ‘CLANG_PATH’ was not declared in this scope
  const std::string compiler = CLANG_PATH;

doe300, 2018-02-04:
Looks like the same problem as in https://github.com/doe300/VC4CL/issues/19.

doe300, 2018-12-15:
Fixed, see https://github.com/doe300/VC4C/issues/64#issuecomment-447562027

closed 2018-12-15 by doe300