doe300/VC4C #35: PyOpenCL demo benchmark fails to compile - bit-cast vector-size error

josiahw, 2018-02-05:
I'm testing out some basic PyOpenCL examples (PyOpenCL detects the library and parameters correctly). The demo benchmark looked like a nice simple kernel to try:
[https://raw.githubusercontent.com/inducer/pyopencl/master/examples/benchmark.py](https://raw.githubusercontent.com/inducer/pyopencl/master/examples/benchmark.py)

This gives an error:
[E] Mon Feb  5 06:43:19 2018: Compiler threw exception: Instruction Mapping: Bit-casts across different vector-sizes are not yet supported!
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Instruction Mapping: Bit-casts across different vector-sizes are not yet supported!

The exception comes from here:
[https://github.com/doe300/VC4C/blob/2a95e8aaf4080eeb237bf542f49c309aa6a85712/src/intermediate/TypeConversions.cpp#L25](https://github.com/doe300/VC4C/blob/2a95e8aaf4080eeb237bf542f49c309aa6a85712/src/intermediate/TypeConversions.cpp#L25)

I'm happy to look at fixing it, but some pointers as to what needs to happen would be great.

doe300, 2018-02-05:
As the error states, `Bit-casts across different vector-sizes are not yet supported!`

AFAIK, in most CPUs, vector-register is ONE extended register with e.g. 128 Bytes, where you use the amount of bits required for the type stored. In that case, a bit-cast across vector-sizes is not converted to any instruction (similar to `reinterpret_cast` in C++).

E.g. the input

    | short[0] | short[1] | short[2] | short[3] | ...

can be viewed as 

    | int[0]              | int[1]              | ...

without any actual conversion.

In the QPU however, a 16-element vector register is more like 16 different registers (with the same address, but only accessible from different element ALUs of one QPU). So to bitcast across vector- (and thus element-)sizes, we need to rotate the input vector and append/split the elements manually.

So the implementation of the bit-cast needs to generate extra code basically for almost every combination of input- and output-types which results in almost 50 different cases.

While some are easier to implement (e.g. (u)char2 -> (u)short 1 is `out[0] = (in[0] & 0xFF) | ((in[1] & 0xFF) << 8)`) others are far more complex (generally, we'll need extra calculations per element of the output-vector).

E.g. (u)short16 -> (u)int8 will require something like this:

    out[0] = (in[0] & 0xFFFF) | ((in[1] & 0xFFFF) << 16)
    out[1] = (in[2] & 0xFFFF) | ((in[3] & 0xFFFF) << 16)
    out[2] = (in[4] & 0xFFFF) | ((in[5] & 0xFFFF) << 16)
    out[3] = (in[6] & 0xFFFF) | ((in[7] & 0xFFFF) << 16)
    ...

I was hoping to come up with an equivalent, but far more efficient solution where I don't need to implement ~50 cases with extra calculation per element of the output-type.
So, if you find one, I would be very happy:)
Otherwise, you could implement all/most/some of the different cases that needs to be handled (e.g. by starting with the case for your sample code).

By the way, I cannot reproduce the error, at least not for the kernel `sum` in line 57 on my setup.

doe300, 2018-02-18:
The error reported has been fixed and the test-program doesn't exist anymore. @josiahw, can this be closed?

josiahw, 2018-03-12:
Yes I can confirm that this is now resolved.

closed 2018-03-12 by josiahw