doe300/VC4C #40: Description of dependencies

nomaddo, 2018-02-26:
Now the dependencies of `.deb` package is as follows.

https://github.com/doe300/VC4C/blob/master/CMakeLists.txt#L344

I am wondering it is right:

- Is `llvm-3.9-dev`  is always necessary?  If `LLVMLIB_FRONTEND` is disabled, `VC4C` might not depend on it?
- Is `clang-3.9` is always necessary?  if `LLVMIR_FRONTEND` is disabled, `VC4C` doesn't depend on it?

@doe300 Any Idea?

doe300, 2018-03-01:
I am investigating the dependencies of VC4C and also VC4CL and remove what is not necessary.

nomaddo, 2018-03-02:
At least `VC4C` with `SPIRV-LLVM` doesn't need `llvm-3.9-dev`: it can run on docker image of `nomaddo/cross-rpi:0.1`.

doe300, 2018-03-13:
I'm not done here. Now that I again have a stable internet connection, I can check the TODOs I left in the CMakeLists.txt (also for VC4CL)

commit 6d4d97e5 (2018-03-16): Removes wrong dependencies, closes #40
closed 2018-03-16 by doe300