doe300/VC4C #44: Optimize simple code

nomaddo, 2018-03-02:
Currently, unused mov instructions from `uniform` are generated at the beginning of kernel functions.

```c
__kernel void loop1 (__global float * a) {
  int id = get_local_id(0);
  float16 v = vload16 (id, a);
  vstore16(v * 2, id, a);
}
```

```asm
// Module with 1 kernels, global data with 0 words (64-bit each), starting at offset 1 words and 0 words of stack-frame
// Kernel 'loop1' with 49 instructions, offset 2, with following parameters: __global float* a (4 B, 1 items)
or -, unif, unif
or -, unif, unif
or r0, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or -, unif, unif
or ra0, unif, unif
or r0, r0, r0
ldi r1, 255
and r0, r0, r1
and r0, r0, r1
shl r0, r0, 4 (4)
shl r0, r0, 2 (2)
add r0, ra0, r0
or r2, r0, r0
mul24 r1, 4 (4), elem_num
v8adds r0, 8 (8), 8 (8)
sub.setf -, elem_num, r0
or.ifnc r0, 0 (0), 0 (0)
add.ifn r0, r2, r1
or tmu0s, r0, r0
nop.load_tmu0.never 
or r0, 2.000000 (33), 2.000000 (33)
fmul r0, r4, r0
or -, mutex_acq, mutex_acq
ldi vpw_setup, vpm_setup(size: 16 words, stride: 1 rows, address: h32(0))
or vpm, r0, r0
ldi vpw_setup, vdw_setup(rows: 1, elements: 16 words, address: h32(0))
ldi vpw_setup, vdw_setup(stride: 0)
or vpw_addr, r2, r2
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
or r0, unif, unif
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -45
nop.never 
nop.never 
nop.never 
not irq, qpu_num
nop.thrend.never 
nop.never 
nop.never 
```

These loads from `uniform` can be removed, ideally. And should be removed, because these are repeatedly executed (looped by `brr.ifallzc (pc+4) + -45`).

nomaddo, 2018-03-02:
Related: https://github.com/doe300/VC4C/issues/14#issuecomment-349919591

doe300, 2018-03-02:
As stated in #45, I will take this one.

nomaddo, 2018-03-05:
Thanks! The output become definitely better:

```c
__kernel void loop1 (__global float * a) {
  int id = get_local_id(0);
  float16 v = vload16 (id, a);
  vstore16(v * 2, id, a);
}
```

```asm
// Module with 1 kernels, global data with 0 words (64-bit each), starting at offset 1 words and 0 words of stack-frame
// Kernel 'loop1' with 37 instructions, offset 2, with following parameters: __global float* a (4 B, 1 items) (lids)
or r0, unif, unif
or ra0, unif, unif
or r0, r0, r0
ldi r1, 255
and r0, r0, r1
and r0, r0, r1
shl r0, r0, 4 (4)
shl r0, r0, 2 (2)
add r0, ra0, r0
or r2, r0, r0
mul24 r1, 4 (4), elem_num
v8adds r0, 8 (8), 8 (8)
sub.setf -, elem_num, r0
or.ifnc r0, 0 (0), 0 (0)
add.ifn r0, r2, r1
or tmu0s, r0, r0
nop.load_tmu0.never
or r0, 2.000000 (33), 2.000000 (33)
fmul r0, r4, r0
or -, mutex_acq, mutex_acq
ldi vpw_setup, vpm_setup(size: 16 words, stride: 1 rows, address: h32(0))
or vpm, r0, r0
ldi vpw_setup, vdw_setup(rows: 1, elements: 16 words, address: h32(0))
ldi vpw_setup, vdw_setup(stride: 0)
or vpw_addr, r2, r2
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
or r0, unif, unif                                    /// (1)
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -33
nop.never
nop.never
nop.never
not irq, qpu_num
nop.thrend.never
nop.never
nop.never
```

nomaddo, 2018-03-07:
In my imagination, we need more improvement for better performance of the compiler.
In the current layout, we cannot apply loop optimizations (like loop unrolling).

In the following code, we can easily apply loop unrolling as follows:

```c
for (i = gid0_start; i < gid0_end; i++)
  for (j = gid1_start; j < gid1_end; j++)
      for (k = gid2_start; k < gid2_end; k++) {
        // main body
      }

```

```c
for (i = gid0_start; i < gid0_end; i++)
  for (j = gid1_start; j < gid1_end; j++) {
      for (k = gid2_start; k < gid2_end - 4; k += 4) // expand 4 times
        // main body
      for (k = k; k < gid2_end; k ++)
        // main body
    }
```

To apply that, we have to know 

- the value of the beginning value of each dimension (of global_id, or local_id)
- the value of end value of each diemension

In cpu code, these values are often known in the entrance of the loops.
But we cannot apply this optimization in ease way, because `VC4CL` pass such parameters by `uniform` now.

My suggestion is changing the layout as follow:

- the start value of global_id_0
- the end value of global_id_0
- the start value of global_id_1
- the end value of global_id_1
- the start value of global_id_2
- the end value of global_id_2
- ......

Then, code will become like

```asm
mov gid0_start, uniform
mov gid0_end, uniform
mov gid1_start, uniform
mov gid1_end, uniform
mov gid2_start, uniform
mov gid2_end, uniform
...

// label_0
mov i, gid0_start
// label_1
mov j, gid1_start
// label_2
mov k, gid2_start

//
// main body
//

iadd k, k, 1                            // (1)
isub.setf -, k, gid2_end
jnz label_2
iadd j, j, 1                             // (2)
isub.setf -, j, gid1_end
jnz label_1
iadd i, i, 1                             // (3)
isub.setf -, i, gid0_end
jnz label_2
//
// end of program
//
```

When we apply loop-unrolling, we expand loops and change increment value of (1), (2), or (3).

I am not sure that we can always apply this method (if we use group_id, examples).
At least, if host code use only `get_global_id`, I want to apply such optimization.

In my opinion, loop unrolling is MUST for performance.
(The reason why we use GPU is only for performance, isn't it?)

@doe300 Do you have idea about that?

doe300, 2018-03-14:
I'm not sure I get your meaning.
In the first code block (3 nested loops iterating over global space), is the code example the kernel-code or is the ``//main body` part the kernel code and the loops the structure of work-items being executed?

For the first case:
I think the kernel-code is "malformed", since one kernel execution only represents a single global-ID

For the second case:
If I understand it correctly, this would extend the optimization already done in step `UnrollWorkGroups` to create actual loops over the work-items into all 3 dimensions and with initially loaded upper and lower bounds. It is only possible to combine work-items from different work-groups, since we cannot merge work-items of the same work-group (this would mess up a lot of things, among others the semaphores).

nomaddo, 2018-03-14:
Sorry for my confusing comment.
My concern is the way to unroll loops of kernel program.

Currently we cannnot apply such optimization, because kernel code cannot know how many times the code is execute in the beginning of the loop.

```c
 __kernel void loop1 (__global float * a){
   int id = get_local_id(0);
   a[id] = a[id] * 2;
}
```

```asm
// Module with 1 kernels, global data with 0 words (64-bit each), starting at offset 1 words and 0 words of stack-frame
// Kernel 'loop1' with 31 instructions, offset 2, with following parameters: __global out float* a (4 B, 1 items), __global float* b (4 B, 1 items) (lids)
// label: %start_of_function
or r0, unif, unif
or ra0, unif, unif
or -, unif, unif
// label: %tmp.0
or r0, r0, r0
ldi r1, 255
and r0, r0, r1
and r0, r0, r1
shl r0, r0, 2 (2)
add r1, ra0, r0
nop.never
or tmu0s, r1, r1
nop.load_tmu0.never
or r0, r4, r4
or -, mutex_acq, mutex_acq
ldi vpw_setup, vpm_setup(size: 16 words, stride: 1 rows, address: h32(0))
fmul vpm, r0, 2.000000 (33)
ldi vpw_setup, vdw_setup(rows: 1, elements: 1 words, address: h32(0))
ldi vpw_setup, vdw_setup(stride: 0)
or vpw_addr, r1, r1
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
// label: %end_of_function
or r0, unif, unif                             // (1)
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -27 // to %start_of_function
nop.never
nop.never
nop.never
not irq, qpu_num
nop.thrend.never
nop.never
nop.never
```

The exit condition of the loop (currently like one-dimesion for-loop) is depend on the value of `uniform` in (1).
To apply unrolling, we have to know how many times the loop will repeat in the beginning of the loop. 
Then, we cannot apply unrlling unless the usage of uniform is changed.

doe300, 2020-12-08:
Closing, since to my understanding this is implemented

closed 2020-12-08 by doe300