doe300/VC4C #46: Compilation failed due to spirv-llvm?

nomaddo, 2018-03-02:
Currently `VC4C` on x86_64 fail to compile as follows:
Debian packages are downloaded from https://circleci.com/gh/nomaddo/VC4C/137#artifacts/containers/0.

```sh
$ docker run -v /tmp:/tmp -v ${HOME}/clABCMarks:/marks --rm -it nomaddo/cross-rpi:0.1

$ dpkg-deb -i vc4cl-stdlib-0.4-Linux.deb
$ dpkg-deb -x  vc4c-0.4-Linux.deb /
$ /opt/SPIRV-LLVM/build/bin/clang -cc1 -triple spir-unknown-unknown -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-all -Wno-gcc-compat -x cl -emit-pch -o /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch -I /usr/local/include/vc4cl-stdlib/ /usr/local/include/vc4cl-stdlib/opencl-c.h
$ VC4C --asm -o /tmp/hoge.s /marks/mul2.cl
VC4C --asm -o /tmp/hoge.s /marks/mul2.cl
[D] Fri Mar  2 13:26:31 2018: Compiling '/marks/mul2.cl' into '/tmp/hoge.s' with options '' ...
[D] Fri Mar  2 13:26:31 2018: Temporary file '/tmp/vc4c-PdE2jQ' created
[D] Fri Mar  2 13:26:31 2018: Temporary file '/tmp/vc4c-Ptoo1K' created
[I] Fri Mar  2 13:26:31 2018: Compiling OpenCL to LLVM-IR with :/opt/SPIRV-LLVM/build/bin/clang -cc1 -triple spir-unknown-unknown  -I /marks -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch -x cl -S -emit-llvm-bc -o /tmp/vc4c-Ptoo1K /marks/mul2.cl
[I] Fri Mar  2 13:26:31 2018: Converting LLVM-IR to SPIR-V with :/opt/SPIRV-LLVM/build/bin/llvm-spirv -o /tmp/vc4c-PdE2jQ /tmp/vc4c-Ptoo1K
[D] Fri Mar  2 13:26:31 2018: Temporary file '/tmp/vc4c-Ptoo1K' deleted
[I] Fri Mar  2 13:26:31 2018: Compilation complete!
[I] Fri Mar  2 13:26:31 2018: Using SPIR-V frontend...
[D] Fri Mar  2 13:26:31 2018: Read SPIR-V binary with 161 words
[D] Fri Mar  2 13:26:31 2018: Starting parsing...
[D] Fri Mar  2 13:26:31 2018: SPIR-V header parsed: magic-number 0x7230203, version 0x10000, generator 6000e, max-ID 21
[D] Fri Mar  2 13:26:31 2018: Using supported capability: Addresses
[D] Fri Mar  2 13:26:31 2018: Using supported capability: Linkage
[D] Fri Mar  2 13:26:31 2018: Using supported capability: Kernel
[D] Fri Mar  2 13:26:31 2018: Using supported capability: Vector16
[D] Fri Mar  2 13:26:31 2018: Importing extended instruction set: OpenCL.std
[D] Fri Mar  2 13:26:31 2018: Using a physical OpenCL memory model
[D] Fri Mar  2 13:26:31 2018: Kernel-method found: loop1
[E] Fri Mar  2 13:26:31 2018: Met unsupported instruction-decoration 11
[E] Fri Mar  2 13:26:31 2018: Unsupported operation:  at ?
[E] Fri Mar  2 13:26:31 2018:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x121 [0x7fead7f2fe13]
[E] Fri Mar  2 13:26:31 2018:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::spirv2qasm::SPIRVParser::parse(vc4c::Module&)+0x53b [0x7fead813ca6d]
[E] Fri Mar  2 13:26:31 2018:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::convert()+0x11f [0x7fead7f3165d]
[E] Fri Mar  2 13:26:31 2018:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x250 [0x7fead7f31d44]
[E] Fri Mar  2 13:26:31 2018:  (5) VC4C : main+0xcf6 [0x5642ad15ddfc]
[E] Fri Mar  2 13:26:31 2018:  (6) /lib/x86_64-linux-gnu/libc.so.6 : __libc_start_main+0xf1 [0x7fead66fc2b1]
[E] Fri Mar  2 13:26:31 2018:  (7) VC4C : _start+0x2a [0x5642ad15c72a]
[D] Fri Mar  2 13:26:31 2018: Temporary file '/tmp/vc4c-PdE2jQ' deleted
[E] Fri Mar  2 13:26:31 2018: Compiler threw exception: Parser: Unsupported operation: 
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Parser: Unsupported operation: 
Aborted (core dumped)
```

@doe300 Do you have any idea?

doe300, 2018-03-02:
The key is this line (though I admit, the error-output is not very good):

> Met unsupported instruction-decoration 11

Some entry is decorated with the *BuiltIn* decoration, which is not yet supported.

What is the content of the `/marks/mul2.cl` file?

nomaddo, 2018-03-05:
`mul2.cl` is as follows:

```c
__kernel void loop1 (__global float * a) {
  int id = get_local_id(0);
  a[id] = a[id] * 2;
}
```

`VC4C` is the latest. I am wondering what the error messages said......

doe300, 2018-03-08:
I cannot reproduce the error on my setup.
I also just saw, that your code to generate the `VC4CLStdLib.h.pch` seems wrong (you take the `opencl-c.h` as input and not the `VC4CLStdLib.h`). I think, you generate a wrong PC file (without the VC4CL sts-lib implementation)

closed 2018-03-11 by doe300