doe300/VC4C #50: Better flattening of basic blocks

nomaddo, 2018-03-06:
```c
kernel void if1 (global float a[]) {
  int id = get_global_id(0);
  if ((id & 0x01) == 0)
    a[id] = 3.0;
}
```

```asm
// Module with 1 kernels, global data with 0 words (64-bit each), starting at offset 1 words and 0 words of stack-frame
// Kernel 'if1' with 46 instructions, offset 2, with following parameters: __global out float* a (4 B, 1 items) (lSize, lids, gidX, offX)
// label %start_of_function
or r1, unif, unif
or r2, unif, unif
or r0, unif, unif
or ra2, unif, unif
or ra1, unif, unif
// label %tmp.0
or ra0, r0, r0; v8min r0, r1, r1
ldi r1, 255
and r3, r0, r1
or r2, ra2, ra2; v8min r0, r2, r2
and r1, r0, r1
mul24 r0, ra0, r3
add r0, r2, r0
add r1, r0, r1
nop.never 
or r0, r1, r1
and.setf -, r0, 1 (1)
xor.ifzc r0, 1 (1), 1 (1); v8min.ifz r0, 1 (1), 1 (1)
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + 4 // to %tmp.1      --- (1)
nop.never 
nop.never 
nop.never 
brr.ifanyz (pc+4) + 10 // to %tmp.2    --- (2)
nop.never 
nop.never 
nop.never 
// label %tmp.1
or r0, r1, r1
shl r0, r0, 2 (2)
or -, mutex_acq, mutex_acq
ldi vpw_setup, vpm_setup(size: 16 words, stride: 1 rows, address: h32(0))
itof vpm, 3 (3)
ldi vpw_setup, vdw_setup(rows: 1, elements: 1 words, address: h32(0))
ldi vpw_setup, vdw_setup(stride: 0)
add vpw_addr, ra1, r0
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
// label %end_of_function
or r0, unif, unif
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -42 // to %start_of_function
nop.never 
nop.never 
nop.never 
not irq, qpu_num
nop.thrend.never 
nop.never 
nop.never 
```

In this code, the branch of (1) can be removed. We should,

- check condition of (2) first
- if the condition is false, fall through to the basic block (of writing to memory)

Anyway, we need better flattening of basic blocks.
Currently, we just use the iterator of traverseing all instructions.
We need more carefully flattening, which is keen to the shape of the graph.

doe300, 2018-03-06:
You are right about the imperfect linear traversation of instructions.
The optimization suggested here was already implemented and then reverted in  75925b274a87b6828a21d85382825a46f052a449 because it caused wrong behaviour. But in theory, it should be possible to remove this branch.

doe300, 2018-03-09:
So my thoughts are:
- rewrite method to store basic blocks in graph according to their predecessors/successors (instead of linear vector as current). This one is fairly easy, since we already have a representation of a CFG. Only insertion/deletion of basic blocks need to be handled extra.
- rewrite basic block to store instructions in graph according to their data and order dependencies (instead of linear as current)
- rewrite optimizations to access/traverse graph-structure (for most optimizations, it makes no difference how the instructions are ordered, for some it would greatly improve performance)
- remove as many uses of linear traversal as possible

Some notes:
While many optimizations will have no problem traversing a graph instead of a linear vector, others wont work any more. So maybe we need to split them into 2 steps: step 1 uses graph structure, step 1.5 converts graph structure into linear structure (e.g. by instruction scheduling), step 2. uses linear structure. Or we create a "linear" traversal of the instruction graph, which represents the order they are later scheduled, then we can fuse step 1 and 2.
Also, traversing data dependency graphs which span more than a basic block will have loops in it, so we need to make sure to avoid infinitely traversing those loops.

nomaddo, 2018-04-25:
In this case, all the order of basic blocks are inefficient as follows:

```c
void kernel test(float global * a, const int n){
  float sum = 0.0;
  for (int i = 0; i < 10; i++)
    sum += a[i];
  a[get_global_id(0)] = sum;
}
```

```
$ ./build/VC4C -fno-unroll-loops --asm -o /tmp/hoge.s test.cl
```

```
// Module with 1 kernels, global data with 0 words (64-bit each), starting at offset 1 words and 0 words of stack-frame
// Kernel 'test' with 61 instructions, offset 2, with following parameters: __global out float* a (4 B, 1 items), int n (4 B, 1 items) (lSize, lids, gidX, offX)
// label: %start_of_function
or ra4, unif, unif
or ra3, unif, unif
or ra2, unif, unif
or ra1, unif, unif
or ra0, unif, unif
or -, unif, unif
or r1, 0 (0), 0 (0); v8min r3, 0 (0), 0 (0)
brr (pc+4) + 19 // to %tmp.1
nop.never 
nop.never 
nop.never 
// label: %tmp.2
ldi r0, 255
and r3, ra4, r0
and r1, ra3, r0
mul24 r0, ra2, r3
add r0, ra1, r0
add r0, r0, r1
shl r0, r0, 2 (2)
or -, mutex_acq, mutex_acq
ldi vpw_setup, vpm_setup(size: 16 words, stride: 1 rows, address: h32(0))
or vpm, r2, r2
ldi vpw_setup, vdw_setup(rows: 1, elements: 1 words, address: h32(0))
ldi vpw_setup, vdw_setup(stride: 0)
add vpw_addr, ra0, r0
or -, vpw_wait, vpw_wait
or mutex_rel, 1 (1), 1 (1)
brr (pc+4) + 21 // to %end_of_function
nop.never 
nop.never 
nop.never 
// label: %tmp.1
or r1, r1, r1
shl r0, r1, 2 (2)
add tmu0s, ra0, r0
nop.load_tmu0.never 
fadd r2, r3, r4
add r1, r1, 1 (1)
or r0, r1, r1
max r0, r0, 10 (10)
xor.setf -, r0, r1
xor.ifz r0, 1 (1), 1 (1); v8min.ifzc r0, 1 (1), 1 (1)
or.setf -, r0, r0
or.ifzc r1, r1, r1; v8min.ifzc r3, r2, r2
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -17 // to %tmp.1
nop.never 
nop.never 
nop.never 
brr.ifanyz (pc+4) + -40 // to %tmp.2
nop.never 
nop.never 
nop.never 
// label: %end_of_function
or r0, unif, unif
or.setf -, elem_num, r0
brr.ifallzc (pc+4) + -57 // to %start_of_function
nop.never 
nop.never 
nop.never 
not irq, qpu_num
nop.thrend.never 
nop.never 
nop.never 
```

commit 259ee21f (2018-06-09): Adds optimization to reorder basic blocks, fixes #50 (#101)

* Adds optimization to reorder basic blocks, fixes #50

* Fix error in test kernel
closed 2018-06-09 by doe300