doe300/VC4C #53: AArch64 requires -fPIC

josiahw, 2018-03-12:
I'm working to build this on a rpi3 with debian/Aarch64 and got a few hundred of these messages when compiling vc4c:

/usr/bin/ld: CMakeFiles/objvc4asm.dir/Parser.cpp.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTVN6Parser7MessageE' which may bind externally can not be used when making a shared object; recompile with -fPIC

The fix looks simple enough?

josiahw, 2018-03-12:
To fix, the following need to be passed to vc4asm (in the cmake file) when building:
 -DCMAKE_CXX_FLAGS="-fPIC"
 -DCMAKE_C_FLAGS="-fPIC"

nomaddo, 2018-03-13:
@doe300 Will we support Aarch64? it is not official, and i am not sure VC4 driver works correctly.

doe300, 2018-03-13:
Officially, only Raspbian is supported. But since adding these flags does no harm, I see no reason to not doing it. 
For any other OS though, the environment has to be set up (as you mentioned), among other things by providing the mailbox kernel-module as well as the bcmhost-library. But if you do that, I see no reason why VC4CL shouldn't work on other OSes.

Terminus-IMRC, 2018-03-13:
Running VC4CL on AArch64 is not yet supported (though some people are trying to enable it) because:

* Mailbox interface (`/dev/vcio`) is not yet ported to AArch64.
* GPU is not (yet) capable of allocating memory by request from CPU in AArch64 mode.

But, yes, adding the flags doesn't matter.

doe300, 2018-03-16:
Just to clarify the OS support:
- VC4C should be OS independent (should run on any modern Linux), so issues filed with other OSes which are not specific to this OS will be looked into (see #61).
- VC4CL on the other hand is only officially supported on latest Raspbian OS

doe300, 2018-04-07:
@josiahw , can this be closed?

closed 2018-05-06 by doe300