doe300/VC4C #57: Use clang-format

doe300, 2018-03-13:
We should use clang-format to make sure, we have a unified formatting.

Any comments on this?
Does anyone have a good template for clang-format, we could use?

This would also apply to the VC4CL project.

nomaddo, 2018-03-14:
I agree with your opinion. Some files are unformatted and we dont want to care for format differences.
As a trial, I made some config to use clang-format.

This patch enable to

- `make clangformat` to make all source-code formatted.
- Check all source-codes are formatted in clang-format in circleci.

Format style is defined in `.clang-format`, which is created by `clang-format -style=LLVM -dump-config > .clang-format` now. You can choose code styles as you like.

doe300, 2018-03-24:
Closed with #59

closed 2018-03-24 by doe300