doe300/VC4C #58: precomplication failed

nomaddo, 2018-03-13:
I am trying to run `VC4C` on fresh raspbian env, but somehow compile failed as follows:

```
pi@raspberrypi:~/VC4C$ ./build/VC4C -o /tmp/hoge ../VC4CL/mul2.cl 
[D] Tue Mar 13 22:47:53 2018: Compiling '../VC4CL/mul2.cl' into '/tmp/hoge' with options '' ...
[D] Tue Mar 13 22:47:53 2018: Temporary file '/tmp/vc4c-PXslFI' created
[I] Tue Mar 13 22:47:53 2018: Compiling OpenCL to LLVM-IR with :/usr/bin/clang-3.9 -cc1 -triple spir-unknown-unknown  -I ../VC4CL -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch -x cl -S -emit-llvm-bc -o /tmp/vc4c-PXslFI ../VC4CL/mul2.cl
[E] Tue Mar 13 22:47:59 2018: Errors in precompilation:
[E] Tue Mar 13 22:47:59 2018: ../VC4CL/mul2.cl:3:12: error: use of undeclared identifier 'i'
  a[i] = a[i] * 2;
           ^
../VC4CL/mul2.cl:3:5: error: use of undeclared identifier 'i'
  a[i] = a[i] * 2;
    ^
2 errors generated.

[E] Tue Mar 13 22:47:59 2018:  (1) /home/pi/VC4C/build/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x76c21d78]
[E] Tue Mar 13 22:47:59 2018:  (2) /home/pi/VC4C/build/libVC4CC.so.1.2 : +0x349ba0 [0x76c55ba0]
[E] Tue Mar 13 22:47:59 2018:  (3) /home/pi/VC4C/build/libVC4CC.so.1.2 : +0x349edc [0x76c55edc]
[E] Tue Mar 13 22:47:59 2018:  (4) /home/pi/VC4C/build/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x36c [0x76c5674c]
[E] Tue Mar 13 22:47:59 2018:  (5) /home/pi/VC4C/build/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x154 [0x76c54d34]
[E] Tue Mar 13 22:47:59 2018:  (6) /home/pi/VC4C/build/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0xd4 [0x76c23ce4]
[E] Tue Mar 13 22:47:59 2018:  (7) ./build/VC4C : main+0xc4c [0x232e4]
[E] Tue Mar 13 22:47:59 2018:  (8) /lib/arm-linux-gnueabihf/libc.so.6 : __libc_start_main+0x114 [0x73f3c678]
[D] Tue Mar 13 22:47:59 2018: Temporary file '/tmp/vc4c-PXslFI' deleted
[E] Tue Mar 13 22:47:59 2018: Compiler threw exception: Pre-compilation: Error in precompilation: ../VC4CL/mul2.cl:3:12: error: use of undeclared identifier 'i'
  a[i] = a[i] * 2;
           ^
../VC4CL/mul2.cl:3:5: error: use of undeclared identifier 'i'
  a[i] = a[i] * 2;
    ^
2 errors generated.

terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: ../VC4CL/mul2.cl:3:12: error: use of undeclared identifier 'i'
  a[i] = a[i] * 2;
           ^
../VC4CL/mul2.cl:3:5: error: use of undeclared identifier 'i'
  a[i] = a[i] * 2;
    ^
2 errors generated.

Aborted
```

mul2.cl is as follows:

```c
kernel void f (global float * a) {
  int id = get_global_id(0);
  a[i] = a[i] * 2;
}
```

This issue happened in x86 env. @doe300 any idea?

josiahw, 2018-03-14:
try using a[id] or rename id to i?

nomaddo, 2018-03-14:
@josiahw  Thanks you so much. I missed that...

closed 2018-03-14 by nomaddo