doe300/VC4C #60: Implement register spilling

doe300, 2018-03-14:
We need to implement register spilling to be able to support more complex kernels.

(see https://github.com/doe300/VC4CL/issues/24)

nomaddo, 2018-04-06:
Notice: 
To spill registers, cache incoherence will be a problem between VPM and TMU.
DMA Load & Store doesn't use L2-cache, but TMU does. 
 
Example (assembled by pi-videocore):

```python
import numpy as np

from videocore.assembler import qpu
from videocore.driver import Driver

@qpu
def dma_store(asm, preload):
    mov(ra0, uniform)
    mov(ra1, uniform)

    shl(r0, element_number, 2)
    iadd(r0, r0, ra0)

    if preload: # ATTENTION: difference of of the value `preload` chenge the result
        # we don't use the value loaded here
        mov(tmu0_s, r0)
        nop(sig='load tmu0')

    setup_vpm_write()
    mov(vpm, element_number)

    setup_dma_store(nrows=1)
    start_dma_store(ra0) # store the value
    wait_dma_store()

    mov(tmu0_s, r0)
    nop(sig='load tmu0') # load the value, which is just store into the buffer

    setup_vpm_write()
    mov(vpm, r4)

    setup_dma_store(nrows=1)
    start_dma_store(ra1) # resultにstore
    wait_dma_store()

    exit()
	
with Driver() as drv:
    print('----- enable preload -----')
    buffer = drv.alloc(16, 'uint32')
    buffer[:] = 0
    result = drv.alloc(16, 'uint32')
    result[:] = 0

    print('[Before]')
    print(buffer)
    print(result)

    drv.execute(
        n_threads=1,
        program=drv.program(dma_store, True),
        uniforms=[buffer.address, result.address]
    )

    print('[After]')
    print(buffer)
    print(result)
	
with Driver() as drv:
    print('----- disable preload -----')
    buffer = drv.alloc(16, 'uint32')
    buffer[:] = 0
    result = drv.alloc(16, 'uint32')
    result[:] = 0

    print('[Before]')
    print(buffer)
    print(result)

    drv.execute(
        n_threads=1,
        program=drv.program(dma_store, False),
        uniforms=[buffer.address, result.address]
    )

    print('[After]')
    print(buffer)
    print(result)
```

doe300, 2018-04-06:
Yeah, I think this is also the problem in #30