doe300/VC4C #64: Issues with rebuilding and default clang

doe300, 2018-03-17:
Sometimes, when VC4C is re-built (CMake already configured) using the default clang, the compilation fails due to `CLANG_PATH` not found.

My theory is that is has something to with `CLANG_PATH` being set as a macro in CMake, not a CMake variable and therefore might not be saved?!

See also https://github.com/doe300/VC4CL/issues/19 and #34.

josiahw, 2018-03-19:
A cleaner alternative is to set up cmake so that you can do:
mkdir build && cd build && cmake .. && make
This lets you rm the build config easily when configuration is changed. 

I also sometimes have to kill the lib directory in VC4C to update the .pch file correctly.

doe300, 2018-03-20:
To update the `VC4CLStdLib.h.pch`, you need to delete the old file and re-build the target `vc4cl-stdlib`.

doe300, 2018-12-15:
Since fe545cd5910c60a12cd5aa6c2c544d7884e822a2, better way (see https://github.com/doe300/VC4C/issues/64#issuecomment-374412301) can be used.

closed 2018-12-15 by doe300