doe300/VC4C #66: Cross-compiled build does not run on Rpi B

doe300, 2018-03-24:
The cross-compiled build does not run on original Raspberry Pi B (not B+, not 2B) due to probably architecture mismatch.
The original error message when trying to execute `VC4C` is:

    Illegal instruction

`cat /proc/cpuinfo` spills out:

    ARMv6-compatible processor rev 7 (v6l)

 `readelf -A /usr/local/bin/VC4C` gives 

    Tag_CPU_name: "7-A"
    Tag_CPU_arch: v7

nomaddo, 2018-03-26:
This is a problem of dockage images.
Currently, we use packages of debian armhf (see https://github.com/nomaddo/cross-rpi/blob/master/Dockerfile#L4).
These seem to include ARMv7 instructions (https://wiki.debian.org/RaspberryPi):

    Debian armhf will not work (see below) for the first version but does work with the second one. The Zero uses the same SoC/CPU as the first version, so they should function identically.

nomaddo, 2018-04-25:
Status:

The problem is mismatch between config of compiler and `.deb` located in raspbian: the configs of cxx11 ABI are different. We may need to rebuild gcc-cross-compiler or llvm and its dependencies.
It is hard to build manually. So I am thinking how to solve this......

nomaddo, 2018-05-08:
solved by
#88 and https://github.com/doe300/VC4CL/pull/30

closed 2018-05-08 by nomaddo