doe300/VC4C #70: Split optimizations and normalization

nomaddo, 2018-03-28:
Currently, `OptimizationPass`s and `OptimizationStep`s represents optimization and normalization to regular VC4 assembler (like `addStartStopSegment`. This transform the intermediate language to language, which is closer to VC4 assembly).

Normalizations should not be applied, but optimizations should.
It's better to split them, and normalizations should be applied in earlier phases.

doe300, 2018-03-28:
You are probably right. The question is only, what order we use:

If we first normalize (convert to code executable on the VC4) and then optimize, the optimizations are limited (e.g. usage-ranges and read-after-write limitations).
If we first optimize and then normalize, the normalization might introduce inefficiencies, we cannot remove anymore (e.g. `nop`s to solve read-after-writes).

We probably need to split into:

1. normalization (phi-elimination, inlining, intrinsics, etc.)
2. optimization
3. resolve hardware limitations (read-after-write, etc.)

nomaddo, 2018-03-28:
I agree, the order is problem. So how to relax real VC4C limitations are considered.
Probably, the language (we use for optimization) should be like that (the same as we currently use) :

- Basic blocks are the basis
- Ignore delay slots
- Ignore register-limitation (but `r4`, `r5` and IO registers are cared)
- Use `move` (instead of `or a, b, b` for ease of optimizations)
- Care the difference of `SmallImmediate` and ordinary immediate values
- Ignore dual issue (instruction scheduler care it)
- Ignore the delay of SFU or vector-rotation (the matter of instruction scheduler)

This language should care for the number of instructions, and almost all optimizations can be done in the language. If some points don't affect the number of instructions, we can relax the limitation about the points.

Mayby the phases are:

- First Normalization (phi-eliminate, inlining. add the basic blocks of loading parameter from `uniform`, etc...)
- First Basic Optimization
- Loop Optimization
- Second Basic Optimization
- Second Normalization (register allocation, scheduling, fill delay slot, solve delays)
- (Second optimization? scheduling sometimes work after register allocation......)
- Codegen

nomaddo, 2018-04-11:
@doe300 can close this issue?

doe300, 2018-04-11:
Not yet, I want to rewrite the optimization steps themselves so we can better utilize the CFG and dependency graphs.

closed 2018-04-11 by doe300