doe300/VC4C #72: Compress multiple constants into one register.

nomaddo, 2018-03-28:
The number of registers in VC4 is very limited.
One of the technique to avoid this, is compression of constants into one register.

For example, `ldi r0, 0` fill in all of element of vector-register `r0`.
If each element of `r0` hold a different value, we can save registers.
So we use vector-rotation with broadcast.

Things we should do is,

- (decide which constants should be packed into a register)
- Load constants with specific setflags
- Load a constant from a reigster (which hold different values in each elements) using vector-rotate (see P20 **Horizontal Vector Rotation** in VideoCore IV Architecture Reference Guide).

doe300, 2018-03-29:
Although this is definitively possible to save registers, I don't think this is the direction we should go in.
The problem is that loading a constant takes a single instruction while rotating registers takes at least 2 to 3.

Also, are you talking about compressing constants (loaded via `loadi`) or parameters (read from `uniform`)?

nomaddo, 2018-03-30:
Sorry for confusing explanation. I am talking about constants from `uniform`.
Yes, getting values from compressed values are expensive than usual as you said. 
For constants (loaded by `ldi` or `mov`), these should not be applied this optimization.

nomaddo, 2018-04-13:
I think we can save instructions to store values from uniform.
It's enough to spend two instructions to store and compress.

https://github.com/nineties/py-videocore/blob/master/examples/sgemm_1thread.py#L37

This uses "load imm per-elmt signed" (page 33 in VC4C reference manual).

nomaddo, 2018-04-16:
If all bits of `Per-element MS (sign) bit` are zero, your are right.
Come to think of it, If you want to set a flag of only 1-bit, we can use just isub like `isub.set -, element_number, 2` and `or.jzs out, unif, unif`.