doe300/VC4C #89: Suppress unrolling

nomaddo, 2018-04-25:
Currently the front-end `clang` unroll loops because `-O3` is specified as compilation option.
This makes harder to do 

- fusion of write back to memory
- vectorization in `VC4C` unrolling
- Suppress explosion in use of registers

It requires careful optimization design, but tt's better to do in our optimization pass.

doe300, 2018-05-05:
I'm not sure about this one. Sure, if we have full control over loop unrolling, we can adapt it more to our platform. But I would not trust the efficiency and correctness of out optimizations over the clang compiler;)

nomaddo, 2018-05-07:
Unrolling inner-loops is easy compared with other loop optimizations. 
Vectorization require some analysis no to make wrong program, but unrolling inner-loops are always correct.