doe300/VC4C #105: TestVC4C yields random segmentation faults on a RPi Zero

alexsee75, 2018-06-15:
While VC4C could be compiled without issues and VC4CL tests worked reasonably well (>80%), I am concerned that several runs of TestVC4C give random segmentation faults at two different points in the test and in three runs I could not get any runs without a segfault so I have no idea how many tests failed or were successfully run. Attached you find several logs (compiled with -DDEBUG=ON)

I would be less concerned if the errors were always at the same place.... something is fishy here, perhaps unitialized variables or memory? Random corruption?

RPi Zero has 384M memory and 2G swap, GPU memory is set to 128M, experimental GPU driver is deactivated.

Should I take a look at the test program, randomize the tests or make them run one-by-one? This would probably the fastest way to find out if it is random corruption or one specific test case that is causing this...
[TestVC4C_2.log](https://github.com/doe300/VC4C/files/2106536/TestVC4C_2.log)
[TestVC4C_3.log](https://github.com/doe300/VC4C/files/2106537/TestVC4C_3.log)
[TestVC4C_4.log](https://github.com/doe300/VC4C/files/2106538/TestVC4C_4.log)

alexsee75, 2018-06-15:
> Can you run TestVC4C with gdb/valgrind and attach the output?

Attached.

> Did you cross-compile or compile natively?

I compiled natively on the RPi zero. Strangely it is necessary to set CROSS_COMPILE=ON to get a correct deb package - otherwise it's AMD64.
[TestVC4C_5.log](https://github.com/doe300/VC4C/files/2106730/TestVC4C_5.log)

doe300, 2018-06-30:
I can reproduce bug and do know why it occurs, but haven't found a solution yet.

A little bit more detail:
The crash occurs in the std-lib emulation test-cases, because a Local is accessed which somehow has already been cleaned up.

closed 2018-07-12 by doe300