doe300/VC4C #107: Compile failed with gcc 4.9, clang-3.9, cmake 3.6.2

edsonke, 2018-07-03:
Hi there,

I compile VC4C failed, can't find any useful message in issues, might need some suggestion of it.
Thanks a lot.

The error message:
```
[ 15%] Building CXX object build/CMakeFiles/VC4CC.dir/BackgroundWorker.cpp.o
c++: error: unrecognized command line option ‘-Wnull-dereference’
c++: error: unrecognized command line option ‘-Wsuggest-override’
```

It is not helpful if i disable those two flags, other compile failure happened:

```
<command-line>:0:0: warning: "_GNU_SOURCE" redefined
<command-line>:0:0: note: this is the location of the previous definition
[ 16%] Building CXX object build/CMakeFiles/VC4CC.dir/BasicBlock.cpp.o
<command-line>:0:0: warning: "_GNU_SOURCE" redefined
<command-line>:0:0: note: this is the location of the previous definition
In file included from /home/pi/bean-ready/qpu/VC4C/src/analysis/ControlFlowGraph.h:10:0,
                 from /home/pi/bean-ready/qpu/VC4C/src/BasicBlock.cpp:10:
/home/pi/bean-ready/qpu/VC4C/src/analysis/../Graph.h: In member function ‘vc4c::Direction vc4c::Edge<NodeType, Relation, Direction>::getDirection() const’:
/home/pi/bean-ready/qpu/VC4C/src/analysis/../Graph.h:483:24: error: ‘Direction’ is not a class, namespace, or enumeration
                 return Direction::NONE;
                        ^
```

---

Environment: 
Linux raspberrypi 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux

cmake version 3.6.2

clang version 3.9.0-4 (tags/RELEASE_390/final)
Target: armv7l-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin

gcc (Raspbian 4.9.2-10+deb8u1) 4.9.2

doe300, 2018-07-03:
VC4C(L) is written with C++14, which is not supported by GCC 4.9 (see https://github.com/doe300/VC4CL/issues/17).

If you update your Raspian to the latest version, it will also update GCC to version 6.3 fixing the compilation issue.

edsonke, 2018-07-03:
@doe300 Thanks for your quick reply, I will try it.

Since I've some reason that still working on Jessie instead of latest Raspbian, 
and I thin these project must build passed before, so I ask about this.

On the other hand, I also did some experiment and build pass on Jessie (with roll back and workarounds ;) )
Here are some reference information, hope somebody might help:

Build pass feature: OpenCL 1.2

VC4C commit: 2becded1c2874fcf4f1faac7b153698b93b68dd7
VC4CL commit: d2bc1eca8b1a3c8b3af9a7144540ffe115dad779

Modify icd_loader.cpp manually, like commit: [Refine compability](https://github.com/edsonke/VC4CL/commit/380058bd56263db6162fbf61ca4398d371f322ed)

Thanks

closed 2018-07-03 by edsonke