doe300/VC4C #108: Compile error

alexzk1, 2018-07-04:
BitcodeReader.cpp:620:55: error: ‘const class llvm::Instruction’ has no member named ‘hasUnsafeAlgebra’
     if(llvm::isa<llvm::FPMathOperator>(&inst) && inst.hasUnsafeAlgebra())

doe300, 2018-07-04:
What version of LLVM do you have installed?

alexzk1, 2018-07-04:
I use arch linux, llvm 6.0.0-4
That I tried on x86 build native app for it (I hope I could use compiler on x86 to cross then for PI).
Now waiting compile on PI, but it has archarm there too.

doe300, 2018-07-04:
Looks related to #106. I myself haven't tested LLVM 6.0 yet, and it seems it does not compile.

alexzk1, 2018-07-04:
Also I spent many hours on another issues....for example this is a must -DMULTI_THREADED=OFF
otherwise it tries to link -lpthreads while proper is -lpthread

doe300, 2018-07-04:
I've never used Arch Linux. Is it so different then other Linuxes? Because I never had a problem with `lptheads`. But maybe I can rewrite this to use the CMake way of finding a threading library.

alexzk1, 2018-07-04:
If you google "cmake pthread" a lot will pop. Somewhere there I saw that -lpthreads is for llvm and -lpthread is for g++. So linker was failing to pass tests and find  -lpthreads. Arch itself is not too different. It's just rolling - always fresh + allows to install binaries(like ubuntu) or build sources (like gentoo) - your choice.

alexzk1, 2018-07-04:
In new header it has function
 /// Determine whether all fast-math-flags are set.
  bool isFast() const;
can it be replacement?

alexzk1, 2018-07-04:
https://github.com/JuliaLang/julia/issues/25869
found here, they did that replacement - unsafe to isFast

alexzk1, 2018-07-05:
On compiling test binary, while it do cpp-logger I think,  "ld failed to find -lpthreads"

alexzk1, 2018-07-05:
Right now I can't repeat that error. With you repo it goes till cpp file fails. However, I was missing llvm initially (or clang). Installed 1 of them later working on package.

alexzk1, 2018-07-05:
Ok, i figured. If I have already installed older package - it fails to build at different stages, for example after assembler is built. So need to cleanse system prior compiling.

doe300, 2018-07-05:
I updated the cpptest-lite module, so the pthread issue should no longer persist. Can you re-check?

alexzk1, 2018-07-05:
Ok, well, I used my latest PKGBUILD but pointed to your repo.

build() {
  cd ${srcdir}/VC4C
  cmake -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_DEBUG=OFF -DBUILD_DEB_PACKAGE=OFF
  make
}

package() {
  cd $srcdir/VC4C
  make DESTDIR="$pkgdir/" install
  mkdir -p $pkgdir/usr/include/vc4cl-stdlib
  install -m644 $srcdir/VC4CLStdLib/include/*.* $pkgdir/usr/include/vc4cl-stdlib
}

I had already installed compiler.
It built on x86_64(couple hours ago it was fail somewhere after assembler, if installed compiler already).
However precompiled headers were not generated again (bcs it found installed). And compiler has hard compiled in that pch again (that what u didnt accept). So now if I updated package it will cleanup that header....This situation with headers compiled into compiler is weird. I would suggest to redesign this part somehow.

alexzk1, 2018-07-05:
It seems you was right. I fixed something in code, but that was not correct place. Compiler still searching for precompiled header in temporary package build folder.

doe300, 2018-07-06:
> Compiler still searching for precompiled header in temporary package build folder.

If you "install" the std-lib headers first, you can point the `VC4CL_STDLIB_HEADER_SOURCE` CMake variable to the installed path and VC4C will look for them there.
This of course does not solve the pre-compilation issue, unless you run the generation command on install (e.g. in the `package()` step of your PKGBUILD).

closed 2018-07-07 by alexzk1