doe300/VC4C #117: OpenCL 2.2: C++ Support?

ax3l, 2018-10-03:
Hi and thank you for the great work!

Just out of curiosity:

With [OpenCL 2.2 C++ support landing in clang](https://www.phoronix.com/scan.php?page=news_item&px=OpenCL-CPP-Clang-Mainline-Begin), are you prospectively trying to support OpenCL 2.2 and maybe even C++ in kernels?

doe300, 2018-10-03:
Yes and no:)

I will try (although not with a high priority) to support OpenCL 2.x features where they can be (easily and efficiently) written for the VideoCore IV hardware.
But I cannot (in the foreseeable future) support any OpenCL 2.x standard as a whole, since it requires features I (for now) cannot implement, e.g.:
* Mandatory 64-bit support (at least `double` support, even in embedded profile)
* Starting kernels from kernel code
* Also will most of the new features not be supported on a while (e.g. pipes and SVM)

So officially OpenCL 2+ won't be supported, but if there are some small specific features someone asks for, I might add them.

Regarding C++ in kernels. As far as I can tell, this is mostly a CLang/LLVM thing, meaning that if the LLVM IR output does not differ from output for the "old" OpenCL C code, it will be supported out of the box (maybe some macros have to be added to actually be able to compile it). Again, if there are just some small things to be added to support OpenCL C++, I might do it.

ax3l, 2018-10-03:
Thank you for the detailed answer!

That sounds wonderful, as I do care more about the OpenCL C++ features as I do care about the 2.x features. Sounds wonderful! :)

closed 2018-10-03 by ax3l