doe300/VC4C #126: Bug of register allocation

nomaddo, 2018-12-06:
```
$ ./src/VC4C -O3 --quiet --asm -o /tmp/hoge.s  testing/test_int.cl 
```

```
[E] Thu Dec  6 19:39:42 2018: Validation-error 'Cannot read register ra2 because it just has been written by the previous instruction.' in: mul24 r0, rb0, ra2
[E] Thu Dec  6 19:39:42 2018: With reference to instruction: add ra2, ra3, r0
[E] Thu Dec  6 19:39:42 2018:  (1) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x121 [0x7fa64364c5dd]
[E] Thu Dec  6 19:39:42 2018:  (2) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x65579a [0x7fa64371279a]
[E] Thu Dec  6 19:39:42 2018:  (3) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x655ebf [0x7fa643712ebf]
[E] Thu Dec  6 19:39:42 2018:  (4) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : Validator::ProcessItem(Validator::state&)+0x2c62 [0x7fa6438cf3e2]
[E] Thu Dec  6 19:39:42 2018:  (5) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : Validator::Validate()+0x21f [0x7fa6438d087f]
[E] Thu Dec  6 19:39:42 2018:  (6) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : vc4c::qpu_asm::CodeGenerator::toMachineCode(vc4c::Method&)+0x17e [0x7fa6437129ce]
[E] Thu Dec  6 19:39:42 2018:  (7) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x5903dc [0x7fa64364d3dc]
[E] Thu Dec  6 19:39:42 2018:  (8) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x5911d8 [0x7fa64364e1d8]
[E] Thu Dec  6 19:39:42 2018:  (9) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : std::function<void (vc4c::Method* const&)>::operator()(vc4c::Method* const&) const+0x49 [0x7fa64365101d]
[E] Thu Dec  6 19:39:42 2018:  (10) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : void vc4c::BackgroundWorker::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)::{lambda()#1}::operator()() const+0xb3 [0x7fa64364fd33]
[E] Thu Dec  6 19:39:42 2018:  (11) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : std::_Function_handler<void (), void vc4c::BackgroundWorker::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)::{lambda()#1}>::_M_invoke(std::_Any_data const&)+0x20 [0x7fa6436538bb]
[E] Thu Dec  6 19:39:42 2018:  (12) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : std::function<void ()>::operator()() const+0x32 [0x7fa643644954]
[E] Thu Dec  6 19:39:42 2018:  (13) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x586e97 [0x7fa643643e97]
[E] Thu Dec  6 19:39:42 2018:  (14) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x587588 [0x7fa643644588]
[E] Thu Dec  6 19:39:42 2018:  (15) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x587525 [0x7fa643644525]
[E] Thu Dec  6 19:39:42 2018:  (16) /home/nomaddo/idein/VC4C/src/libVC4CC.so.1.2 : +0x587504 [0x7fa643644504]
[E] Thu Dec  6 19:39:42 2018:  (17) /usr/lib/x86_64-linux-gnu/libstdc++.so.6 : +0xb9e6f [0x7fa64018de6f]
[E] Thu Dec  6 19:39:42 2018:  (18) /lib/x86_64-linux-gnu/libpthread.so.0 : +0x7f2a [0x7fa643082f2a]
[E] Thu Dec  6 19:39:42 2018:  (19) /lib/x86_64-linux-gnu/libc.so.6 : clone+0x3f [0x7fa63fe5eedf]
[E] Thu Dec  6 19:39:42 2018: Background worker threw error: Verifier: vc4asm verification error: Warning V20.1: Cannot read register ra2 because it just has been written by the previous instruction.
  instruction at 0x250
  referring to instruction at 0x248
[E] Thu Dec  6 19:39:42 2018: While running worker task: CodeGenerator
[E] Thu Dec  6 19:39:42 2018: Compiler threw exception: Verifier: vc4asm verification error: Warning V20.1: Cannot read register ra2 because it just has been written by the previous instruction.
  instruction at 0x250
  referring to instruction at 0x248
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Verifier: vc4asm verification error: Warning V20.1: Cannot read register ra2 because it just has been written by the previous instruction.
  instruction at 0x250
  referring to instruction at 0x248
```

doe300, 2018-12-15:
I cannot reproduce the issue with the latest master (7968f6731adc3b85e1aa660403e9ad1bb1e9e891), can you re-check?

nomaddo, 2018-12-15:
With https://github.com/doe300/VC4C/commit/87417e47d8158266b265ef0dfa89fc3220249347, the problem reproduce.
With latest master (https://github.com/doe300/VC4C/commit/7968f6731adc3b85e1aa660403e9ad1bb1e9e891) , it does't.

Probably, the reason is that your commits change the instructions. I think the regsiter-allocator still has the bug.

doe300, 2018-12-15:
Yes, most likely I just suppressed it by changing the relevant code.

doe300, 2019-12-21:
The register allocation has been fixed several times since then. Given that this bug is no longer reproducible, I closed the ticket.

closed 2019-12-21 by doe300