doe300/VC4C #127: Undefined behaviors in TestVC4C

nomaddo, 2018-12-08:
The problem comes from cpptest-lite?

```
$ valgrind ./build/test/TestVC4C --regressions
==17084== Memcheck, a memory error detector
==17084== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==17084== Using Valgrind-3.12.0.SVN and LibVEX; rerun with -h for copyright info
==17084== Command: ./build/test/TestVC4C --regressions
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x89627DC: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x890165E: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x89015C1: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084== 
==17084== Invalid read of size 32
==17084==    at 0x89668E1: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x299B12: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x29B6FE: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x566559B: ??? (in /home/nomaddo/VC4C/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.0.9)
==17084==  Address 0x98a9050 is 0 bytes inside a block of size 20 alloc'd
==17084==    at 0x4A3221F: operator new(unsigned long) (vg_replace_malloc.c:334)
==17084==    by 0x56663B6: ??? (in /home/nomaddo/VC4C/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.0.9)
==17084==    by 0x29BE38: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x988751F: ???
==17084==    by 0x983083F: ???
==17084==    by 0x9830897: ???
==17084==    by 0xFFEFFF3CF: ???
==17084==    by 0x29B6FE: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0xFFEFFF3F7: ???
==17084==    by 0x988751F: ???
==17084==    by 0x56F81: ???
==17084==    by 0x983083F: ???
==17084== 
./example/fft2_2.cl
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x8877B27: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x3FFFFFFFFF: ???
==17084== 
==17084== Use of uninitialised value of size 8
==17084==    at 0x8877B37: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x3FFFFFFFFF: ???
==17084== 
==17084== Use of uninitialised value of size 8
==17084==    at 0x8877B90: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x3FFFFFFFFF: ???
==17084== 
==17084== Use of uninitialised value of size 8
==17084==    at 0x8877C47: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0xC1FFFFFFFF: ???
==17084== 
==17084== Use of uninitialised value of size 8
==17084==    at 0x8877CF4: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0xC1FFFFFFFF: ???
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x8955278: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x546530E: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x51ED304: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x280463: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x299B12: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x29B6FE: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x566559B: ??? (in /home/nomaddo/VC4C/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.0.9)
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x8966752: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x546530E: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x51ED304: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x280463: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x299B12: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x29B6FE: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x566559B: ??? (in /home/nomaddo/VC4C/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.0.9)
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x89668A4: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x546530E: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x51ED304: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x280463: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x299B12: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x29B6FE: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x566559B: ??? (in /home/nomaddo/VC4C/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.0.9)
==17084== 
==17084== Conditional jump or move depends on uninitialised value(s)
==17084==    at 0x8901C21: ??? (in /lib/x86_64-linux-gnu/libc-2.27.so)
==17084==    by 0x546530E: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x51ED304: ??? (in /home/nomaddo/VC4C/build/src/libVC4CC.so.0.4)
==17084==    by 0x280463: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x299B12: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x29B6FE: ??? (in /home/nomaddo/VC4C/build/test/TestVC4C)
==17084==    by 0x566559B: ??? (in /home/nomaddo/VC4C/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.0.9)
==17084== 

... and more warnings ...
```

nomaddo, 2018-12-08:
Maybe the problem of my environment...

closed 2018-12-08 by nomaddo

doe300, 2018-12-08:
I didn't check all the regression tests, but for some simpler tests, I don't get a single warning from valgrind. Did you update the cpptest sub-module to the latest version?