doe300/VC4C #131: CompileError -  (c++: error: LLVMCore: No such file or directory)

abhiTronix, 2019-01-19:
```
[ 82%] Linking CXX shared library libVC4CC.so
c++: error: LLVMCore: No such file or directory
src/CMakeFiles/VC4CC.dir/build.make:1923: recipe for target 'src/libVC4CC.so.0.4' failed
make[2]: *** [src/libVC4CC.so.0.4] Error 1
CMakeFiles/Makefile2:513: recipe for target 'src/CMakeFiles/VC4CC.dir/all' failed
make[1]: *** [src/CMakeFiles/VC4CC.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
```
I'm getting this error no matter what cmake configration I try. I'm building this library on my Raspberry Pi 3 B(Stretch-Latest) with Clang 6.0.0 and similiar error with clang 7.0.0? Any help?

abhiTronix, 2019-01-19:
Solved by installing `llvm-4.0` and `clang-4.0` from official Raspberry Pi Repo.

ghostop14, 2019-11-27:
Did you ever find a solution to this issue?  I'm encountering the same build issue.

doe300, 2019-11-27:
@ghostop14, you might need to install the proper LLVM development package. What version of LLVM/Clang do you have installed?

ghostop14, 2019-11-27:
Hi doe300, I'm running on an RPi3 with the latest Raspbian all updated.  I installed llvm and clang from the repo, but looks like llvm 3.8 and 3.9 are both installed, same with clang... 3.8 and 3.9.  Think forcably removing 3.8 might do the trick?

ghostop14, 2019-11-27:
Just to be sure, I did uninstall the 3.8 versions.  Here's more detail on the failure:

[ 12%] Performing build step for 'variant-project'
[ 14%] Built target variant-project-build
[ 14%] Linking CXX shared library libVC4CC.so
c++: error: LLVMCore: No such file or directory
src/CMakeFiles/VC4CC.dir/build.make:2233: recipe for target 'src/libVC4CC.so.0.4.9999' failed
make[2]: *** [src/libVC4CC.so.0.4.9999] Error 1
CMakeFiles/Makefile2:415: recipe for target 'src/CMakeFiles/VC4CC.dir/all' failed
make[1]: *** [src/CMakeFiles/VC4CC.dir/all] Error 2
Makefile:160: recipe for target 'all' failed

Do you have a quick apt install line for Raspbian to make sure all the prereq's are correct? (I think I have everything but I could be missing something simple).  Here's what I have:
apt install clang-3.9 clang-format-3.9 libclang-3.9-dev llvm-3.9 llvm-3.9-dev

ghostop14, 2019-11-27:
Got it.  I had both the apt llvm installed and the Khronos versions.  The Khronos was 3.6 which was causing the issue.  For the requirements section in the readme, I think the ", and Khronos" was confusing in the compiler section.  I originally took it to mean both were required.  Maybe suggest the readme do them as a number list like 1.) or 2.) ?

Thanks!

doe300, 2019-11-27:
Glad you found the issue. You are right, they should be either-or.

commit a405b941 (2019-12-07): Small fixes

- Reverts removal of SPIRV-Tools-opt target
- Handles ODR violations when linking multiple LLVM modules with conflicting VC4CL std-lib function definitions
- Improves linking test
- Improves documentation, closes #131
closed 2019-12-07 by doe300