doe300/VC4C #132: llvm-config.cmake not found

as3ii, 2019-01-24:
I'm trying to compile it on arch linux arm on aarch64 (raspberry pi 3). 
Here the output of cmake:

```

-- VC4CL standard library headers found: /home/as3ii/VC4C/../VC4CLStdLib/include/VC4CLStdLib.h
-- Enabling multi-threaded optimizations
-- CLang compiler found: /usr/bin/clang
-- Using enhanced OpenCL features of Clang 3.9+!
CMake Warning at CMakeLists.txt:245 (find_package):
  By not providing "FindLLVM.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LLVM", but
  CMake did not find one.

  Could not find a package configuration file provided by "LLVM" with any of
  the following names:

    LLVMConfig.cmake
    llvm-config.cmake

  Add the installation prefix of "LLVM" to CMAKE_PREFIX_PATH or set
  "LLVM_DIR" to a directory containing one of the above files.  If "LLVM"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Warning at CMakeLists.txt:269 (message):
  LLVM library front-end enabled, but LLVM library was not found!

CMake Warning at CMakeLists.txt:274 (message):
   Neither SPIR-V nor LLVM library front-end are configured!

-- found clang-format: /usr/bin/clang-format
-- build deb package...
-- Debian package expects VC4CL standard library PCH to be located in: /home/as3ii/VC4C/../VC4CLStdLib/include/VC4CLStdLib.h.pch
-- Configuring done
-- Generating done
-- Build files have been written to: /home/as3ii/VC4C

```

doe300, 2019-01-25:
Do you have `llvm-dev` or any specific version of it (e.g. `llvm-3.9-dev`) installed? These packages should provide the `LLVMConfig.cmake` file.

as3ii, 2019-01-25:
> 
> 
> Do you have `llvm-dev` or any specific version of it (e.g. `llvm-3.9-dev`) installed? These packages should provide the `LLVMConfig.cmake` file.

only llvm-libs 7.0.1-2 which was installed with clang  7.0.1-1 and contains the LLVM runtime libraries

doe300, 2019-01-25:
Then you need to install the corresponding `llvm*-dev` package (`llvm-7-dev` I think)

as3ii, 2019-01-25:
ok, I found the right package, thanks

closed 2019-01-25 by as3ii