doe300/VC4C #137: Compilation ERROR, libLLVM-3.9 missing

MaierF87, 2019-08-05:
Hey, 
tried to compile the fibonacci example with the following command 
`VC4C --llvm --hex -o fibonacci.hex fibonacci.cl`

leading into 
`VC4C: error while loading shared libraries: libLLVM-3.9.so.1: cannot open shared object file: No such file or directory`

so I searching the webs on how to get that library and found an awnser on https://apt.llvm.org/

which basicly means that i added following lines to /etc/apt/sources.list

```
deb http://apt.llvm.org/stretch/ llvm-toolchain-stretch main
deb-src http://apt.llvm.org/stretch/ llvm-toolchain-stretch main
# 7 
deb http://apt.llvm.org/stretch/ llvm-toolchain-stretch-7 main
deb-src http://apt.llvm.org/stretch/ llvm-toolchain-stretch-7 main
# 8 
deb http://apt.llvm.org/stretch/ llvm-toolchain-stretch-8 main
deb-src http://apt.llvm.org/stretch/ llvm-toolchain-stretch-8 main
```

then i ran 
`sudo apt -o Acquire::AllowInsecureRepositories=true update`
with following output
`Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Get:3 http://apt.llvm.org/stretch llvm-toolchain-stretch InRelease [4,225 B]
Get:4 http://apt.llvm.org/stretch llvm-toolchain-stretch-7 InRelease [4,221 B]
Get:5 http://apt.llvm.org/stretch llvm-toolchain-stretch-8 InRelease [4,235 B]
Get:6 http://apt.llvm.org/stretch llvm-toolchain-stretch/main Sources [3,018 B]
Get:7 http://apt.llvm.org/stretch llvm-toolchain-stretch-7/main Sources [2,499 B]
Get:8 http://apt.llvm.org/stretch llvm-toolchain-stretch-8/main Sources [2,515 B]
Fetched 20.7 kB in 2s (9,088 B/s)  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://apt.llvm.org/stretch llvm-toolchain-stretch InRelease' doesn't support architecture 'armhf'
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://apt.llvm.org/stretch llvm-toolchain-stretch-7 InRelease' doesn't support architecture 'armhf'
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://apt.llvm.org/stretch llvm-toolchain-stretch-8 InRelease' doesn't support architecture 'armhf'
`

seems like I cant get libLLVM ....

if it helps here is my clinfo:

`
sudo clinfo
Number of platforms                               1
  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
  Platform Vendor                                 doe300
  Platform Version                                OpenCL 1.2 VC4CL 0.4.9999
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_khr_spir cl_khr_create_command_queue cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU
  Device Vendor                                   Broadcom
  Device Vendor ID                                0xa5c
  Device Version                                  OpenCL 1.2 VC4CL 0.4.9999
  Driver Version                                  0.4.9999
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  EMBEDDED_PROFILE
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                No
[... 59 lines cut ...]
  Device Extensions                               cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_nv_pragma_unroll cl_arm_core_id cl_ext_atomic_counters_32 cl_khr_initialize_memory cl_arm_integer_dot_product_int8 cl_arm_integer_dot_product_accumulate_int8 cl_arm_integer_dot_product_accumulate_int16 cl_arm_integer_dot_product_accumulate_saturate_int8 cl_khr_il_program cl_khr_spir cl_khr_create_command_queue cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  OpenCL for the Raspberry Pi VideoCore IV GPU
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [VC4CL]
  clCreateContext(NULL, ...) [default]            Success [VC4CL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_DEFAULT)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.12
  ICD loader Profile                              OpenCL 2.2
`

doe300, 2019-08-05:
From the error, I assume you run Raspbian buster?
Is llvm-3.9 in the buster repository?
If not, which LLVM version is in the standard Raspbian buster repository?
When you re-run the configuration step of the VC4C cmake step, which version of LLVM is listed there?

MaierF87, 2019-08-05:
Thanks for the fast reply!

Yes, im using raspbian buster, seems like buster comes with llvm-7.0.1

rerunning the cmake step for VC4C resulted in
`~/VC4C $ cmake -DSPIRV_FRONTEND=OFF .
-- VC4CL standard library headers found: /home/pi/VC4C/../VC4CLStdLib/include/
-- Enabling multi-threaded optimizations
-- CLang compiler found: /usr/bin/clang
-- Using enhanced OpenCL features of Clang 3.9+!
-- Compiling LLVM library front-end with LLVM in version 7.0.1 located in '/usr/lib/llvm-7/lib'
CMake Warning at src/CMakeLists.txt:192 (message):
  clang-format not found: strongly recommend to use it before commit!

-- build deb package...
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/VC4C`

doe300, 2019-08-05:
Okay, so it finds LLVM 7.0 which looks correct. But why it links against version 3.9 I don't know. Especially, since it does not complain at compile time...
What is the output of `ldd vc4c` and `ldd libVC4CC.so`?

MaierF87, 2019-08-05:
No such file or directory
in both cases

doe300, 2019-08-05:
You probably have to specify the whole path, e.g. `ldd <vc4c project root>/build/src/vc4c` and `ldd <vc4c project root>/build/src/libVC4CC.so`.

MaierF87, 2019-08-05:
youre right, sorry im new to linux 

here are the results:
```

ldd /home/pi/VC4C/src/vc4c
	linux-vdso.so.1 (0x7eb28000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f3d000)
	libVC4CC.so.1.2 => /home/pi/VC4C/src/libVC4CC.so.1.2 (0x7659f000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76546000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76533000)
	libLLVM-7.so.1 => /usr/lib/arm-linux-gnueabihf/libLLVM-7.so.1 (0x732a5000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x7315e000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x730dc000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x730af000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x72f61000)
	/lib/ld-linux-armhf.so.3 (0x76f52000)
	libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0x72f48000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x72f30000)
	libedit.so.2 => /usr/lib/arm-linux-gnueabihf/libedit.so.2 (0x72ef8000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x72ecd000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x72eb6000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0x72e85000)
	libxml2.so.2 => /usr/lib/arm-linux-gnueabihf/libxml2.so.2 (0x72d09000)
	libbsd.so.0 => /usr/lib/arm-linux-gnueabihf/libbsd.so.0 (0x72ce1000)
	libicui18n.so.63 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.63 (0x72a93000)
	libicuuc.so.63 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.63 (0x7290d000)
	libicudata.so.63 => /usr/lib/arm-linux-gnueabihf/libicudata.so.63 (0x70f0f000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x70ede000)
pi@raspberrypi:~ $ ldd /home/pi/VC4C/src/libVC4CC.so
	linux-vdso.so.1 (0x7eb6c000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76556000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x764fd000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x764ea000)
	libLLVM-7.so.1 => /usr/lib/arm-linux-gnueabihf/libLLVM-7.so.1 (0x7325c000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x73115000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x73093000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x73066000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x72f18000)
	/lib/ld-linux-armhf.so.3 (0x76f09000)
	libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0x72eff000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x72ee7000)
	libedit.so.2 => /usr/lib/arm-linux-gnueabihf/libedit.so.2 (0x72eaf000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x72e84000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x72e6d000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0x72e3c000)
	libxml2.so.2 => /usr/lib/arm-linux-gnueabihf/libxml2.so.2 (0x72cc0000)
	libbsd.so.0 => /usr/lib/arm-linux-gnueabihf/libbsd.so.0 (0x72c98000)
	libicui18n.so.63 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.63 (0x72a4a000)
	libicuuc.so.63 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.63 (0x728c4000)
	libicudata.so.63 => /usr/lib/arm-linux-gnueabihf/libicudata.so.63 (0x70ec6000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x70e95000)
```

doe300, 2019-08-05:
Ok, that is very confusing:
- there are a lot of libraries I don't know what they do, e.g. libicuuc, libicudata
- there is no mention of the libLLVM-3.9.so.1 that is reported missing

What does `/usr/bin/clang --version` report? And `ldd /usr/bin/clang`?

MaierF87, 2019-08-05:
/usr/bin/clang --version
clang version 7.0.1-8+rpi2 (tags/RELEASE_701/final)
Target: armv6k-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin

pi@raspberrypi:~ $ ldd /usr/bin/clang
	linux-vdso.so.1 (0x7eef3000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f70000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76f17000)
	libLLVM-7.so.1 => /usr/lib/arm-linux-gnueabihf/libLLVM-7.so.1 (0x73c89000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x73b42000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x73ac0000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x73a93000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x73945000)
	/lib/ld-linux-armhf.so.3 (0x76f85000)
	libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0x7392c000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x73914000)
	libedit.so.2 => /usr/lib/arm-linux-gnueabihf/libedit.so.2 (0x738dc000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x738b1000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x7389a000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x73887000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0x73856000)
	libxml2.so.2 => /usr/lib/arm-linux-gnueabihf/libxml2.so.2 (0x736da000)
	libbsd.so.0 => /usr/lib/arm-linux-gnueabihf/libbsd.so.0 (0x736b2000)
	libicui18n.so.63 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.63 (0x73464000)
	libicuuc.so.63 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.63 (0x732de000)
	libicudata.so.63 => /usr/lib/arm-linux-gnueabihf/libicudata.so.63 (0x718e0000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x718af000)

doe300, 2019-08-05:
So, one last question, then I am out of ideas:

Did you update your OS from a Raspbian stretch or did you install a fresh Raspbian buster?

MaierF87, 2019-08-05:
installed a fresh raspbian buster, Release date: 2019-07-10

doe300, 2019-08-05:
Okay, then sorry, I have no idea why it tried to find libLLVM-3.9.so ...

MaierF87, 2019-08-05:
guess ill try a fresh install on another sd-card, maybe that magically solves the problem
thanks for your time tho :)

MaierF87, 2019-08-05:
So finished installing and... it seems the current version of VC4CL does not support raspbian bust?

followed the same install guide and trying the same fibonacci example. this resulted in:
`VC4C --llvm --hex -o fibonacci.hex fibonacci.cl
bash: VC4C: command not found`

so I tried using your installation guide stopping at VC4C with 
```
sudo dpkg -i /tmp/vc4c.deb
Selecting previously unselected package vc4c.
(Reading database ... 160397 files and directories currently installed.)
Preparing to unpack /tmp/vc4c.deb ...
Unpacking vc4c (0.4.1514-2019-08-04) ...
dpkg: dependency problems prevent configuration of vc4c:
 vc4c depends on clang-3.9; however:
  Package clang-3.9 is not installed.
 vc4c depends on llvm-3.9; however:
  Package llvm-3.9 is not installed.
 vc4c depends on llvm-3.9-dev; however:
  Package llvm-3.9-dev is not installed.

dpkg: error processing package vc4c (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 vc4c
```
so I tried to install clang using
`sudo apt-get install clang

Reading package lists... Done
Building dependency tree       
Reading state information... Done
clang is already the newest version (1:7.0-47).
clang set to manually installed.
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 vc4c : Depends: clang-3.9 but it is not installable
        Depends: llvm-3.9 but it is not installable
        Depends: llvm-3.9-dev but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).`

`pi@raspberrypi:~ $ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  vc4c
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 175 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 160415 files and directories currently installed.)
Removing vc4c (0.4.1514-2019-08-04) ...
dpkg: warning: while removing vc4c, directory '/usr/local/share' not empty so not removed
dpkg: warning: while removing vc4c, directory '/usr/local/lib' not empty so not removed`

and here are the steps we tried last time, it seems the same unknown libraries were installed again, on a completely fresh install no other programs were installed (only the preinstalled ones):

`pi@raspberrypi:~/VC4CL-test $ ldd ~/VC4C/src/vc4c
	linux-vdso.so.1 (0x7efcf000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f88000)
	libVC4CC.so.1.2 => /home/pi/VC4C/src/libVC4CC.so.1.2 (0x765a2000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76560000)
	libLLVM-7.so.1 => /usr/lib/arm-linux-gnueabihf/libLLVM-7.so.1 (0x732d2000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x7318b000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x73109000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x730dc000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x72f8e000)
	/lib/ld-linux-armhf.so.3 (0x76f9d000)
[... 44 lines cut ...]
pi@raspberrypi:~/VC4CL-test $ ldd /usr/bin/clang
	linux-vdso.so.1 (0x7eec1000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f11000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ecf000)
	libLLVM-7.so.1 => /usr/lib/arm-linux-gnueabihf/libLLVM-7.so.1 (0x73c41000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x73afa000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x73a78000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x73a4b000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x738fd000)
	/lib/ld-linux-armhf.so.3 (0x76f26000)
	libatomic.so.1 => /usr/lib/arm-linux-gnueabihf/libatomic.so.1 (0x738e4000)
	libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0x738cc000)
	libedit.so.2 => /usr/lib/arm-linux-gnueabihf/libedit.so.2 (0x73894000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x73869000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x73852000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x7383f000)
	libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0x7380e000)
	libxml2.so.2 => /usr/lib/arm-linux-gnueabihf/libxml2.so.2 (0x73692000)
	libbsd.so.0 => /usr/lib/arm-linux-gnueabihf/libbsd.so.0 (0x7366a000)
	libicui18n.so.63 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.63 (0x7341c000)
	libicuuc.so.63 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.63 (0x73296000)
	libicudata.so.63 => /usr/lib/arm-linux-gnueabihf/libicudata.so.63 (0x71898000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x71867000)`

maybe this helps

doe300, 2019-08-06:
> it seems the current version of VC4CL does not support raspbian buster?

Yes and no.
- The prebuilt debian packages at the moment do not support Raspbian buster yet, since it upgrades the preinstalled LLVM version. I created #138 to track this.
- When building manually, VC4C should support any LLVM version (up to version 7 at least).

closed 2020-03-05 by doe300