doe300/VC4C #139: Use of unsupported SPIR-V extension

waayaa, 2019-11-14:
How do I fix this?
```
hashcat (v5.1.0-1428-g11c8a404) starting...

You have enabled --force to bypass dangerous warnings and errors!
This can hide serious problems and should only be done when debugging.
Do not report hashcat issues encountered when using --force.
OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 76/76 MB allocatable, 1MCU

Hashes: 1 digests; 1 unique digests, 1 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates
Rules: 1

Applicable optimizers:
* Zero-Byte
* Single-Hash
* Single-Salt
* Slow-Hash-SIMD-LOOP

Minimum password length supported by kernel: 8
Maximum password length supported by kernel: 63

Watchdog: Hardware monitoring interface not found on your system.
Watchdog: Temperature abort trigger disabled.

* Device #1: build_options '-D KERNEL_STATIC -I OpenCL -I /usr/local/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D HAS_VPERM=0 -D HAS_VADD3=0 -D HAS_VBFE=0 -D HAS_BFE=0 -D HAS_LOP3=0 -D HAS_MOV64=0 -D HAS_PRMT=0 -D VECT_SIZE=1 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=1 -D DGST_R2=2 -D DGST_R3=3 -D DGST_ELEM=4 -D KERN_TYPE=16800 -D ATTACK_EXEC=10 -D ATTACK_KERN=0 -D _unroll '
* Device #1: build_options_module '-D KERNEL_STATIC -I OpenCL -I /usr/local/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D HAS_VPERM=0 -D HAS_VADD3=0 -D HAS_VBFE=0 -D HAS_BFE=0 -D HAS_LOP3=0 -D HAS_MOV64=0 -D HAS_PRMT=0 -D VECT_SIZE=1 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=1 -D DGST_R2=2 -D DGST_R3=3 -D DGST_ELEM=4 -D KERN_TYPE=16800 -D ATTACK_EXEC=10 -D ATTACK_KERN=0 -D _unroll  '
* Device #1: Kernel m16800-pure.bf3b99b0.kernel not found in cache! Building may take a while...
clBuildProgram(): CL_BUILD_PROGRAM_FAILURE

[E] Wed Nov 13 17:51:59 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xdc [0x7f9f63c6f4]
[E] Wed Nov 13 17:51:59 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::spirv2qasm::SPIRVParser::parseInstruction(spv_parsed_instruction_t const*)+0x204 [0x7f9f95c74c]
[E] Wed Nov 13 17:51:59 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : +0xfea884 [0x7f9f959884]
[E] Wed Nov 13 17:51:59 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : +0x1050d74 [0x7f9f9bfd74]
[E] Wed Nov 13 17:51:59 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : +0x1050608 [0x7f9f9bf608]
[E] Wed Nov 13 17:51:59 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : +0x10502fc [0x7f9f9bf2fc]
[E] Wed Nov 13 17:51:59 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : spvBinaryParse+0x84 [0x7f9f9c1208]
[E] Wed Nov 13 17:51:59 2019:  (8) /usr/local/lib/libVC4CC.so.1.2 : vc4c::spirv2qasm::SPIRVParser::parse(vc4c::Module&)+0x228 [0x7f9f959e08]
[E] Wed Nov 13 17:51:59 2019:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::convert()+0xe8 [0x7f9f63d4fc]
[E] Wed Nov 13 17:51:59 2019:  (10) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x1c4 [0x7f9f63dbf4]
[E] Wed Nov 13 17:51:59 2019:  (11) /usr/local/lib/libVC4CL.so : +0x139ac4 [0x7fa02d7ac4]
[E] Wed Nov 13 17:51:59 2019:  (12) /usr/local/lib/libVC4CL.so : vc4cl::Program::link(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<vc4cl::object_wrapper<vc4cl::Program>, std::allocator<vc4cl::object_wrapper<vc4cl::Program> > > const&)+0x110 [0x7fa02d7e10]
[E] Wed Nov 13 17:51:59 2019:  (13) /usr/local/lib/libVC4CL.so : +0x13af58 [0x7fa02d8f58]
[E] Wed Nov 13 17:51:59 2019:  (14) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x3d0 [0x7fa02da814]
[E] Wed Nov 13 17:51:59 2019:  (15) hashcat : +0x40030 [0x558d585030]
[E] Wed Nov 13 17:51:59 2019:  (16) hashcat : +0x485f0 [0x558d58d5f0]
[E] Wed Nov 13 17:51:59 2019:  (17) hashcat : +0x903c [0x558d54e03c]
[E] Wed Nov 13 17:51:59 2019:  (18) hashcat : +0x95fc [0x558d54e5fc]
[E] Wed Nov 13 17:51:59 2019:  (19) hashcat : +0x7104 [0x558d54c104]
[E] Wed Nov 13 17:51:59 2019:  (20) /lib/aarch64-linux-gnu/libc.so.6 : __libc_start_main+0xe4 [0x7fa6fe62ac]
[E] Wed Nov 13 17:51:59 2019:  (21) hashcat : +0x5614 [0x558d54a614]
[E] Wed Nov 13 17:51:59 2019: Compiler threw exception: Parser: Use of unsupported SPIR-V extension: SPV_KHR_no_integer_wrap_decoration

* Device #1: Kernel /usr/local/share/hashcat/OpenCL/m16800-pure.cl build failed - proceeding without this device.

Started: Wed Nov 13 17:46:49 2019
Stopped: Wed Nov 13 17:51:59 2019
terminate called without an active exception
Aborted

```

doe300, 2019-11-14:
There is not much that you can do.

I will have a look at implementing the necessary missing extension [SPV_KHR_no_integer_wrap_decoration](http://htmlpreview.github.io/?https://github.com/KhronosGroup/SPIRV-Registry/blob/master/extensions/KHR/SPV_KHR_no_integer_wrap_decoration.html), which looks like it should be very easily doable .

waayaa, 2019-11-14:
I can contribute to implement the fix if you give me a pointer to the code. I guess it will be time consuming to find the specific detail without your input ;)

doe300, 2019-11-14:
Of course, input is always welcome.

Where you might want to start:
- Add support for the `SPV_KHR_no_integer_wrap_decoration` extension in https://github.com/doe300/VC4C/blob/master/src/spirv/SPIRVParser.cpp#L507, e.g. by just not throwing the exception, we don't need to handle the extension otherwise for now.
- In https://github.com/doe300/VC4C/blob/master/src/spirv/SPIRVParser.cpp#L289 add support for the `NoSignedWrap` and `NoUnsignedWrap` decorations to the first block of decorations.
- In https://github.com/doe300/VC4C/blob/master/src/spirv/SPIRVParser.cpp#L268, similar to how the `SaturatedConversion` is handled, set the `intermediate::InstructionDecorations::SIGNED_OVERFLOW_IS_UB` and `intermediate::InstructionDecorations::UNSIGNED_OVERFLOW_IS_UB` for the two now decorations.

Since the corresponding InstructionDecorations already exist for the decorations (e.g. for the LLVM front-end), only the conversion has to be added to the front-end.

waayaa, 2019-11-15:
I have done as you said. 
1. Still on the command I presented earlier I am now getting a segmentation fault.
2. For the llvm spirv modules/dependencies and some modules/dependencies not part of your gits I use pre installed binary. Will this affect the process? If so I probably have to do a clean and do from scratch as you instructed.
```
Thread 8 "Normalization" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fa8bfb1e0 (LWP 6720)]
0x0000007fb809a4e4 in mpark::detail::base<(mpark::detail::Trait)1, std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal>::valueless_by_exception (this=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1055
1055	        return index_ == static_cast<index_t<Ts...>>(-1);
(gdb) backtrace
#0  0x0000007fb809a4e4 in mpark::detail::base<(mpark::detail::Trait)1, std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal>::valueless_by_exception (this=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1055
#1  0x0000007fb809a510 in mpark::detail::base<(mpark::detail::Trait)1, std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal>::index (this=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1059
#2  0x0000007fb80cfc08 in mpark::variant<std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal>::index (this=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1719
#3  0x0000007fb80cfd5c in mpark::holds_alternative<1ul, std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal> (v=...)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1745
#4  0x0000007fb80cfd90 in mpark::detail::generic_get_if<1ul, mpark::variant<std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal> const> (v=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1818
#5  0x0000007fb80cfdcc in mpark::get_if<1ul, std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal> (v=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1834
#6  0x0000007fb80cfa4c in mpark::get_if<vc4c::Literal, std::vector<vc4c::CompoundConstant, std::allocator<vc4c::CompoundConstant> >, vc4c::Literal> (v=0x78)
    at /root/Downloads/VC4C/build/variant/src/variant-project/include/mpark/variant.hpp:1846
#7  0x0000007fb80cde1c in vc4c::CompoundConstant::getScalar (this=0x70)
    at /root/Downloads/VC4C/src/GlobalValues.cpp:67
#8  0x0000007fb82fcc78 in vc4c::normalization::getConstantValue (source=...) at /root/Downloads/VC4C/src/normalization/MemoryMapChecks.cpp:49
#9  0x0000007fb8300ef0 in canLowerToRegisterReadOnly (method=..., baseAddr=0x55594ad260, access=...) at /root/Downloads/VC4C/src/normalization/MemoryMapChecks.cpp:648
#10 0x0000007fb82fcc20 in vc4c::normalization::checkMemoryMapping (method=..., baseAddr=0x55594ad260, access=...) at /root/Downloads/VC4C/src/normalization/MemoryMapChecks.cpp:42
#11 0x0000007fb82f19d4 in vc4c::normalization::mapMemoryAccess (module=..., method=..., config=...) at /root/Downloads/VC4C/src/normalization/MemoryAccess.cpp:831
#12 0x0000007fb831f424 in vc4c::normalization::Normalizer::normalizeMethod (this=0x7fffffd1a0, module=..., method=...) at /root/Downloads/VC4C/src/normalization/Normalizer.cpp:255
#13 0x0000007fb831e670 in vc4c::normalization::Normalizer::<lambda(vc4c::Method*)>::operator()(vc4c::Method *) const (__closure=0x7fffffce18, kernelFunc=0x55594a9df0) at /root/Downloads/VC4C/src/normalization/Normalizer.cpp:219
#14 0x0000007fb8320bf0 in std::_Function_handler<void(vc4c::Method* const&), vc4c::normalization::Normalizer::normalize(vc4c::Module&) const::<lambda(vc4c::Method*)> >::_M_invoke(const std::_Any_data &, vc4c::Method * const&) (__functor=..., __args#0=@0x55561d5e60: 0x55594a9df0) at /usr/include/c++/9/bits/std_function.h:300
#15 0x0000007fb809b3e4 in std::function<void (vc4c::Method* const&)>::operator()(vc4c::Method* const&) const (this=0x7fffffce18, __args#0=@0x55561d5e60: 0x55594a9df0) at /usr/include/c++/9/bits/std_function.h:688
#16 0x0000007fb809990c in vc4c::ThreadPool::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&)::{lambda()#1}::operator()() const (this=0x5568478348) at /root/Downloads/VC4C/src/ThreadPool.h:39
#17 0x0000007fb809c9d8 in std::_Function_handler<void (), vc4c::ThreadPool::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&)::{lambda()#1}>::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/9/bits/std_function.h:300
#18 0x0000007fb80f2bc8 in std::function<void ()>::operator()() const (this=0x5568478348) at /usr/include/c++/9/bits/std_function.h:688
#19 0x0000007fb80f2824 in std::__invoke_impl<void, std::function<void ()>&>(std::__invoke_other, std::function<void ()>&) (__f=...) at /usr/include/c++/9/bits/invoke.h:60
[... cut ...]
#42 0x0000007fb80eb6dc in std::thread::_Invoker<std::tuple<vc4c::ThreadPool::ThreadPool(const string&, unsigned int)::<lambda()> > >::operator()(void) (this=0x55596486e8) at /usr/include/c++/9/thread:251
#43 0x0000007fb80eb6bc in std::thread::_State_impl<std::thread::_Invoker<std::tuple<vc4c::ThreadPool::ThreadPool(const string&, unsigned int)::<lambda()> > > >::_M_run(void) (this=0x55596486e0) at /usr/include/c++/9/thread:195
#44 0x0000007fb789773c in ?? () from /lib/aarch64-linux-gnu/libstdc++.so.6
#45 0x0000007fbf699888 in start_thread (arg=0x7fffffc8ef) at pthread_create.c:486
#46 0x0000007fbf53770c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```

doe300, 2019-11-16:
> For the llvm spirv modules/dependencies and some modules/dependencies not part of your gits I use pre installed binary.

That is completely okay.

commit d90472ba (2019-11-27): Adds support for the SPV_KHR_no_integer_wrap_decoration extension

Fixes #139
closed 2019-11-27 by doe300