doe300/VC4C #143: Running Tensorflow model via TVM on OpenCL

marcin-sielski, 2020-04-20:
I am experiencing following error on Rasberry Pi while trying to run tensorflow model over TVM on OpenCL.
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Normalizer: Invalid local type for memory area: (p) i32* %tmp.3262

doe300, 2020-04-21:
Can you give me more details about what were you trying to do?

- What tensorflow sources did you build?
- What command did you run?
- If possible, please post any log previous to that error.

marcin-sielski, 2020-04-21:
Hi @doe300,

My apologies but I am new to the OpenCL and the whole TVM stack. I will try to explain in more details what I did.
I have compiled VC4C, VC4CL, VC4CLStdLib, TVM (full package) and installed it on Raspberry Pi together with Tensorflow and tried to follow the tutorial at:
https://opensourceforu.com/2019/06/the-capabilities-of-tensor-virtual-machine-an-open-deep-learning-compiler-stack/
The tutorial has some drawbacks but I was able to walk through most of the example:
1. I have imported and compiled the mobilenet into relay representation for "opencl" (example uses llvm) - Step 3
2. I have saved the compiler output - Step 4
3. I have executed successfully most of the steps from Step 5  
While trying to execute the model:
```
# Execute the Model
module.run()
# Get the first output
out = module.get_output(0)
```
I get 
```
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Normalizer: Invalid local type for memory area: (p) i32* %tmp.326
```
Thank you for help.

Best Regards

Marcin Sielski

marcin-sielski, 2020-05-22:
Hi @doe300,

Do you have any ideas how to troubleshoot the issue?

I wonder what the message 'Invalid local type for memory area: (p) i32* %tmp.326' mean.

Thank you for help in advance.

Best Regards

Marcin Sielski

doe300, 2020-05-23:
The error means that the OpenCL kernel accesses memory in a way that is not yet supported by the VC4C compiler.

Without knowing the actual kernel code, there is not much that can be done here. If you compile the VC4CL host-side stack in debug mode (with the CMake variable `BUILD_DEBUG` set), the kernel code which fails to compile will be dumped into a file (look for lines in the standard output like "Dumping program sources to (tmp/vc4c-source-xxx.cl") which the could be used to analyze the problem.

marcin-sielski, 2020-06-28:
@doe300 
I finally managed to generate kernel code. I am not sure where too look at:

[vc4cl-source-1804289383.cl.tar.gz](https://github.com/doe300/VC4C/files/4841806/vc4cl-source-1804289383.cl.tar.gz)

Thank you for help in advance.

Best Regards

Marcin Sielski

doe300, 2020-06-28:
Thanks for the source code.

I do not get the error reported above, but I do get some register-association errors. I will have a look into them.