doe300/VC4C #147: ‘LOGGER’ is not a member of ‘logging’

ipapadop, 2020-08-12:
Happens after cb1a0923d25dbaad636cb4b91ecb3687c1f0e29e

The following solves it
```
diff --git a/src/Compiler.cpp b/src/Compiler.cpp
index 9c51fcf..505f767 100644
--- a/src/Compiler.cpp
+++ b/src/Compiler.cpp
@@ -173,12 +173,12 @@ std::size_t Compiler::compile(std::istream& input, std::ostream& output, const C
     }
 }
 
-std::unique_ptr<logging::Logger> logging::LOGGER(new logging::ColoredLogger(std::wcout, logging::Level::WARNING));
+std::unique_ptr<logging::Logger> logging::DEFAULT_LOGGER(new logging::ColoredLogger(std::wcout, logging::Level::WARNING));
 
 void vc4c::setLogger(std::wostream& outputStream, const bool coloredOutput, const LogLevel level)
 {
     if(coloredOutput)
-        logging::LOGGER = std::make_unique<logging::ColoredLogger>(outputStream, static_cast<logging::Level>(level));
+        logging::DEFAULT_LOGGER = std::make_unique<logging::ColoredLogger>(outputStream, static_cast<logging::Level>(level));
     else
-        logging::LOGGER = std::make_unique<logging::StreamLogger>(outputStream, static_cast<logging::Level>(level));
+        logging::DEFAULT_LOGGER = std::make_unique<logging::StreamLogger>(outputStream, static_cast<logging::Level>(level));
 }
diff --git a/test/test.cpp b/test/test.cpp
index 1719dfd..1504ac6 100644
--- a/test/test.cpp
+++ b/test/test.cpp
@@ -131,7 +131,7 @@ static Test::Suite* newOptimizationsTest()
 int main(int argc, char** argv)
 {
     //only output errors
-    logging::LOGGER.reset(new logging::ConsoleLogger(logging::Level::WARNING));
+    logging::DEFAULT_LOGGER.reset(new logging::ConsoleLogger(logging::Level::WARNING));
 
     Test::registerSuite(Test::newInstance<TestOptimizationSteps>, "test-optimization-steps", "Runs unit tests on the single optimization steps");
     Test::registerSuite(newOptimizationsTest, "test-optimizations", "Runs smoke tests on the single optimization steps");
```

doe300, 2020-08-12:
Yeah I know, I saw it this morning. The https://github.com/doe300/VC4C/tree/devel branch also fixes this. I just wont get around to committing a proper fix for the next few days, sorry

doe300, 2020-08-18:
This is now fixed with the latest master

closed 2020-08-18 by doe300