doe300/VC4C #150: Build fails on clean Pi

emiliodallatorre, 2020-12-30:
Hi,
I'm trying to build VC4C, but I keep getting the following errors.

`/home/pi/opencl/VC4C/src/llvm/BitcodeReader.cpp: In function ‘vc4c::intermediate::InstructionDecorations toInstructionDecorations(const llvm::Instruction&)’:
/home/pi/opencl/VC4C/src/llvm/BitcodeReader.cpp:563:13: error: ‘const class llvm::Instruction’ has no member named ‘hasNoSignedWrap’; did you mean ‘hasNoSignedZeros’?
     if(inst.hasNoSignedWrap())
             ^~~~~~~~~~~~~~~
/home/pi/opencl/VC4C/src/llvm/BitcodeReader.cpp:565:13: error: ‘const class llvm::Instruction’ has no member named ‘hasNoUnsignedWrap’; did you mean ‘hasNoSignedZeros’?
     if(inst.hasNoUnsignedWrap())
             ^~~~~~~~~~~~~~~~~
/home/pi/opencl/VC4C/src/llvm/BitcodeReader.cpp:568:13: error: ‘const class llvm::Instruction’ has no member named ‘isExact’; did you mean ‘isCast’?
     if(inst.isExact())
             ^~~~~~~
/home/pi/opencl/VC4C/src/llvm/BitcodeReader.cpp: In member function ‘vc4c::Value vc4c::llvm2qasm::BitcodeReader::toConstant(vc4c::Module&, const llvm::Value*, vc4c::Method*, vc4c::llvm2qasm::LLVMInstructionList*)’:
/home/pi/opencl/VC4C/src/llvm/BitcodeReader.cpp:1168:18: warning: unused variable ‘constant’ [-Wunused-variable]
     else if(auto constant = llvm::dyn_cast<const llvm::ConstantAggregateZero>(val))
                  ^~~~~~~~
src/CMakeFiles/VC4CC.dir/build.make:887: recipe for target 'src/CMakeFiles/VC4CC.dir/llvm/BitcodeReader.cpp.o' failed
make[2]: *** [src/CMakeFiles/VC4CC.dir/llvm/BitcodeReader.cpp.o] Error 1
CMakeFiles/Makefile2:1938: recipe for target 'src/CMakeFiles/VC4CC.dir/all' failed
make[1]: *** [src/CMakeFiles/VC4CC.dir/all] Error 2
Makefile:165: recipe for target 'all' failed
make: *** [all] Error 2`

I think it could be linked to my Make version, 3.19...

doe300, 2020-12-31:
Na, looks like some header mismatch with libllvm, which version do you have installed?

emiliodallatorre, 2020-12-31:
> Na, looks like some header mismatch with libllvm, which version do you have installed?

I guess I'm to noob to know how to check it...

doe300, 2020-12-31:
Can you send the output of these commands:

- `sudo dpkg -s llvm-dev`
- `ls /usr/bin/clang*`
- `/usr/bin/clang --version` (only if there is such an file `/usr/bin/clang` found by the above command)

emiliodallatorre, 2020-12-31:
Yessir:

```
pi@Pi-3:~ $ sudo dpkg -s llvm-dev
dpkg-query: package 'llvm-dev' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
pi@Pi-3:~ $ ls /usr/bin/clang*
/usr/bin/clang
/usr/bin/clang++
/usr/bin/clang-3.8
/usr/bin/clang++-3.8
/usr/bin/clang-apply-replacements
/usr/bin/clang-apply-replacements-3.8
/usr/bin/clang-check
/usr/bin/clang-check-3.8
/usr/bin/clang-cl-3.8
/usr/bin/clang-format
/usr/bin/clang-format-3.8
/usr/bin/clang-format-diff
/usr/bin/clang-format-diff-3.8
/usr/bin/clang-query
/usr/bin/clang-query-3.8
/usr/bin/clang-rename-3.8
/usr/bin/clang-tblgen
/usr/bin/clang-tblgen-3.8
/usr/bin/clang-tidy
/usr/bin/clang-tidy-3.8
/usr/bin/clang-tidy-diff-3.8.py
pi@Pi-3:~ $ /usr/bin/clang --version
clang version 3.8.1-24+rpi1 (tags/RELEASE_381/final)
Target: armv6--linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin
pi@Pi-3:~ $ 
````

emiliodallatorre, 2020-12-31:
Also, I can add that – even after installing `llvm-dev` – I keep getting the same error...

doe300, 2021-01-01:
You seem to have a very very old system running on your Raspberry Pi. clang/llvm 3.8 was already outdated when I started this project and was never supported.

So you could either fix the code, e.g. by commenting out the offending lines (they are mostly informative and not required), which I cannot guarantee that it won't blow up somewhere else. Or you could (and I would recommend this) upgrade your OS and the clang/llvm version installed. VC4C should work fine with clang/llvm 4.0 and up.

emiliodallatorre, 2021-01-01:
> You seem to have a very very old system running on your Raspberry Pi. clang/llvm 3.8 was already outdated when I started this project and was never supported.
>
> So you could either fix the code, e.g. [...]

I'm sorry to bother with such stupid questions, UNIX is definitely not my world... How can I install Clang greater than 4.0? If I use `apt-get`, I get version 3.8...

doe300, 2021-01-01:
What does `sudo apt-cache search llvm` report?

Also, can you post the output of `cat /etc/os-release` to see your actual OS version?

emiliodallatorre, 2021-01-01:
Yessir:
```
pi@Pi-3:~ $ sudo apt-cache search llvm
autofdo - AutoFDO Profile Toolchain
clang - C, C++ and Objective-C compiler (LLVM based)
clang-3.5 - C, C++ and Objective-C compiler (LLVM based)
clang-3.5-doc - C, C++ and Objective-C compiler (LLVM based) - Documentation
clang-3.5-examples - Clang examples
clang-3.6 - C, C++ and Objective-C compiler (LLVM based)
clang-3.6-doc - C, C++ and Objective-C compiler (LLVM based) - Documentation
clang-3.6-examples - Clang examples
clang-3.8 - C, C++ and Objective-C compiler (LLVM based)
clang-3.8-doc - C, C++ and Objective-C compiler (LLVM based) - Documentation
clang-3.8-examples - Clang examples
clang-3.9 - C, C++ and Objective-C compiler (LLVM based)
clang-3.9-doc - C, C++ and Objective-C compiler (LLVM based) - Documentation
clang-3.9-examples - Clang examples
clang-4.0 - C, C++ and Objective-C compiler
clang-4.0-doc - C, C++ and Objective-C compiler - Documentation
clang-4.0-examples - Clang examples
clang-7-doc - C, C++ and Objective-C compiler - Documentation
clang-tools-4.0 - clang-based tools for C/C++ developments
emscripten - LLVM-to-JavaScript Compiler
emscripten-doc - LLVM-to-JavaScript Compiler
ldc - LLVM D Compiler
libc++-dev - LLVM C++ Standard library (development files)
libc++-helpers - LLVM C++ Standard library - build helpers
libc++-test - LLVM C++ Standard library (test cases)
libc++1 - LLVM C++ Standard library
libc++abi-dev - LLVM low level support for a standard C++ library (development files)
libc++abi1 - LLVM low level support for a standard C++ library
libclang-3.5-dev - clang library - Development package
libclang-3.6-dev - clang library - Development package
libclang-3.8-dev - clang library - Development package
libclang-3.9-dev - clang library - Development package
libclang-4.0-dev - clang library - Development package
libclang-common-3.5-dev - clang library - Common development package
libclang-common-3.6-dev - clang library - Common development package
libclang-common-3.8-dev - clang library - Common development package
libclang-common-3.9-dev - clang library - Common development package
libclang-common-4.0-dev - clang library - Common development package
libclang-dev - clang library - Development package
libclang-perl - Perl bindings to the Clang compiler's indexing interface
libclang1 - C, C++ and Objective-C compiler (LLVM based)
libclang1-3.5 - C interface to the clang library
libclang1-3.5-dbg - clang library
libclang1-3.6 - C interface to the clang library
libclang1-3.6-dbg - clang library
libclang1-3.8 - C interface to the clang library
libclang1-3.8-dbg - clang library
libclang1-3.9 - C interface to the clang library
libclang1-3.9-dbg - clang library
libclang1-4.0 - C interface to the clang library
libclang1-4.0-dbg - clang library (debug)
libfuzzer-3.8-dev - Library for coverage-guided fuzz testing
libfuzzer-3.9-dev - Library for coverage-guided fuzz testing
libfuzzer-4.0-dev - Library for coverage-guided fuzz testing
libiomp-dev - Intel OpenMP runtime - dev package - transition package
libiomp-doc - Intel OpenMP runtime - Documentation -  Transition package
libiomp5 - Intel OpenMP runtime - transition package
libiomp5-dbg - Intel OpenMP runtime - Debug transition package
liblld-4.0 - LLVM-based linker, library
liblld-4.0-dev - LLVM-based linker, header files
liblldb-3.5 - Next generation, high-performance debugger, library
liblldb-3.5-dev - Next generation, high-performance debugger - Header files
liblldb-3.6 - Next generation, high-performance debugger, library
liblldb-3.6-dev - Next generation, high-performance debugger - Header files
liblldb-3.8 - Next generation, high-performance debugger, library
liblldb-3.8-dbg - Next generation, high-performance debugger, debugging libraries
liblldb-3.8-dev - Next generation, high-performance debugger - Header files
[... 115 lines cut ...]
llvm-6.0-dev-dbgsym - Debug symbols for llvm-6.0-dev
llvm-6.0-doc - Modular compiler and toolchain technologies, documentation
llvm-6.0-examples - Modular compiler and toolchain technologies, examples
llvm-6.0-runtime - Modular compiler and toolchain technologies, IR interpreter
llvm-6.0-runtime-dbgsym - Debug symbols for llvm-6.0-runtime
llvm-6.0-tools - Modular compiler and toolchain technologies, tools
llvm-6.0-tools-dbgsym - Debug symbols for llvm-6.0-tools
python-clang-6.0 - Clang Python Bindings
python-lldb-6.0 - Next generation, high-performance debugger, python lib
```

and

```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

doe300, 2021-01-02:
If you install `sudo apt install clang-6.0 libllvm6.0 llvm-6.0-dev` (and probably you might need to uninstall `sudo apt remove clang llvm`), then VC4C should compile correctly

emiliodallatorre, 2021-01-02:
Yes sir, I can confirm this is working, I apologize for the loss of time...!

closed 2021-01-02 by emiliodallatorre