doe300/VC4C #155: many errors linking vc4clStdLib

jorisrobijn, 2021-05-10:
I get an enormous amount of errors when vc4c is being linked.
G++ 6.3.0, clang 7.0.1
VC4CLStdLib did not give any error when building.
I have no idea what this has to do with, hopefully someone can help me further.

Scanning dependencies of target VC4C
[ 54%] Building CXX object src/CMakeFiles/VC4C.dir/main.cpp.o
[ 55%] Linking CXX executable vc4c
[E] Mon May 10 19:33:28 2021: Errors in precompilation:
[E] Mon May 10 19:33:28 2021: warning: unknown warning option '-Wdouble-promotion'; did you mean '-Wdocumentation'?
In file included from /home/pi/src/VC4C/../VC4CLStdLib/include//VC4CLStdLib.h:17:
In file included from /home/pi/src/VC4C/../VC4CLStdLib/include/_config.h:12:
/home/pi/src/VC4C/../VC4CLStdLib/include/opencl-c.h:6784:21: warning: incompatible redeclaration of library function 'acos'
float __ovld __cnfn acos(float);
                    ^
/home/pi/src/VC4C/../VC4CLStdLib/include/opencl-c.h:6784:21: note: 'acos' is a builtin with type 'double (double)'
/home/pi/src/VC4C/../VC4CLStdLib/include/opencl-c.h:6810:21: warning: incompatible redeclaration of library function 'acosh'
float __ovld __cnfn acosh(float);
                    ^
/home/pi/src/VC4C/../VC4CLStdLib/include/opencl-c.h:6810:21: note: 'acosh' is a builtin with type 'double (double)'
/home/pi/src/VC4C/../VC4CLStdLib/include/opencl-c.h:6862:21: warning: incompatible redeclaration of library function 'asin'
float __ovld __cnfn asin(float);
                    ^
(...)

/home/pi/src/VC4C/../VC4CLStdLib/include/_math.h:1089:17: note: candidate function
/home/pi/src/VC4C/../VC4CLStdLib/include/_overloads.h:154:21: note: expanded from macro 'SIMPLE_1'
        INLINE FUNC_1(ret, func, argType, argName) \
                           ^
/home/pi/src/VC4C/../VC4CLStdLib/include/_overloads.h:35:49: note: expanded from macro 'FUNC_1'
#define FUNC_1(ret, func, argType, argName) ret func(argType argName) OVERLOADABLE
                                                ^
In file included from /home/pi/src/VC4C/../VC4CLStdLib/include//VC4CLStdLib.h:25:
/home/pi/src/VC4C/../VC4CLStdLib/include/_work_items.h:60:15: error: 'overloadable' function 'get_global_linear_id' must have a prototype
INLINE size_t get_global_linear_id() OVERLOADABLE CONST
              ^
/home/pi/src/VC4C/../VC4CLStdLib/include/_work_items.h:67:15: error: 'overloadable' function 'get_local_linear_id' must have a prototype
INLINE size_t get_local_linear_id() OVERLOADABLE CONST
              ^
51 warnings and 128 errors generated.

[E] Mon May 10 19:33:29 2021: Received signal: SIGABRT
[E] Mon May 10 19:33:29 2021:  (1) /home/pi/src/VC4C/build/src/libVC4CC.so.1.2 : +0xa25178 [0x768ad178]
[E] Mon May 10 19:33:29 2021:  (2) /lib/arm-linux-gnueabihf/libc.so.6 : __default_rt_sa_restorer+0 [0x729316c0]
[E] Mon May 10 19:33:29 2021:  (3) /lib/arm-linux-gnueabihf/libc.so.6 : gsignal+0xa0 [0x7293045c]
[E] Mon May 10 19:33:29 2021:  (4) /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 : __gnu_cxx::__verbose_terminate_handler()+0xfc [0x72b64008]
src/CMakeFiles/VC4C.dir/build.make:97: recipe for target 'src/vc4c' failed
make[2]: *** [src/vc4c] Error 6
make[2]: *** Deleting file 'src/vc4c'
CMakeFiles/Makefile2:443: recipe for target 'src/CMakeFiles/VC4C.dir/all' failed
make[1]: *** [src/CMakeFiles/VC4C.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

doe300, 2021-05-12:
What does `/usr/bin/clang --version` print?

I don't need the full output, but can you post the first few lines (maybe until the first error) of the following command? The logged clang command etc. is more interesting than the error dump itself.

    <build folder>/src/vc4c --verbose --precompile-stdlib -o <output directory> <path/to/vc4cl-stdlib>/include/VC4CLStdLib.h

jorisrobijn, 2021-05-12:
pi@liveatc:~/src/VC4C$ clang --version
Raspbian clang version 3.5.0-10+rpi1 (tags/RELEASE_350/final) (based on LLVM 3.5.0)
Target: arm-unknown-linux-gnueabihf
Thread model: posix

But cmake detects 7.0.1, which is also installed:

pi@liveatc:~/src/VC4C/build$ clang-7 --version
clang version 7.0.1-8~deb9u4 (tags/RELEASE_701/final)
Target: armv7l-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin

ii  opencl-c-headers    2.1-1          all            OpenCL (Open Computing Language) C header f
ii  opencl-clhpp-header 2.0.10-1       all            C++ headers for OpenCL development
un  opencl-clhpp-header <none>         <none>         (no description available)
ii  opencl-headers      2.1-1          all            OpenCL (Open Computing Language) header fil

You want to see what clang command line is issued by your precompile action. How do I get to see that?

doe300, 2021-05-12:
If you run

    <build folder>/src/vc4c --verbose --precompile-stdlib -o <output directory> <path/to/vc4cl-stdlib>/include/VC4CLStdLib.h

it should be printed in the first lines.

But since `clang` is reported to be 3.5.0 this is could be the problem, since clang 3.5 is far too old. To use the clang-7, you could configure `cmake` with the `-DCLANG_FOUND=/usr/bin/clang-7` parameter, this should fix the compilation errors.

The underlying problem is probably in https://github.com/doe300/VC4C/blob/master/cmake/clang.cmake#L6, because the non-versioned (default) clang is prefered over any supported version. And on your system the non-versioned clang is 3.5.

It could be required that you also manually set the `LLVM_LINK_FOUND` and other variables checked in this file.

jorisrobijn, 2021-05-13:
Instead I changed the symbolic link of clang from 3.5 to 7.0.1, and the retried. Now it did precompile fine. So although it detects clang 7 specifically (see my earlier output), it doesn't use it, and as you wrote it uses the versionless clang command.
This is apparently the case on stretch, installing clang-7 doesn't change the clang symbolic link.
Thanks!

closed 2021-05-13 by jorisrobijn