doe300/VC4C #156: Compile failure with Clang/LLVM 11

zingaburga, 2021-05-25:
I used to have Clang/LLVM 8 and Clang/LLVM 11 installed, however I was getting issues when running OpenCL code which appeared to point to a conflict between the two (like it was using v8 compiled code against v11 tools or something like that).

To try to remedy that, I removed Clang/LLVM 8, leaving only 11 behind, and rebuilt, but this seems to now fail (I'm guessing it originally decided to build against version 8).

From a quick glance, it looks like the key issue might be not being able to convert strings to/from llvm::StringRef.

Build log:

~~~
/home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp: In constructor ‘vc4c::llvm2qasm::BitcodeReader::BitcodeReader(std::istream&, vc4c::SourceType)’:
/home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp:129:111: error: no matching function for call to ‘vc4c::CompilationError::CompilationError(vc4c::CompilationStep, const char [29], llvm::StringRef)’
  129 |            throw CompilationError(CompilationStep::PARSER, "Error parsing LLVM IR module", error.getMessage());
      |                                                                                                              ^

In file included from /home/pi/dev/opencl/VC4C/src/llvm/../Locals.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../GlobalValues.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../Module.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../Parser.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp:7:
/home/pi/dev/opencl/VC4C/include/CompilationError.h:74:9: note: candidate: ‘vc4c::CompilationError::CompilationError(vc4c::CompilationError&&)’
   74 |         CompilationError(CompilationError&&) = default; // RPi cross-compiler throws on noexcept
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:74:9: note:   candidate expects 1 argument, 3 provided
/home/pi/dev/opencl/VC4C/include/CompilationError.h:73:9: note: candidate: ‘vc4c::CompilationError::CompilationError(const vc4c::CompilationError&)’
   73 |         CompilationError(const CompilationError&) = default;
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:73:9: note:   candidate expects 1 argument, 3 provided
/home/pi/dev/opencl/VC4C/include/CompilationError.h:72:9: note: candidate: ‘vc4c::CompilationError::CompilationError(vc4c::CompilationStep, const string&, const string&)’
   72 |         CompilationError(CompilationStep step, const std::string& message, const std::string& object);
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:72:95: note:   no known conversion for argument 3 from ‘llvm::StringRef’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
   72 |         CompilationError(CompilationStep step, const std::string& message, const std::string& object);
      |                                                                            ~~~~~~~~~~~~~~~~~~~^~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:71:9: note: candidate: ‘vc4c::CompilationError::CompilationError(vc4c::CompilationStep, const string&)’
   71 |         CompilationError(CompilationStep step, const std::string& message);
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:71:9: note:   candidate expects 2 arguments, 3 provided
/home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp: In function ‘void extractKernelMetadata(vc4c::Method&, const llvm::Function&, const llvm::Module&, const llvm::LLVMContext&)’:
/home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp:164:13: warning: unused variable ‘metadata’ [-Wunused-variable]
  164 |     if(auto metadata = func.getMetadata("kernel_arg_access_qual"))
      |             ^~~~~~~~
/home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp:179:72: error: no match for ‘operator=’ (operand types are ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} and ‘llvm::StringRef’)
  179 |                 kernel.parameters.at(i).origTypeName = name->getString();
      |                                                                        ^
In file included from /usr/include/c++/10/string:55,
                 from /usr/include/c++/10/stdexcept:39,
                 from /home/pi/dev/opencl/VC4C/include/CompilationError.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../Locals.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../GlobalValues.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../Module.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/../Parser.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp:7:
/usr/include/c++/10/bits/basic_string.h:665:7: note: candidate: ‘std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::operator=(const std::__cxx11::basic_string<_CharT, _Traits, _Alloc>&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’
  665 |       operator=(const basic_string& __str)
      |       ^~~~~~~~
/usr/include/c++/10/bits/basic_string.h:665:37: note:   no known conversion for argument 1 from ‘llvm::StringRef’ to ‘const std::__cxx11::basic_string<char>&’
  665 |       operator=(const basic_string& __str)
      |                 ~~~~~~~~~~~~~~~~~~~~^~~~~
/usr/include/c++/10/bits/basic_string.h:675:7: note: candidate: ‘std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::operator=(const _CharT*) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’
  675 |       operator=(const _CharT* __s)
      |       ^~~~~~~~
/usr/include/c++/10/bits/basic_string.h:675:31: note:   no known conversion for argument 1 from ‘llvm::StringRef’ to ‘const char*’
  675 |       operator=(const _CharT* __s)
      |                 ~~~~~~~~~~~~~~^~~
/usr/include/c++/10/bits/basic_string.h:686:7: note: candidate: ‘std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::operator=(_CharT) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>]’
  686 |       operator=(_CharT __c)
      |       ^~~~~~~~
/usr/include/c++/10/bits/basic_string.h:686:24: note:   no known conversion for argument 1 from ‘llvm::StringRef’ to ‘char’
[... 1318 lines cut ...]
                 from /home/pi/dev/opencl/VC4C/src/llvm/../Parser.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.h:10,
                 from /home/pi/dev/opencl/VC4C/src/llvm/BitcodeReader.cpp:7:
/home/pi/dev/opencl/VC4C/include/CompilationError.h:74:9: note: candidate: ‘vc4c::CompilationError::CompilationError(vc4c::CompilationError&&)’
   74 |         CompilationError(CompilationError&&) = default; // RPi cross-compiler throws on noexcept
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:74:9: note:   candidate expects 1 argument, 3 provided
/home/pi/dev/opencl/VC4C/include/CompilationError.h:73:9: note: candidate: ‘vc4c::CompilationError::CompilationError(const vc4c::CompilationError&)’
   73 |         CompilationError(const CompilationError&) = default;
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:73:9: note:   candidate expects 1 argument, 3 provided
/home/pi/dev/opencl/VC4C/include/CompilationError.h:72:9: note: candidate: ‘vc4c::CompilationError::CompilationError(vc4c::CompilationStep, const string&, const string&)’
   72 |         CompilationError(CompilationStep step, const std::string& message, const std::string& object);
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:72:95: note:   no known conversion for argument 3 from ‘llvm::StringRef’ to ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’}
   72 |         CompilationError(CompilationStep step, const std::string& message, const std::string& object);
      |                                                                            ~~~~~~~~~~~~~~~~~~~^~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:71:9: note: candidate: ‘vc4c::CompilationError::CompilationError(vc4c::CompilationStep, const string&)’
   71 |         CompilationError(CompilationStep step, const std::string& message);
      |         ^~~~~~~~~~~~~~~~
/home/pi/dev/opencl/VC4C/include/CompilationError.h:71:9: note:   candidate expects 2 arguments, 3 provided
^Cmake[2]: *** [src/CMakeFiles/VC4CC.dir/build.make:836: src/CMakeFiles/VC4CC.dir/llvm/BitcodeReader.cpp.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:2065: src/CMakeFiles/VC4CC.dir/all] Interrupt
make: *** [Makefile:182: all] Interrupt
~~~

doe300, 2021-05-25:
Could be a while until I update the repo itself, but [this patch](https://gist.github.com/doe300/a1c29a6f656eabd3fcd46bbc992ea9d2) should fix clang 11 compilation and usage.

zingaburga, 2021-05-26:
Thanks so much for the quick patch!

The cmake patch didn't apply for me - it seems like the file uses tab indentation, whilst the patch is looking for spaces.  I applied the changes manually though, and it all works great!