doe300/VC4C #157: compile failure with clang6.0 llvm-6.0

uniartisan, 2021-06-11:
my system is raspberry os (64 bit), I have read similar issues: #108 #150, but the issue still occurs.
here is my command:
```
sudo apt install clang-6.0 libllvm6.0 llvm-6.0-dev
cmake -DMULTI_THREADED=OFF .
```

here is CmakeError.txt:
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_59bd7/fast && /usr/bin/make -f CMakeFiles/cmTC_59bd7.dir/build.make CMakeFiles/cmTC_59bd7.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_59bd7.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_59bd7.dir/src.c.o   -c /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_59bd7
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_59bd7.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_59bd7.dir/src.c.o  -o cmTC_59bd7 
/usr/bin/ld: CMakeFiles/cmTC_59bd7.dir/src.c.o: in function `main':
src.c:(.text+0x34): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x3c): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x48): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_59bd7.dir/build.make:87：cmTC_59bd7] 错误 1
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
make: *** [Makefile:121：cmTC_59bd7/fast] 错误 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_16556/fast && /usr/bin/make -f CMakeFiles/cmTC_16556.dir/build.make CMakeFiles/cmTC_16556.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_16556.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_16556.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_16556
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_16556.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_16556.dir/CheckFunctionExists.c.o  -o cmTC_16556  -lpthreads 
/usr/bin/ld: 找不到 -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_16556.dir/build.make:87：cmTC_16556] 错误 1
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
make: *** [Makefile:121：cmTC_16556/fast] 错误 2

Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_68bb2/fast && /usr/bin/make -f CMakeFiles/cmTC_68bb2.dir/build.make CMakeFiles/cmTC_68bb2.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_68bb2.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_68bb2.dir/src.c.o   -c /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_68bb2
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_68bb2.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_68bb2.dir/src.c.o  -o cmTC_68bb2 
/usr/bin/ld: CMakeFiles/cmTC_68bb2.dir/src.c.o: in function `main':
src.c:(.text+0x34): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x3c): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x48): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_68bb2.dir/build.make:87：cmTC_68bb2] 错误 1
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
make: *** [Makefile:121：cmTC_68bb2/fast] 错误 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_9b2e8/fast && /usr/bin/make -f CMakeFiles/cmTC_9b2e8.dir/build.make CMakeFiles/cmTC_9b2e8.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_9b2e8.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_9b2e8.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_9b2e8
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_9b2e8.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_9b2e8.dir/CheckFunctionExists.c.o  -o cmTC_9b2e8  -lpthreads 
/usr/bin/ld: 找不到 -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_9b2e8.dir/build.make:87：cmTC_9b2e8] 错误 1
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
make: *** [Makefile:121：cmTC_9b2e8/fast] 错误 2
```

How can I solve this?  Wish some one can help me :)

doe300, 2021-06-11:
You seem to not have any pthread library installed. For a minimal fix you could try to install `sudo apt-get install libpthread-stubs0-dev`, otherwise I would recommend installing `sudo apt install build-essential` which also installs GCC, but should provide all other development requirements.

uniartisan, 2021-06-11:
> You seem to not have any pthread library installed. For a minimal fix you could try to install `sudo apt-get install libpthread-stubs0-dev`, otherwise I would recommend installing `sudo apt install build-essential` which also installs GCC, but should provide all other development requirements.

```
pi@raspberrypi:~ $ sudo apt-get install libpthread-stubs0-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
libpthread-stubs0-dev 已经是最新版 (0.4-1)。
libpthread-stubs0-dev 已设置为手动安装。
下列软件包是自动安装的并且现在不需要了：
  llvm-7-runtime llvm-runtime lxplug-volume python-colorzero rpi-eeprom-images
使用'sudo apt autoremove'来卸载它(它们)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。
pi@raspberrypi:~ $ sudo apt install build-essential
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
build-essential 已经是最新版 (12.6)。
下列软件包是自动安装的并且现在不需要了：
  llvm-7-runtime llvm-runtime lxplug-volume python-colorzero rpi-eeprom-images
使用'sudo apt autoremove'来卸载它(它们)。
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。
```

It seems that all the library are installed.  However, the issue still occurs.
here is full output, which may help:
```
cat /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeOutput.log
The system is: Linux - 5.10.17-v8+ - aarch64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/pi/workspace/pi/VC4C/CMakeFiles/3.16.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/pi/workspace/pi/VC4C/CMakeFiles/3.16.3/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_d606a/fast && /usr/bin/make -f CMakeFiles/cmTC_d606a.dir/build.make CMakeFiles/cmTC_d606a.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_d606a.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_d606a.dir/testCCompiler.c.o   -c /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_d606a
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d606a.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_d606a.dir/testCCompiler.c.o  -o cmTC_d606a 
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_bbcd1/fast && /usr/bin/make -f CMakeFiles/cmTC_bbcd1.dir/build.make CMakeFiles/cmTC_bbcd1.dir/build
make[1]: Entering directory '/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp'
[... 851 lines cut ...]

Determining if the include file pthread.h exists passed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_d6cc2/fast && /usr/bin/make -f CMakeFiles/cmTC_d6cc2.dir/build.make CMakeFiles/cmTC_d6cc2.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_d6cc2.dir/CheckIncludeFile.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_d6cc2.dir/CheckIncludeFile.c.o   -c /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp/CheckIncludeFile.c
Linking C executable cmTC_d6cc2
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_d6cc2.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_d6cc2.dir/CheckIncludeFile.c.o  -o cmTC_d6cc2 
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”

Determining if the function pthread_create exists in the pthread passed with the following output:
Change Dir: /home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_51f15/fast && /usr/bin/make -f CMakeFiles/cmTC_51f15.dir/build.make CMakeFiles/cmTC_51f15.dir/build
make[1]: 进入目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
Building C object CMakeFiles/cmTC_51f15.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_51f15.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_51f15
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_51f15.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_51f15.dir/CheckFunctionExists.c.o  -o cmTC_51f15  -lpthread 
make[1]: 离开目录“/home/pi/workspace/pi/VC4C/CMakeFiles/CMakeTmp”
```

uniartisan, 2021-06-11:
Update:
```
pi@raspberrypi:~/workspace/pi/VC4C/CMakeFiles $ find /lib -name "*pthread*"
/lib/arm-linux-gnueabihf/libpthread.so.0
/lib/arm-linux-gnueabihf/libpthread-2.28.so
/lib/aarch64-linux-gnu/libpthread.so.0
/lib/aarch64-linux-gnu/libpthread-2.28.so
```

doe300, 2021-06-11:
What are your versions of:

- CMake
- `/usr/bin/cc`

uniartisan, 2021-06-12:
Cmake:3.16.3-3~bpo10+1
gcc/g++:4:8.3.0-1

Thx a lot for your time, I have verified my project on the Raspberry Pi 4B. :)

doe300, 2021-06-12:
Just FYI, if you are not aware, VC4CL won't work at all on 64-bit Raspbian or Raspberry Pi 4.

To solve the build problem, the only thing I can think of is to update your GCC, 4.8 is quiet old, maybe that fixes something...

uniartisan, 2021-06-12:
> Just FYI, if you are not aware, VC4CL won't work at all on 64-bit Raspbian or Raspberry Pi 4.
> 
> To solve the build problem, the only thing I can think of is to update your GCC, 4.8 is quite old, maybe that fixes something...

Thank you for your reply, VC4CL only works on Raspberry Pi 3B. I just adjust my ML program to 4B and test it in order to save time.
I will try build gcc-6 next week, and I will update if this solve the build problem.

By the way:
64-bit Raspbian is based on Debian, and its package is directly from  debian arm64, which means some package is really old.

jDietschi, 2023-06-19:
> 
> Thank you for your reply, VC4CL only works on Raspberry Pi 3B. I just adjust my ML program to 4B and test it in order to save time. I will try build gcc-6 next week, and I will update if this solve the build problem.
> 

Is this really only compatible if Raspberry Pi 3B? I'm trying to set it up on my Raspberry Pi 3A and failing so far with the same CMakeErrors as @uniartisan 

Also thank you so much for all your work!