doe300/VC4C #158: Build fail on Raspbian 10

Zenitur, 2021-08-23:
```
Scanning dependencies of target VC4C
[ 54%] Building CXX object src/CMakeFiles/VC4C.dir/main.cpp.o
[ 55%] Linking CXX executable vc4c
[E] Mon Aug 23 16:24:56 2021: Errors in precompilation:
[E] Mon Aug 23 16:24:56 2021: In file included from /home/pi/opencl2/VC4C/../VC4CLStdLib/include//VC4CLStdLib.h:25:
/home/pi/opencl2/VC4C/../VC4CLStdLib/include/_work_items.h:60:15: error: 'overloadable' function 'get_global_linear_id' must have a prototype
INLINE size_t get_global_linear_id() OVERLOADABLE CONST
              ^
/home/pi/opencl2/VC4C/../VC4CLStdLib/include/_work_items.h:67:15: error: 'overloadable' function 'get_local_linear_id' must have a prototype
INLINE size_t get_local_linear_id() OVERLOADABLE CONST
              ^
2 errors generated.

[E] Mon Aug 23 16:24:56 2021:  (1) /home/pi/opencl2/VC4C/src/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x7679a584]
[E] Mon Aug 23 16:24:56 2021:  (2) /home/pi/opencl2/VC4C/src/libVC4CC.so.1.2 : vc4c::precompilation::runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x1bc [0x76ca4fec]
[E] Mon Aug 23 16:24:56 2021:  (3) /home/pi/opencl2/VC4C/src/libVC4CC.so.1.2 : vc4c::precompilation::precompileStandardLibraryFiles(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x3d8 [0x76cac644]
[E] Mon Aug 23 16:24:56 2021:  (4) /home/pi/opencl2/VC4C/src/vc4c : main+0x94c [0x3b638]
[E] Mon Aug 23 16:24:56 2021:  (5) /lib/arm-linux-gnueabihf/libc.so.6 : __libc_start_main+0x10c [0x72b3e718]
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: In file included from /home/pi/opencl2/VC4C/../VC4CLStdLib/include//VC4CLStdLib.h:25:
/home/pi/opencl2/VC4C/../VC4CLStdLib/include/_work_items.h:60:15: error: 'overloadable' function 'get_global_linear_id' must have a prototype
INLINE size_t get_global_linear_id() OVERLOADABLE CONST
              ^
/home/pi/opencl2/VC4C/../VC4CLStdLib/include/_work_items.h:67:15: error: 'overloadable' function 'get_local_linear_id' must have a prototype
INLINE size_t get_local_linear_id() OVERLOADABLE CONST
              ^
2 errors generated.

[E] Mon Aug 23 16:24:56 2021: Received signal: SIGABRT
[E] Mon Aug 23 16:24:56 2021:  (1) /home/pi/opencl2/VC4C/src/libVC4CC.so.1.2 : +0x99e1e4 [0x7681e1e4]
[E] Mon Aug 23 16:24:56 2021:  (2) /lib/arm-linux-gnueabihf/libc.so.6 : __default_rt_sa_restorer+0 [0x72b54130]
```
[image]

Zenitur, 2021-08-24:
Also [the topic](https://www.raspberrypi.org/forums/viewtopic.php?t=194952) on the Raspberry Pi forum is now locked and we cannot post any comment.

doe300, 2021-08-24:
I need some more information on the issue, can you configure the project with the CMake variable `-DVC4CL_STDLIB_PRECOMPILE=OFF`, recompile (should now work) and then post the output of `<build dir>/src/vc4c --version`?