doe300/VC4C #159: c++: fatal error: Killed signal terminated program cc1plus

shooshx, 2021-11-20:
I'm trying to follow the instructions in https://qengineering.eu/install-opencl-on-raspberry-pi-3.html
on my pi3 and I'm getting:

```
/home/pi/opencl/VC4C/build/cpptest-lite/src/cpptest-lite-project/include/comparisons.h: In instantiation of ‘bool Test::Comparisons::isSame(const T1&, const T2&) [with T1 = int; T2 = unsigned int]’:
/home/pi/opencl/VC4C/test/TestInstructions.cpp:1626:5:   required from here
/home/pi/opencl/VC4C/build/cpptest-lite/src/cpptest-lite-project/include/comparisons.h:70:16: warning: comparison of integer expressions of different signedness: ‘const int’ and ‘const unsigned int’ [-Wsign-compare]
    return val1 == val2;
           ~~~~~^~~~~~~
c++: fatal error: Killed signal terminated program cc1plus
```

looks like the compiler is just killed? maybe out of memory?
what do?

doe300, 2021-11-20:
Can you post the output of `vc4c --version`? Also can you check your out-of-memory theory by running some system monitor (e.g. htop) while running the test code and checking whether the system memory runs full?

shooshx, 2021-11-20:
```
pi@retropie:~/opencl/VC4C/build $ /home/pi/opencl/VC4C/build/src/vc4c --version
Running VC4C in version: 0.4.9999 (7958032)
Build configuration: debug mode; multi-threaded optimization; builtin SPIR-V front-end; LLVM library front-end with libLLVM 7; vc4asm verification

Standard library location:
        header in /home/pi/opencl/VC4C/../VC4CLStdLib/include/defines.h
        PCH in /home/pi/opencl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.h.pch
        LLVM module in /home/pi/opencl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.bc
        SPIR-V module in
Tool locations:
        clang in /usr/bin/clang (default)
        llvm-spirv not in $PATH
        llvm-link in /usr/bin/llvm-link-7 (default)
        opt in /usr/bin/opt-7 (default)
        llvm-dis in /usr/bin/llvm-dis-7 (default)
        llvm-as in /usr/bin/llvm-as-7 (default)
        spirv-link not in $PATH
```

looking at htop while it's running I am seeing it climb up to 718M/746M just before being killed. So yea, pretty sure that's it.
Is it possible to build without these tests?

shooshx, 2021-11-20:
ok, that was silly, it just occurred to me to close VSCode running in SSH, that freed about 200MB and allowed this to run without issue. Sorry for the bother :)

closed 2021-11-20 by shooshx