doe300/VC4C #160: Build Error

ixiDev, 2022-06-19:
Hi there.
I try to build this package on Raspberry Pi3B with ubuntu rasp 22.04 LTS
and I get this error 

```
VC4C/src/llvm/BitcodeReader.cpp:1045:39: error: ‘const class llvm::CallInst’ has no member named ‘getNumArgOperands’; did you mean ‘getNumOperands’?
 1045 |         for(unsigned i = 0; i < call->getNumArgOperands(); ++i)
      |                                       ^~~~~~~~~~~~~~~~~
      |                                       getNumOperands
/home/pi/Downloads/VC4C/src/llvm/BitcodeReader.cpp: In member function ‘vc4c::Value vc4c::llvm2qasm::BitcodeReader::toConstant(vc4c::Module&, const llvm::Value*, vc4c::Method*, vc4c::llvm2qasm::LLVMInstructionList*)’:
/VC4C/src/llvm/BitcodeReader.cpp:1420:18: warning: unused variable ‘constant’ [-Wunused-variable]

```
**clang --version**
```
Ubuntu clang version 14.0.0-1ubuntu1
Target: arm-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin
```

**gcc --version**
```
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I change **getNumArgOperands** to **getNumOperands** and the compilation success but after I install all packages (VC4C,VC4CL) and run clinfo here is the output

```
pi@rpi3:~/Downloads/VC4CL$ clinfo 
[E] Sun Jun 19 04:25:41 2022: Received signal: SIGSEGV
[E] Sun Jun 19 04:25:41 2022:  (1) /usr/local/lib/libVC4CC.so.1.2 : +0xeda9c [0xb6abda9c]
[E] Sun Jun 19 04:25:41 2022:  (2) /lib/arm-linux-gnueabihf/libc.so.6 : +0x2d6f0 [0xb6ded6f0]
```

doe300, 2022-06-19:
What do you mean with "ubuntu rasp 22.04 LTS"? Are you using Raspberry Pi OS or an Ubuntu? In any case, VC4CL does not work anymore with the "new" Raspberry Pi OS and any other OS might not have the required firmware runtime libraries installed at all.

ixiDev, 2022-06-19:
> What do you mean with "ubuntu rasp 22.04 LTS"? Are you using Raspberry Pi OS or an Ubuntu? In any case, VC4CL does not work anymore with the "new" Raspberry Pi OS and any other OS might not have the required firmware runtime libraries installed at all.

Thanks for reply <3 

I test it right now on the latest Rasbian and it's not working, its compile but I get the same error 
```
[E] Sun Jun 19 07:44:54 2022: Received signal: SIGSEGV
[E] Sun Jun 19 07:44:54 2022:  (1) /usr/lib/libVC4CC.so.1.2 : +0x107a4c [0x7695ea4c]
[E] Sun Jun 19 07:44:54 2022:  (2) /lib/arm-linux-gnueabihf/libc.so.6 : __default_rt_sa_restorer+0 [0x76d7fdb0]
```

is there anything to do or just go back to old Raspbien ? thanks again

doe300, 2022-06-19:
You will need to provide a more detailed stack trace for me to find out the reason for the crash. Can you for example run the code with GDB and then post the stack trace on crash?

Something like:

    gdb clinfo
    (in gdb shell) run
    (in gdb shell when the SEGFAULT is caught) bt

ixiDev, 2022-06-19:
Her  is the stack trace from gdb 

```
Reading symbols from clinfo...
(No debugging symbols found in clinfo)
(gdb) run
Starting program: /usr/bin/clinfo 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00075118 in ?? ()
(gdb) backtrace 
#0  0x00075118 in ?? ()
#1  0x76f9e8e4 in ?? () from /lib/arm-linux-gnueabihf/libOpenCL.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

```

I think the error comes from the prebuild libOpenCL.so that comes with the latest Rasbian OS because when I build VC4CL it linked I guess with /lib/arm-linux-gnueabihf/libOpenCL.so  but here is ldd for libVC4CL.so

```
VC4CL/build/src $ ldd libVC4CL.so.0.4.9999 
        linux-vdso.so.1 (0x7eb4e000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76eb9000)
        libVC4CC.so.1.2 => /home/pi/Downloads/VC4C/build/src/libVC4CC.so.1.2 (0x76a92000)
        libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0x7690a000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x768dd000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76789000)
        /lib/ld-linux-armhf.so.3 (0x76f9d000)
        libatomic.so.1 => /lib/arm-linux-gnueabihf/libatomic.so.1 (0x76770000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76744000)
        libLLVM-6.0.so.1 => /lib/arm-linux-gnueabihf/libLLVM-6.0.so.1 (0x73629000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x735ba000)
        libffi.so.7 => /lib/arm-linux-gnueabihf/libffi.so.7 (0x735a2000)
        libedit.so.2 => /lib/arm-linux-gnueabihf/libedit.so.2 (0x73568000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x73540000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x7352c000)
        libtinfo.so.6 => /lib/arm-linux-gnueabihf/libtinfo.so.6 (0x734fa000)
        libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0x734d8000)
        libmd.so.0 => /lib/arm-linux-gnueabihf/libmd.so.0 (0x734bd000)
```

her is build log 

```

pi@rpi3:~/Downloads/VC4C/build $ cmake -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release  -DCMAKE_INSTALL_PREFIX=/usr/  ..
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- VC4CL standard library headers found: /home/pi/Downloads/VC4C/../VC4CLStdLib/include/
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CMake 3.14+ FetchContent to include dependencies...
-- CLang compiler found: /usr/bin/clang-11
-- LLVM-dis found: /usr/bin/llvm-dis-11
-- LLVM-as found: /usr/bin/llvm-as-11
-- LLVM-link found: /usr/bin/llvm-link-11
-- Compiling LLVM library front-end with LLVM in version 11.0.1 located in '/usr/lib/llvm-11/lib'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- found clang-format: /usr/bin/clang-format
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Downloads/VC4C/build
```