doe300/VC4C #161: Compilation error

solsticedhiver, 2022-09-17:
when trying to compile vc4c-git-r709.e3eed1e-3 package from AUR on archlinux, this ends  with

```
[ 41%] Building CXX object test/CMakeFiles/TestData.dir/FloatTests.cpp.o
[100%] Linking CXX shared library libcpptest-lite.so
[100%] Built target cpptest-lite
[ 41%] Built target cpptest-lite-project-build
[ 41%] Building CXX object test/CMakeFiles/TestData.dir/IntegerTests.cpp.o
[ 42%] Building CXX object test/CMakeFiles/TestData.dir/MathTests.cpp.o
In file included from /home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/ArithmeticTests.cpp:6:
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/TestEntries.h: In instantiation of 'std::vector<_RealType> test_data::transform(const std::vector<T>&, const Func&) [with R = float; T = float; Func = registerTypeTests<float, int>(const std::string&)::<lambda(float)>]':
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/ArithmeticTests.cpp:430:25:   required from 'void registerTypeTests(const std::string&) [with T = float; IntType = int; std::string = std::__cxx11::basic_string<char>]'
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/ArithmeticTests.cpp:609:38:   required from here
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/TestEntries.h:107:29: warning: conversion from 'int' to '__gnu_cxx::__alloc_traits<std::allocator<float>, float>::value_type' {aka 'float'} may change value [-Wconversion]
  107 |             result[i] = func(arg[i]);
      |                         ~~~~^~~~~~~~
[ 42%] Building CXX object test/CMakeFiles/TestData.dir/MemoryTests.cpp.o
In file included from /home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/MathTests.cpp:6:
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/TestEntries.h: In instantiation of 'std::vector<_RealType> test_data::transform(const std::vector<T>&, const std::vector<U>&, const Func&) [with R = float; T = float; U = int; Func = std::function<float(float, float)>]':
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/MathTests.cpp:376:67:   required from here
/home/solstice/.cache/paru/clone/vc4c-git/src/VC4C/test/TestEntries.h:119:29: warning: conversion from '__gnu_cxx::__alloc_traits<std::allocator<int>, int>::value_type' {aka 'int'} to 'float' may change value [-Wconversion]
  119 |             result[i] = func(arg0[i], arg1[i]);
      |                         ~~~~^~~~~~~~~~~~~~~~~~
[ 43%] Building CXX object test/CMakeFiles/TestData.dir/RelationalTests.cpp.o
[ 43%] Building CXX object test/CMakeFiles/TestData.dir/VectorTests.cpp.o
[ 44%] Building CXX object test/CMakeFiles/TestData.dir/TestData.cpp.o
[ 44%] Building CXX object test/CMakeFiles/TestData.dir/test_files/test_files.cpp.o
[ 44%] Linking CXX static library libvc4c_testdata.a
[ 44%] Built target TestData
make: *** [Makefile:166: all] Error 2
```
using gcc 12.2

doe300, 2022-09-17:
I can't see an actual error in this log. Can you post the full build log (does not have to be a clean build, just so that we have the actual error message).

solsticedhiver, 2022-09-17:
I don't know what happened. I have updated the post above with a complete output, I hope

doe300, 2022-09-19:
I fixed it in the cpplog project.  To get the fix into VC4C, you can https://github.com/doe300/VC4C/blob/master/cmake/cpplog.cmake#L19 and replace the `v0.6` with `master`.

solsticedhiver, 2022-09-19:
there is a problem.
If I restart the compilation I get another error than 2 days ago. I don't know what is going on:
```
[100%] Built target cpptest-lite
[ 43%] Built target cpptest-lite-project-build
[ 43%] Building CXX object test/CMakeFiles/TestData.dir/IntegerTests.cpp.o
In file included from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/ArithmeticTests.cpp:6:
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/TestEntries.h: In instantiation of 'std::vector<_RealType> test_data::transform(const std::vector<T>&, const Func&) [with R = float; T = float; Func = registerTypeTests<float, int>(const std::string&)::<lambda(float)>]':
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/ArithmeticTests.cpp:430:25:   required from 'void registerTypeTests(const std::string&) [with T = float; IntType = int; std::string = std::__cxx11::basic_string<char>]'
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/ArithmeticTests.cpp:609:38:   required from here
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/TestEntries.h:107:29: warning: conversion from 'int' to '__gnu_cxx::__alloc_traits<std::allocator<float>, float>::value_type' {aka 'float'} may change value [-Wconversion]
  107 |             result[i] = func(arg[i]);
      |                         ~~~~^~~~~~~~
[ 44%] Building CXX object test/CMakeFiles/TestData.dir/MathTests.cpp.o
[ 44%] Building CXX object test/CMakeFiles/TestData.dir/MemoryTests.cpp.o
In file included from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/MathTests.cpp:6:
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/TestEntries.h: In instantiation of 'std::vector<_RealType> test_data::transform(const std::vector<T>&, const std::vector<U>&, const Func&) [with R = float; T = float; U = int; Func = std::function<float(float, float)>]':
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/MathTests.cpp:376:67:   required from here
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/test/TestEntries.h:119:29: warning: conversion from '__gnu_cxx::__alloc_traits<std::allocator<int>, int>::value_type' {aka 'int'} to 'float' may change value [-Wconversion]
  119 |             result[i] = func(arg0[i], arg1[i]);
      |                         ~~~~^~~~~~~~~~~~~~~~~~
[ 45%] Building CXX object test/CMakeFiles/TestData.dir/RelationalTests.cpp.o
[ 45%] Building CXX object test/CMakeFiles/TestData.dir/VectorTests.cpp.o
[ 46%] Building CXX object test/CMakeFiles/TestData.dir/TestData.cpp.o
[ 46%] Building CXX object test/CMakeFiles/TestData.dir/test_files/test_files.cpp.o
[ 46%] Linking CXX static library libvc4c_testdata.a
[ 46%] Built target TestData
make: *** [Makefile:166: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
It even gives nother error with another run!

If I changes cpplog to master like you said, I got another error:

```[ 44%] Built target cpptest-lite-project-build
[ 45%] Building CXX object src/CMakeFiles/VC4CC.dir/GlobalValues.cpp.o
[ 46%] Building CXX object src/CMakeFiles/VC4CC.dir/Compiler.cpp.o
[ 46%] Building CXX object src/CMakeFiles/VC4CC.dir/Method.cpp.o
[ 46%] Building CXX object src/CMakeFiles/VC4CC.dir/Module.cpp.o
<command-line>: warning: "_GNU_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
In file included from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/GlobalValues.cpp:9:
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/SIMDVector.h: In constructor 'constexpr vc4c::SIMDVector::SIMDVector(std::initializer_list<vc4c::Literal>)':
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/SIMDVector.h:53:24: error: call to non-'constexpr' function '_OIter std::copy_n(_IIter, _Size, _OIter) [with _IIter = const vc4c::Literal*; _Size = long unsigned int; _OIter = vc4c::Literal*]'
   53 |             std::copy_n(list.begin(), std::min(NATIVE_VECTOR_SIZE, list.size()), elements.begin());
      |             ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/12.2.0/algorithm:61,
                 from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/tools/SmallMap.h:11,
                 from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/Locals.h:14,
                 from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/GlobalValues.h:10,
                 from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/GlobalValues.cpp:7:
/usr/include/c++/12.2.0/bits/stl_algo.h:746:5: note: '_OIter std::copy_n(_IIter, _Size, _OIter) [with _IIter = const vc4c::Literal*; _Size = long unsigned int; _OIter = vc4c::Literal*]' declared here
  746 |     copy_n(_InputIterator __first, _Size __n, _OutputIterator __result)
      |     ^~~~~~
make[2]: *** [src/CMakeFiles/VC4CC.dir/build.make:146: src/CMakeFiles/VC4CC.dir/GlobalValues.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
<command-line>: warning: "_GNU_SOURCE" redefined
<command-line>: note: this is the location of the previous definition
In file included from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/Module.h:12,
                 from /home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/Module.cpp:7:
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/SIMDVector.h: In constructor 'constexpr vc4c::SIMDVector::SIMDVector(std::initializer_list<vc4c::Literal>)':
/home/solstice/prog/dvcs/vc4c-git/src/VC4C/src/SIMDVector.h:53:24: error: call to non-'constexpr' function '_OIter std::copy_n(_IIter, _Size, _OIter) [with _IIter = const vc4c::Literal*; _Size = long unsigned int; _OIter = vc4c::Literal*]'
[... cut ...]
make: *** [Makefile:166: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

In fact, I tried to compile this project by error, and I was later told to use another one.

So I am gonna leave you there. Sorry.

If you want to test it yourself, you can try to install an archlinux VM/container. For easier install, you can try to install a manjaro VM, which is more easy and straightofrward, but will give you almost the same packages and build system that archlinux.

sorry not to help you on that.

BelegroM, 2022-12-13:
Hello all, as it is my first comment.

Looks like directories during compilation get messed up (have similar problem on manjaro, which is arch based anyway). I get the same error as @solsticedhiver, and from what i checked so far with this problem:

> make[2]: *** [src/CMakeFiles/VC4CC.dir/build.make:104: src/CMakeFiles/VC4CC.dir/Compiler.cpp.o] Error 1

That particular line in build.make looks like this:
`cd MY_HOME_DIR_HERE/vc4c-git/src/VC4C/build/src && /usr/bin/c++ $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -MD -MT src/CMakeFiles/VC4CC.dir/Compiler.cpp.o -MF CMakeFiles/VC4CC.dir/Compiler.cpp.o.d -o CMakeFiles/VC4CC.dir/Compiler.cpp.o -c /home/bob/AUR/vc4c-git/src/VC4C/src/Compiler.cpp`

The problem is `src/CMakeFiles/VC4CC.dir/Compiler.cpp.o` which makes this line cause error - should be like the next ones 'CMakeFiles/VC4CC.dir/Compiler.cpp.o'

commit b4b5cce3 (2023-01-07): Fix compilation for recent changes in cpplog and add support for LLVM 14

Fixes #161 and #162
closed 2023-01-07 by doe300