doe300/VC4C #165: RPI 3A+ VC4C Build error

jDietschi, 2023-06-19:
When building the VC4C Library (running make) it fails while trying to Link CXX shared library libVC4CC.so:

`
[  4%] Built target GetGitCommit
[  4%] Performing update step for 'cpplog-project'
From https://github.com/doe300/cpplog
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
[  4%] Performing configure step for 'cpplog-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/admin/opencl/VC4C/build/cpplog/src/cpplog-project-build
[  4%] Performing build step for 'cpplog-project'
[ 50%] Built target cpplog-static
[100%] Built target cpplog
[  4%] Built target cpplog-project-build
[  4%] Linking CXX shared library libVC4CC.so
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
make[2]: *** [src/CMakeFiles/VC4CC.dir/build.make:1555: src/libVC4CC.so.0.4.9999] Error 1
make[2]: *** Deleting file 'src/libVC4CC.so.0.4.9999'
make[1]: *** [CMakeFiles/Makefile2:353: src/CMakeFiles/VC4CC.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
`

The output in VC4C/build/CMakeFiles/CMakeError.log is the same as in issue 
`
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_c2e1a/fast && /usr/bin/make -f CMakeFiles/cmTC_c2e1a.dir/build.make CMakeFiles/cmTC_c2e1a.dir/build
make[1]: Entering directory '/home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_c2e1a.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_c2e1a.dir/src.c.o   -c /home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_c2e1a
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c2e1a.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_c2e1a.dir/src.c.o  -o cmTC_c2e1a
/usr/bin/ld: CMakeFiles/cmTC_c2e1a.dir/src.c.o: in function `main':
src.c:(.text+0x40): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x4c): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x5c): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_c2e1a.dir/build.make:87: cmTC_c2e1a] Error 1
make[1]: Leaving directory '/home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_c2e1a/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_9fcf5/fast && /usr/bin/make -f CMakeFiles/cmTC_9fcf5.dir/build.make CMakeFiles/cmTC_9fcf5.dir/build
make[1]: Entering directory '/home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_9fcf5.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_9fcf5.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_9fcf5
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_9fcf5.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_9fcf5.dir/CheckFunctionExists.c.o  -o cmTC_9fcf5  -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_9fcf5.dir/build.make:87: cmTC_9fcf5] Error 1
make[1]: Leaving directory '/home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_9fcf5/fast] Error 2

`

uname -a
`
Linux raspi01 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux
`

clang version:
`
clang version 7.0.1-8+rpi3+deb10u2 (tags/RELEASE_701/final)
Target: armv6k-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin
`

gcc version:
`
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
`

/proc/cpuinfo
`
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 9020e0
Serial          : 00000000c049d860
Model           : Raspberry Pi 3 Model A Plus Rev 1.0
`

doe300, 2023-06-19:
> collect2: fatal error: ld terminated with signal 9 [Killed]

The linker command seems to have been killed externally. Did you maybe run out of memory?

jDietschi, 2023-06-19:
I don't think so
I just ran another attempt without the kernel downgrade and got the same Error
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake cmTC_33e6d/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_33e6d.dir/build.make CMakeFiles/cmTC_33e6d.dir/build
gmake[1]: Entering directory '/home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_33e6d.dir/src.c.o
/usr/bin/cc   -DCMAKE_HAVE_LIBC_PTHREAD -o CMakeFiles/cmTC_33e6d.dir/src.c.o -c /home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_33e6d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_33e6d.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD -rdynamic CMakeFiles/cmTC_33e6d.dir/src.c.o -o cmTC_33e6d
/usr/bin/ld: CMakeFiles/cmTC_33e6d.dir/src.c.o: in function `main':
src.c:(.text+0x40): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x4c): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x58): undefined reference to `pthread_cancel'
/usr/bin/ld: src.c:(.text+0x68): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_33e6d.dir/build.make:106: cmTC_33e6d] Error 1
gmake[1]: Leaving directory '/home/admin/opencl/VC4C/build/CMakeFiles/CMakeTmp'
gmake: *** [Makefile:140: cmTC_33e6d/fast] Error 2
```
`
my swapfile is 4Gigabyte

admin@raspi01:~/opencl/VC4C/build/CMakeFiles $ free -h
                      total        used        free      shared  buff/cache   available
Mem:           426Mi        37Mi       148Mi       0.0Ki       240Mi       337Mi
Swap:          4.0Gi        18Mi       4.0Gi

What else could I do to avoid running out of memory?

And is this really compatible for Raspberry Pi 3A+?

doe300, 2023-06-19:
From the last log you posted, it looks like it worked, didn't it?

jDietschi, 2023-06-19:
It looks like the building worked correctly. All the output was successfull.
running clinfo still gives me the output
platforms
```
Number of platforms                               0`
```

doe300, 2023-06-19:
To be able to use VC4CL globally, you'll need to configure it with the CMake switch `BUILD_ICD=ON`, rebuild and reinstall. Then it should be listed (although it requires root rights).
Alternatively of course you could `LD_PRELOAD` the library before any OpenCL-client process invocation.

doe300, 2023-06-19:
The CMakeError.log is irrelevant. This is from an internal CMake capabilities test.

Can you run the clinfo with the `OCL_ICD_ENABLE_TRACE=1` and `VC4CL_DEBUG=api` environment variables?

Also, does the file `/etc/OpenCL/vendors/VC4CL.icd` exist and what is its content? Should be the path to the VC4CL library.

jDietschi, 2023-06-20:
Hi @doe300 

After recompiling the whole projekt using the BUILD_ICD=ON switch it seems functional!

Thank you so much for your support.
my `/etc/OpenCL/vendors/VC4CL.icd` file does correctly show the path to the VC4CL library:
/usr/local/libVC4CL.so

closed 2023-06-20 by jDietschi